openwrt/packages #17035: kmod-cfg80211: Package kmod-cfg80211 is missing dependencies for the following libraries (rfkill.ko)

grosjo, 2021-10-31:
Compiling latest git gives :

Package kmod-cfg80211 is missing dependencies for the following libraries:
rfkill.ko
make[3]: *** [Makefile:562: /data/src/openwrt/kuwfi830/bin/targets/ath79/generic/packages/kmod-cfg80211_5.4.155+5.15-rc6-1-1_mips_24kc.ipk] Error 1
make[3]: Leaving directory '/data/src/openwrt/kuwfi830/package/kernel/mac80211'
time: package/kernel/mac80211/compile#61.82#6.18#70.14
    ERROR: package/kernel/mac80211 failed to build.

dangowrt, 2021-10-31:
This is an issue with OpenWrt core rather than packages, so this is the wrong place for filing the bug. Use https://bugs.openwrt.org/ instead.

Apart from that, we shouldn't even be building `rfkill.ko` to begin with. Did you change kernel config to include it?

closed 2021-10-31 by dangowrt

grosjo, 2021-10-31:
Can you clarify " we shouldn't even be building rfkill.ko " ? Why not and if not, how to achieve avoiding rfkill ?

dangowrt, 2021-10-31:
There is a package called `kmod-rfkill` which is meant to be used in cases where rfkill cannot avoided (Bluetooth).
For WiFi, we generally build without rfkill support by default, there is a setting in menuconfig for that and unless you've touched that (ie. deliberately enabling rfkill in your build), cfg80211 should not require it.

[image]

In case you did enable rfkill in menuconfig you have discovered a small bug which requires this patch
```diff
diff --git a/package/kernel/mac80211/Makefile b/package/kernel/mac80211/Makefile
index 89e078b0c3..20795323d8 100644
--- a/package/kernel/mac80211/Makefile
+++ b/package/kernel/mac80211/Makefile
@@ -99,6 +99,7 @@ define KernelPackage/cfg80211
   TITLE:=cfg80211 - wireless configuration API
   DEPENDS+= +iw +iwinfo +wireless-regdb
   ABI_VERSION:=$(PKG_VERSION)-$(PKG_RELEASE)
+  $(call AddDepends/rfkill)
   FILES:= \
        $(PKG_BUILD_DIR)/compat/compat.ko \
        $(PKG_BUILD_DIR)/net/wireless/cfg80211.ko
```