openwrt/packages #17040: acme: can not find dns api hook

MarceloRuiz, 2021-11-01:
Maintainer: @tohojo 
Environment: arm, wrt1900ac, openwrt-21.02 branch (git-21.231.26241-422c175) / OpenWrt 21.02.0 r16279-5cc0535800 

Description:

Acme fails to create the certificate with dns challenge:
```
daemon.err run-acme[19902]: d_api
daemon.err run-acme[19902]: Can not find dns api hook for: dns_dynu
```

```
root@wrt1900ac:/usr/lib/acme# ./acme.sh --version
https://github.com/acmesh-official/acme.sh
v3.0.1

root@wrt1900ac:/# ls /usr/lib/acme/
acme.sh   run-acme
```
'acme.sh' and 'run-acme.sh' are installed in '/usr/lib/acme/' but the directory does not contain anything else, but if I run './acme.sh --upgrade' the script downloads everything to '/root/.acme.sh/', and this directory contains the dnsapi folder that contains the missing scripts:

```
root@wrt1900ac:/# ls /root/.acme.sh/
account.conf  acme.sh       deploy        dnsapi        notify

root@wrt1900ac2:/# ls /root/.acme.sh/dnsapi/
README.md             dns_clouddns.sh       dns_doapi.sh          dns_gandi_livedns.sh  dns_jd.sh             dns_misaka.sh         dns_nsone.sh          dns_rackspace.sh      dns_vultr.sh
dns_1984hosting.sh    dns_cloudns.sh        dns_domeneshop.sh     dns_gcloud.sh         dns_joker.sh          dns_myapi.sh          dns_nsupdate.sh       dns_rcode0.sh         dns_websupport.sh
dns_acmedns.sh        dns_cn.sh             dns_dp.sh             dns_gd.sh             dns_kappernet.sh      dns_mydevil.sh        dns_nw.sh             dns_regru.sh          dns_world4you.sh
dns_acmeproxy.sh      dns_conoha.sh         dns_dpi.sh            dns_gdnsdk.sh         dns_kas.sh            dns_mydnsjp.sh        dns_oci.sh            dns_scaleway.sh       dns_yandex.sh
dns_active24.sh       dns_constellix.sh     dns_dreamhost.sh      dns_he.sh             dns_kinghost.sh       dns_namecheap.sh      dns_one.sh            dns_schlundtech.sh    dns_zilore.sh
dns_ad.sh             dns_cpanel.sh         dns_duckdns.sh        dns_hetzner.sh        dns_knot.sh           dns_namecom.sh        dns_online.sh         dns_selectel.sh       dns_zone.sh
dns_ali.sh            dns_cx.sh             dns_durabledns.sh     dns_hexonet.sh        dns_leaseweb.sh       dns_namesilo.sh       dns_openprovider.sh   dns_servercow.sh      dns_zonomi.sh
dns_anx.sh            dns_cyon.sh           dns_dyn.sh            dns_hostingde.sh      dns_lexicon.sh        dns_nederhost.sh      dns_openstack.sh      dns_simply.sh
dns_arvan.sh          dns_da.sh             dns_dynu.sh           dns_huaweicloud.sh    dns_linode.sh         dns_neodigit.sh       dns_opnsense.sh       dns_tele3.sh
dns_aurora.sh         dns_ddnss.sh          dns_dynv6.sh          dns_infoblox.sh       dns_linode_v4.sh      dns_netcup.sh         dns_ovh.sh            dns_transip.sh
dns_autodns.sh        dns_desec.sh          dns_easydns.sh        dns_infomaniak.sh     dns_loopia.sh         dns_netlify.sh        dns_pdns.sh           dns_ultra.sh
dns_aws.sh            dns_df.sh             dns_edgedns.sh        dns_internetbs.sh     dns_lua.sh            dns_nic.sh            dns_pleskxml.sh       dns_unoeuro.sh
dns_azion.sh          dns_dgon.sh           dns_euserv.sh         dns_inwx.sh           dns_maradns.sh        dns_njalla.sh         dns_pointhq.sh        dns_variomedia.sh
dns_azure.sh          dns_dnsimple.sh       dns_exoscale.sh       dns_ionos.sh          dns_me.sh             dns_nm.sh             dns_porkbun.sh        dns_veesp.sh
dns_cf.sh             dns_do.sh             dns_freedns.sh        dns_ispconfig.sh      dns_miab.sh           dns_nsd.sh            dns_rackcorp.sh       dns_vscale.sh
```
Are the files/folders installed by the acme upgrade missing in the default installation directory? If there are somewhere else, it seems that 'acme.sh' cannot find them

brada4, 2021-11-01:
they are in acme-dnsapi

brada4, 2021-11-01:
i am just another user, probably rename issue w error examples, that busybox sed does not cut it

tohojo, 2021-11-01:
Andrew ***@***.***> writes:

> i am just another user, probably rename issue w error examples, that
> busybox sed does not cut it

We're not doing anything particularly fancy with sed - what's the error,
exactly?

MarceloRuiz, 2021-11-02:
There are multiple lines on the log that say 'No -i support in sed'. Here is one example:
 
```
[Tue Nov  2 11:07:46 EDT 2021] Retrying post
[Tue Nov  2 11:07:46 EDT 2021] POST
[Tue Nov  2 11:07:46 EDT 2021] _post_url='https://acme-staging-v02.api.letsencrypt.org/acme/authz-v3/829713678'
[Tue Nov  2 11:07:46 EDT 2021] _WGET='wget -q --content-on-error '
[Tue Nov  2 11:07:47 EDT 2021] No -i support in sed
```

Just to be clear, the sed -i related errors are gone after installing 'sed' using opkg.

My point regarding dependencies was that if 'luci-app-acme' provides an option to use dns validation, maybe it should mark 'acme-dnsapi' and 'sed' as dependencies so they are installed automatically. Should I create a new issue for this?

tohojo, 2021-11-02:
MarceloRuiz ***@***.***> writes:

> There are multiple lines on the log that say 'No -i support in sed'. Here is one example:
>
> ```
> [Tue Nov  2 11:07:46 EDT 2021] Retrying post
> [Tue Nov  2 11:07:46 EDT 2021] POST
> [Tue Nov  2 11:07:46 [...]

But the question is more why is your busybox 'sed' not supporting -i?
Works just fine on my openwrt install:

sed -i
BusyBox v1.33.1 (2021-08-31 22:20:08 UTC) multi-call binary.

Usage: sed [-i[SFX]] [-nrE] [-f FILE]... [-e CMD]... [FILE]...
or: sed [-i[SFX]] [-nrE] CMD [FILE]...

	-e CMD	Add CMD to sed commands to be executed
	-f FILE	Add FILE contents to sed commands to be executed
	-i[SFX]	Edit files in-place (otherwise sends to stdout)
		Optionally back files up, appending SFX
	-n	Suppress automatic printing of pattern space
	-r,-E	Use extended regex syntax

If no -e or -f, the first non-option argument is the sed command string.
Remaining arguments are input files (stdin if none).

> My point regarding dependencies was that if 'luci-app-acme' provides
> an option to use dns validation, maybe it should mark 'acme-dnsapi'
> and 'sed' as dependencies so they are installed automatically. Should
> I create a new issue for this?

No, the split was on purpose (into acme and acme-dnsapi): if you don't
need the DNS API it should be possible to install acme without it. If we
introduce a dependency this will no longer be possible.

The sed issue should be fixed, though; see above.

MarceloRuiz, 2021-11-02:
I have no idea of what is going on with sed, because I get the same output you get:
```
root@wrt1900ac2:~# sed -i
BusyBox v1.33.1 (2021-10-09 02:34:51 UTC) multi-call binary.

Usage: sed [-i[SFX]] [-nrE] [-f FILE]... [-e CMD]... [FILE]...
or: sed [-i[SFX]] [-nrE] CMD [FILE]...

	-e CMD	Add CMD to sed commands to be executed
	-f FILE	Add FILE contents to sed commands to be executed
	-i[SFX]	Edit files in-place (otherwise sends to stdout)
		Optionally back files up, appending SFX
	-n	Suppress automatic printing of pattern space
	-r,-E	Use extended regex syntax

If no -e or -f, the first non-option argument is the sed command string.
Remaining arguments are input files (stdin if none).
```
Regarding the package splitting, I understand the advantages of having 'acme-dnsapi' separated from 'acme'. I thought the separation was there for users that don't use luci.
If we consider things from the average user point of view, I think the expectation is that once someone installs 'luci-app-acme' everything related to it should be provided for it to just work out-of-the-box. 
Maybe another solution would be that when someone selects DNS as the validation method, the UI could check if 'acme-dnsapi' is installed and provide a clear error message (I know the message 'Using this mode requires the acme-dnsapi package to be installed.' is there, but if an error message just appears when the user makes the selection, it will certainly catch his/hers attention).
Anyway, I know all the information is in the UI and that it was clearly my fault not to catch the need to install 'acme-dnsapi'.

tohojo, 2021-11-02:
MarceloRuiz ***@***.***> writes:

> I have no idea of what is going on with sed, because I get the same
> output you get:

Ah, this message is actually coming from acme.sh itself, because it
looks for a different pattern in the 'sed' help output:

https://github.com/acmesh-official/acme.sh/blob/master/acme.sh#L905-L912

But as you can also see from the text immediately after the debug
message, it's actually completely harmless, as it'll just fall back to a
different replacement method... So it's safe to just ignore this
particular error message.

MarceloRuiz, 2021-11-02:
@toho 
Thanks for clarifying the error can be ignored.
Does luci-app-acme provide any way to configure deploy hooks and notifications? I noticed those directories that are part of the original 'acme.sh' script installation are missing.

tohojo, 2021-11-02:
MarceloRuiz ***@***.***> writes:

> @toho 
> Thanks for clarifying the error can be ignored.
> Does luci-app-acme provide any way to configure deploy hooks and
> notifications? I noticed those directories that are part of the
> original 'acme.sh' script installation are missing.

Nope :)

closed 2021-11-03 by MarceloRuiz