openwrt/packages #17062: xtables-addons build fails on master X86_64

Arduous, 2021-11-04:
Maintainer: @pprindeville 
Environment: (Generic x86_64, OpenWrt master)

Description:

Build fails on a brand new checked out environment. Thank you

OpenWrt branch is master, commit 1ee6d3d24e328e29fcf068e90953c760305462d4
Packages feed is master, commit e7b21be6fad60b447b30414967f413534c62e59d

Relevant .config architecture excerpt

```
CONFIG_TARGET_x86=y
# CONFIG_TARGET_x86_generic is not set
# CONFIG_TARGET_x86_legacy is not set
# CONFIG_TARGET_x86_geode is not set
CONFIG_TARGET_x86_64=y
# CONFIG_TARGET_MULTI_PROFILE is not set
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_HAS_SUBTARGETS=y
CONFIG_HAS_DEVICES=y
CONFIG_TARGET_BOARD="x86"
CONFIG_TARGET_SUBTARGET="64"
CONFIG_TARGET_PROFILE="DEVICE_generic"
CONFIG_TARGET_ARCH_PACKAGES="x86_64"
CONFIG_DEFAULT_TARGET_OPTIMIZATION="-Os -pipe"
```

```
time: package/network/utils/iptables/compile#0.21#0.05#0.25
make[3]: Entering directory '/opt/samuel/openwrt_x86_64/feeds/packages/net/xtables-addons'
rm -f /opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/.built
touch /opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/.built_check
make  -C /opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13 KCFLAGS="-fmacro-prefix-map=/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/opt/samuel/openwrt_x86_64/staging_dir/host/include -I/opt/samuel/openwrt_x86_64/staging_dir/hostpkg/include -I/opt/samuel/openwrt_x86_64/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Nov  4 06:00:21 2021" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/opt/samuel/openwrt_x86_64/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/opt/samuel/openwrt_x86_64/staging_dir/host/lib" CONFIG_SHELL="bash" V=1  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/symvers/button-hotplug.symvers" KERNELRELEASE=5.10.76 DESTDIR="/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/ipkg-install" DEPMOD="/bin/true" all
make[4]: Entering directory '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13'
make  all-recursive
make[5]: Entering directory '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13'
Making all in extensions
make[6]: Entering directory '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions'
Xtables-addons 3.13 - Linux 5.10.76
if [ -n "/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.76" ]; then make -C /opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.76 M=/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions modules; fi;
make[7]: Entering directory '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.76'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (                \
echo >&2;                                                       \
echo >&2 "  ERROR: Kernel configuration is invalid.";           \
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";      \
echo >&2 ;                                                      \
/bin/false)
make -f ./scripts/Makefile.build obj=/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions \
single-build= \
need-builtin=1 need-modorder=1
make -f ./scripts/Makefile.build obj=/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions/ACCOUNT \
 \
need-builtin= \
need-modorder=1
make[9]: *** No rule to make target '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions/ACCOUNT/xt_ACCOUNT.o', needed by '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions/ACCOUNT/xt_ACCOUNT.mod'.  Stop.
make[8]: *** [scripts/Makefile.build:497: /opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions/ACCOUNT] Error 2
make[7]: *** [Makefile:1822: /opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions] Error 2
make[7]: Leaving directory '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.76'
make[6]: *** [Makefile:461: modules] Error 2
make[6]: Leaving directory '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions'
make[5]: *** [Makefile:495: all-recursive] Error 1
make[5]: Leaving directory '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13'
make[4]: *** [Makefile:380: all] Error 2
make[4]: Leaving directory '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13'
make[3]: *** [Makefile:177: /opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/.built] Error 2
make[3]: Leaving directory '/opt/samuel/openwrt_x86_64/feeds/packages/net/xtables-addons'
time: package/feeds/packages/xtables-addons/compile#0.96#0.25#1.18
    ERROR: package/feeds/packages/xtables-addons failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/xtables-addons/compile] Error 1
make[2]: Leaving directory '/opt/samuel/openwrt_x86_64'
make[1]: *** [package/Makefile:110: /opt/samuel/openwrt_x86_64/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/opt/samuel/openwrt_x86_64'
make: *** [/opt/samuel/openwrt_x86_64/include/toplevel.mk:230 : world] Erreur 2

```
[config_x86_64.txt](https://github.com/openwrt/packages/files/7477156/config_x86_64.txt)

jow-, 2021-11-04:
Please retry after the following:
`find staging_dir/host -type f -name '*elf*' | xargs rm -v`

Arduous, 2021-11-04:
Thank you. I ran the first version of your snippet. Some files were deleted, but the same error happened.

I then `make dirclean` but xtables-addons build keeps on failing.

```
samuel@bioman2:/opt/samuel/openwrt_x86_64$ find staging_dir/host -type f -name "*elf*" | xargs rm -v
'staging_dir/host/bin/mklibs-readelf' supprimé
'staging_dir/host/bin/patchelf' supprimé
'staging_dir/host/stamp/.patchelf_installed' supprimé
'staging_dir/host/include/elf.h' supprimé
'staging_dir/host/share/cmake-3.21/Help/module/FindSelfPackers.rst' supprimé
'staging_dir/host/share/cmake-3.21/Modules/FindSelfPackers.cmake' supprimé
```

Arduous, 2021-12-03:
I tried again on a vanilla checkout on current master.
If I run `make menuconfig`and leave everything to its default value (Atheros ATH79/Generic/8devices Carambola2) outside the selection Network/Firewall/iptgeoip , I can build the image(s). If I change the target to x86/x86_64/Generic x86/64 the compilation of xtables-addons fails with the above exposed error.

It seems to me that it is target specific.

pprindeville, 2021-12-11:
@jow- Any ideas?

Arduous, 2022-03-13:
My latest build attempt, on r19090, completed successfully!

Arduous, 2022-04-13:
Hello, I am reopening this issue, as only the **first** build after a `make dirclean` with a very conservative `make -j1 V=sc` is completing successfully. Any subsequent build fails invariably with the same error as before.

I am currently on the master branch. It seems that the xtables-addons is now in version 3.19 .

With the following command, the error message is slightly different

```
make -j1 V=99 package/xtables-addons/{clean,prepare,compile,install}

.....
make[3]: Entering directory '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.19'
make  all-recursive
make[4]: Entering directory '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.19'
Making all in extensions
make[5]: Entering directory '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.19/extensions'
Xtables-addons 3.19 - Linux 5.10.110
if [ -n "/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.110" ]; then make -C /opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.110 M=/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.19/extensions modules; fi;
make[6]: Entering directory '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.110'
make[8]: *** No rule to make target '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.19/extensions/ACCOUNT/xt_ACCOUNT.o', needed by '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.19/extensions/ACCOUNT/xt_ACCOUNT.mod'.  Stop.
make[7]: *** [scripts/Makefile.build:497: /opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.19/extensions/ACCOUNT] Error 2
make[6]: *** [Makefile:1822: /opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.19/extensions] Error 2
make[6]: Leaving directory '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.110'
make[5]: *** [Makefile:461: modules] Error 2
make[5]: Leaving directory '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.19/extensions'
make[4]: *** [Makefile:495: all-recursive] Error 1
make[4]: Leaving directory '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.19'
make[3]: *** [Makefile:380: all] Error 2
make[3]: Leaving directory '/opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.19'
make[2]: *** [Makefile:177: /opt/samuel/openwrt_x86_64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.19/.built] Error 2
make[2]: Leaving directory '/opt/samuel/openwrt_x86_64/feeds/packages/net/xtables-addons'
time: package/feeds/packages/xtables-addons/compile#7.07#0.73#8.73
    ERROR: package/feeds/packages/xtables-addons failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/xtables-addons/compile] Error 1
make[1]: Leaving directory '/opt/samuel/openwrt_x86_64'
make: *** [/opt/samuel/openwrt_x86_64/include/toplevel.mk:230 : package/xtables-addons/compile] Erreur 2
```

Arduous, 2022-04-18:
I made several tests over the past days.

Setup:

1. Cleanup with make dist-clean
2. Update and installation of the packages
3. Copy of a saved .config and make meunconfig
4. make menuconfig
5. make download

I was able to build successfully the image on an old laptop, even with 4 concurrent threads (Core i5-2520M). This laptop was reinstalled with Ubuntu 21.10 2 months ago.
I am not able to build the image on my desktop (Ryzen 9 3900X), even with a single concurrent thread, starting either from scratch or from the SDK. This desktop is in 21.10, and has a long history of updates. I am building OpenWRT there for at least 2 years.

I will perform some more tests on my desktop like building within a VM, building older revisions booting it with a USB stick, performing a full memtest, and report.

At the end of the day, it looks more like a problem with the build system or my computer. Feel free to advise on a more appropriate ticket title.

vndroid, 2022-09-12:
Same issues

```
make[4]: Entering directory '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21'
make  all-recursive
make[5]: Entering directory '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21'
Making all in extensions
make[6]: Entering directory '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions'
Xtables-addons 3.21 - Linux 5.10.138
if [ -n "/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.138" ]; then make -C /data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.138 M=/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions modules; fi;
make[7]: Entering directory '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.138'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (                \
echo >&2;                                                       \
echo >&2 "  ERROR: Kernel configuration is invalid.";           \
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";      \
echo >&2 ;                                                      \
/bin/false)
make -f ./scripts/Makefile.build obj=/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions \
single-build= \
need-builtin=1 need-modorder=1
make -f ./scripts/Makefile.build obj=/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/ACCOUNT \
 \
need-builtin= \
need-modorder=1
make[9]: *** No rule to make target '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/ACCOUNT/xt_ACCOUNT.o', needed by '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/ACCOUNT/xt_ACCOUNT.mod'.  Stop.
make[8]: *** [scripts/Makefile.build:503: /data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/ACCOUNT] Error 2
make[7]: *** [Makefile:1832: /data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions] Error 2
make[7]: Leaving directory '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.138'
make[6]: *** [Makefile:461: modules] Error 2
make[6]: Leaving directory '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions'
make[5]: *** [Makefile:495: all-recursive] Error 1
make[5]: Leaving directory '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21'
make[4]: *** [Makefile:380: all] Error 2
make[4]: Leaving directory '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21'
make[3]: *** [Makefile:177: /data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/.built] Error 2
make[3]: Leaving directory '/data/openwrt/feeds/packages/net/xtables-addons'
time: package/feeds/packages/xtables-addons/compile#5.86#0.65#7.44
    ERROR: package/feeds/packages/xtables-addons failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/xtables-addons/compile] Error 1
make[2]: Leaving directory '/data/openwrt'
make[1]: *** [package/Makefile:110: /data/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/data/openwrt'
make: *** [/data/openwrt/include/toplevel.mk:230: world] Error 2
```

vndroid, 2022-09-12:
> Same issues
>
> ```
> make[4]: Entering directory '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21'
> make  all-recursive
> make[5]: Entering directory [...]

With the following command, build the xtables single.

```bash
make[3]: Entering directory '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21'
make  all-recursive
make[4]: Entering directory '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21'
Making all in extensions
make[5]: Entering directory '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions'
Xtables-addons 3.21 - Linux 5.10.138
if [ -n "/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.138" ]; then make -C /data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.138 M=/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions modules; fi;
make[6]: Entering directory '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.138'
make[8]: *** No rule to make target '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/ACCOUNT/xt_ACCOUNT.o', needed by '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/ACCOUNT/xt_ACCOUNT.mod'.  Stop.
make[7]: *** [scripts/Makefile.build:503: /data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/ACCOUNT] Error 2
make[6]: *** [Makefile:1832: /data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions] Error 2
make[6]: Leaving directory '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.138'
make[5]: *** [Makefile:461: modules] Error 2
make[5]: Leaving directory '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions'
make[4]: *** [Makefile:495: all-recursive] Error 1
make[4]: Leaving directory '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21'
make[3]: *** [Makefile:380: all] Error 2
make[3]: Leaving directory '/data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21'
make[2]: *** [Makefile:177: /data/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/.built] Error 2
make[2]: Leaving directory '/data/openwrt/feeds/packages/net/xtables-addons'
time: package/feeds/packages/xtables-addons/compile#11.53#1.44#13.59
    ERROR: package/feeds/packages/xtables-addons failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/xtables-addons/compile] Error 1
make[1]: Leaving directory '/data/openwrt'
make: *** [/data/openwrt/include/toplevel.mk:230: package/xtables-addons/compile] Error 2
```

Arduous, 2022-09-12:
On which OS/HW are you encountering the issue?

vndroid, 2022-09-12:
builder: Ubuntu 22.04
HW: x86_64

Arduous, 2022-09-12:
More specifically, what is the exact CPU model?

vndroid, 2022-09-12:
> More specifically, what is the exact CPU model?

Compiled using xeon e2378, ready to use in ESXi.

Arduous, 2022-09-12:
Which completely voids my suspicion that it was related to AMD... TY

pprindeville, 2022-09-19:
I'm building on Ubuntu 22.04 LTS on an VM w/ Xeon-D h/w and CPUID passthru without issues.

hifitime, 2022-11-24:
Same issues
builder: Ubuntu 22.04
HW: x86_64
Target：arm64

make -j8
`
 make[3] -C package/utils/f2fs-tools clean-build
 make[3] -C package/utils/f2fs-tools compile
 make[3] -C feeds/store/luci/taskd compile
 make[3] -C package/istoreos-files compile
 make[3] -C package/system/fstools clean-build
 make[3] -C package/system/fstools compile
 make[3] -C feeds/luci/modules/luci-base clean-build
 make[3] -C feeds/luci/modules/luci-base compile
 make[3] -C feeds/store/luci/luci-lib-taskd compile
 make[3] -C package/boot/uboot-envtools clean-build
 make[3] -C package/boot/uboot-envtools compile
 make[3] -C feeds/luci/libs/luci-lib-ipkg compile
    ERROR: package/transplant/xtables-addons failed to build.
 make[3] -C package/kernel/mac80211 clean-build
 make[3] -C package/kernel/mac80211 compile
 make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/wu/build/istoreos/istoreos/include/toplevel.mk:234: world] Error 1
`

Arduous, 2022-11-27:
For some reason, I am now able to build xtables-addons on both machines. I did not do anything particular, outside of git pulling the latest OpenWRT and packages.
Most striking difference for me is the kernel, which was upgraded from 5.10 to 5.15.

Arduous, 2023-05-25:
Compilation consistently succeeds now for months. Closing.

closed 2023-05-25 by Arduous

pprindeville, 2023-07-11:
I'm seeing issues with `xtables-addons` failing for various platforms:

https://github.com/openwrt/packages/pull/21528

even architectures (like `x86_64`) that build for me just fine locally.