openwrt/packages #17072: v4l2tools: fails to build on arm_cortex-a7_neon-vfpv4

PolynomialDivision, 2021-11-05:
@mpromonet 

```
[0/1] Install the project...
-- Install configuration: "Release"
-- Installing: /builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/v4l2tools-0.1.5/ipkg-install/usr/bin/v4l2tools-0.1.5
Package v4l2tools is missing dependencies for the following libraries:
libvpx.so.6
make[3]: *** [Makefile:47: /builder/shared-workdir/build/sdk/bin/packages/arm_cortex-a7_neon-vfpv4/packages/v4l2tools_0.1.5-1_arm_cortex-a7_neon-vfpv4.ipk] Error 1
time: package/feeds/packages/v4l2tools/compile#8.64#1.44#32.62
```

https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a7_neon-vfpv4/packages/v4l2tools/compile.txt

PolynomialDivision, 2021-11-05:
https://github.com/openwrt/packages/pull/17075

neheb, 2021-11-12:
Should be fixed now.

closed 2021-11-26 by PolynomialDivision