openwrt/packages #17074: mariadb: fails to build on arm_cortex-a7_neon-vfpv4

PolynomialDivision, 2021-11-05:
@nxhack

```
In file included from /usr/include/c++/8/memory:62,
                 from ../deps/v8/src/libplatform/default-foreground-task-runner.h:8,
                 from ../deps/v8/src/libplatform/default-foreground-task-runner.cc:5:
/usr/include/c++/8/bits/stl_algobase.h:59:10: fatal error: bits/c++config.h: No such file or directory
 #include <bits/c++config.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```
https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a7_neon-vfpv4/packages/node/compile.txt

PolynomialDivision, 2021-11-05:
https://github.com/openwrt/packages/pull/17076

nxhack, 2021-11-05:
@PolynomialDivision 

```
  *** The following preparations must be made on the host side. ***
      1. gcc 8.3 or higher is required.
      2. To build a 32-bit target, gcc-multilib, g++-multilib are required.
      3. Requires libatomic package. (If necessary, install the 32-bit library at the same time.)
     ex) sudo apt-get install gcc-multilib g++-multilib
```

PolynomialDivision, 2021-11-05:
@nxhack Thanks already saw it. Just looked how I can also tell that buildbot. ;) I'm just playing around so it will not appear anymore in faillogs.

nxhack, 2021-11-05:
@PolynomialDivision 
It seems to be the policy of the buildbot administrator, and the same goes for CI. I can't do anything about it.

PolynomialDivision, 2021-11-05:
@nxhack I thought for such things we can require hosts dependencies? I am not experienced in it so I need some time to find a working solution. Did you already try and it was not working?

nxhack, 2021-11-05:
@PolynomialDivision 
Read the source code of Openwrt's build system; OS package dependencies are not supported.

https://github.com/openwrt/openwrt/blob/master/include/package-dumpinfo.mk
https://github.com/openwrt/openwrt/blob/master/scripts/metadata.pm

PolynomialDivision, 2021-11-05:
You mean there is no way to not fail this package in buildbot?

@aparcar @neheb ?

neheb, 2021-11-05:
Nope. Upstream does not want to add GCC-multilib

PolynomialDivision, 2021-11-05:
But why does [luajit](https://github.com/openwrt/packages/blob/master/lang/luajit/Makefile) not fail? [Here](https://openwrt.org/docs/guide-developer/toolchain/install-buildsystem#debianubuntu) it is written that it also depends on gcc-multilib.

PolynomialDivision, 2021-11-05:
Is that here the "magic"?
https://github.com/openwrt/packages/blob/79a9549940193d7b641cfd160fc22ff08dec8d4c/lang/luajit/Makefile#L30-L37

BKPepe, 2021-11-05:
> But why does [luajit](https://github.com/openwrt/packages/blob/master/lang/luajit/Makefile) not fail? [Here](https://openwrt.org/docs/guide-developer/toolchain/install-buildsystem#debianubuntu) it is written that it also depends on gcc-multilib.

Because I fixed it. https://github.com/openwrt/packages/pull/17076#issuecomment-961734023

closed 2021-12-13 by aparcar