openwrt/packages #17088: tailscale: backport to 19.07

bam80, 2021-11-07:
Maintainer: @ja-pa 
Environment: (mipsel_24kc, Kroks KNdRt31R19, OpenWrt 19.07)

Description: This is a request to backport the Tailscale new package #15539 found in 21.02 to 19.07

BKPepe, 2021-11-18:
Hello @bambo80,

Thank you for your feature request. However, I can not comply with it. 

OpenWrt 19.07 is the stable branch, and until March 2022, it is in the security maintenance mode, and then it will be End-of-Life. It means that no new packages are going to be introduced for this version.

It is suggested to use the latest OpenWrt stable release which is 21.02.

Refer to this announcement: https://lists.openwrt.org/pipermail/openwrt-announce/2021-November/000023.html

closed 2021-11-18 by BKPepe

bam80, 2021-11-18:
Thank you for the explanation!

About 21.02, I can't use it right now because my device has forked OpenWrt.
I opened a discussion on forum about merging it upstream, maybe I could do it myself:
https://forum.openwrt.org/t/merge-support-for-kroks-routers/111746