openwrt/packages #17092: lm-sensors compilation failed

Co1lin, 2021-11-07:
When I tried to compile the openwrt for my router, I encountered this error: 

```
arm-openwrt-linux-muslgnueabi-gcc  -o prog/sensors/sensors prog/sensors/main.ro prog/sensors/chips.ro -liconv -Llib -lsensors /home/colin/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: cannot find -liconv
collect2: error: ld returned 1 exit status
make[4]: *** [prog/sensors/Module.mk:43: prog/sensors/sensors] Error 1
make[4]: Leaving directory '/home/colin/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lm-sensors-3-6-0'
make[3]: *** [Makefile:129: /home/colin/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lm-sensors-3-6-0/.built] Error 2
make[3]: Leaving directory '/home/colin/openwrt/feeds/packages/utils/lm-sensors'
time: package/feeds/packages/lm-sensors/compile#0.18#0.08#0.44
    ERROR: package/feeds/packages/lm-sensors failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/lm-sensors/compile] Error 1
```

It seems that there's something wrong with `lm-sensors`. How can I fix it?

hzjnet, 2022-07-05:
我也遇见了  不知道如何处理

hnws, 2024-12-31:
The issue still happens in latest 24.10-rc4.
I believe the root cause is that lm_sensors uses 
`LIBICONV := $(shell if /sbin/ldconfig -p | grep -q '/libiconv\.so$$' ; then echo \-liconv; else echo; fi)` in `prog/sensors/Module.mk` to detect libiconv. This somehow detects host libiconv package.
In the lm-sensors, the EXLDFLAGS are set to empty. And there is no explicit mention of libiconv-full package as dependency. libiconv was removed after openwrt 22.03. And currently in openwrt there is only libiconv-full which installs .so files into a sub directory `libiconv-full/lib` inside `/usr/lib`.

So the solution is either

1. let libiconv-full to install .so into standard directory: `/usr/lib`.
2. change `EXLDFLAGS` in lm-sensors to be `EXLDFLAGS="-L$(STAGING_DIR)/usr/lib/libiconv-full/lib/"`

ChrisEAlfred, 2025-05-02:
What is the complete path and file to change EXLDFLAGS for lm-sensors?

hnws, 2025-06-03:
utils/lm-sensors/Makefile
line:90