openwrt/packages #17093: OWRT 21.02.1 - "opkg install ipset" fails. Software build with imagebuilder.

ghmon, 2021-11-07:
GL:root:~:# uname -a
Linux df-tsn-v031-OpenWrt 5.4.154 #0 Sun Oct 24 09:01:35 2021 mips GNU/Linux

GL:root:~:# opkg install ipset
+ opkg install ipset
Installing ipset (7.6-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.0/packages/mipsel_24kc/base/ipset_7.6-1_mipsel_24kc.ipk
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-81b5fa8a3bdde9109c65df5b7a2d086e) for kmod-nf-reject
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-81b5fa8a3bdde9109c65df5b7a2d086e) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-81b5fa8a3bdde9109c65df5b7a2d086e) for kmod-ipt-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-81b5fa8a3bdde9109c65df5b7a2d086e) for kmod-nfnetlink
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nfnetlink found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-81b5fa8a3bdde9109c65df5b7a2d086e) for kmod-ipt-ipset
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-ipset found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ipset:
 * 	kernel (= 5.4.143-1-81b5fa8a3bdde9109c65df5b7a2d086e)
 * opkg_install_cmd: Cannot install package ipset.

stangri, 2021-11-07:
Your `/etc/config/distfeeds.conf` probably contains links to the 21.02.0 packages instead of 21.02.1. Was it restored from backup or otherwise edited?

ghmon, 2021-11-07:
> Your `/etc/config/distfeeds.conf` probably contains links to the 21.02.0 packages instead of 21.02.1. Was it restored from backup or otherwise edited?

Oh yes... My fault. I had a wrong (not needed) feed conf in "files/" of the imagebuilder directory... Sorry!

closed 2021-11-07 by ghmon