openwrt/packages #17095: auc: error on start

bam80, 2021-11-08:
Maintainer: @aparcar
Environment: (Qualcomm Atheros QCA956X ver 1 rev 0, TP-Link Archer C6 v2, OpenWrt 19.07.3 r11063-85e04e9f46 / LuCI openwrt-19.07 branch git-20.136.49537-fb2f363)

Description:
Can't run the util:
```
# auc 
auc (0.1.4)
cannot request packagelist from rpcd
```

aparcar, 2021-11-08:
Restart `rpcd` via `service rpcd restart`. It should work now.

@dangowrt please merge this openwrt.git fix you proposed a while ago.

bam80, 2021-11-08:
Some progress but still no luck:
```
root@OpenWrt:~# /etc/init.d/rpcd restart
root@OpenWrt:~# auc
auc (0.1.4)
running OpenWrt 19.07.3 on ath79/generic (tplink,archer-c6-v2)
checking https://sysupgrade.openwrt.org for release upgrade or updated packages
HTTP error 404
server reply invalid.

aparcar, 2021-11-08:
I agree with @dangowrt , sadly some parts are missing which makes it impossible to offer sysupgrades reliable on 19.07.x. Please upgrade to 21.02.x and enjoy attended sysupgrades from then on.

bam80, 2021-11-08:
> Please upgrade to 21.02.x 

Already did, thanks.

It's a bit off-topic, but still [The OpenWrt Firmware Selector ](https://forum.openwrt.org/t/the-openwrt-firmware-selector/81721) seems better fit for me as I don't actually wand _to rebuild_ the image - what I want is just a convenient way to upgrade with official builds..

aparcar, 2021-11-08:
@bam80 you can just run `auc` and it will upgrade your device

dangowrt, 2021-11-09:
Yes, `auc` always uses the (official) ImageBuilder through @aparcar's [asu](https://github.com/aparcar/asu).

Upgrading the device using the released sysupgrade images from https://downloads.openwrt.org always means that you loose all user-installed packages, which can result in loosing WAN access (as you additional packages may be needed even for that). `auc` and `luci-app-attendedsysupgrade` try to solve this problem.

If you do not trust the ImageBuilder-servers provided by others (and it's generally a good idea not to trust anything) you can easily run your own server using [Docker](https://github.com/aparcar/asu#docker).

bam80, 2021-11-09:
Loosing an user-installed packages is probably not a problem as those can be installed in script afterwards.

Loosing a WAN access is an exception and the only case when the rebuilding looks reasonable (do not taking into account you could squeeze a bit more space embedding your packages in the image, of course)

aparcar, 2021-11-09:
@bam80 I don't get your point. If you want convenience use this service with one of provided apps, if you want to manually verify things, do it manually. Not loosing user installed packages is the very point of this service.

Since there is no way for a router to know if it's running on vanilla Buildbot settings (e.g. having the default packages installed) or some special packages are added, there is no easy way to implement that.

closed 2021-11-09 by aparcar

bam80, 2021-11-09:
I want convenience not sacrificing security.  

> Since there is no way for a router to know if it's running on vanilla Buildbot settings 

Maybe clever checksum comparison of data in flash partitions will show us? 

Either way, I don't think that guess is necessary - the util could just provide us an option to flash custom-build image with all the packages, or vanilla image with no packages (for now).
As list of installed packages can be saved on upgrade by stock OpenWrt already - I don't see a problem with latter case..

aparcar, 2021-11-10:
PR welcome.

dangowrt, 2021-11-10:
As the service carries out a supposedly reproducible task, one way out of the dilemma can be to let `asu` also provide a hash of the image having it's (build-host specific) signature stripped, so this hash is identical and would present a reproducible 1:1 mapping of request hash => image hash.
Then paranoid users could have the same build carried out by several builders and we could make clients make sure the hashes match between all build providers and with the downloaded file.

aparcar, 2021-11-10:
While possible this seems silly since buildbot images don't ship auc nor the LuCI app. You would never have a buildbot checksum to compare to

dangowrt, 2021-11-10:
> While possible this seems silly since buildbot images don't ship auc nor the LuCI app. You would never have a buildbot checksum to compare to

Imho the problem right now is if I send the exact same request for an image to two different `asu` instances, they will return two different images which different hashes.
This is because each instance runs with an individual `usign` private key which is used to sign the image using `ucert` before handing it out to the client. If images (and hence hashes) resulting from the same request would be shipped in a way which makes them easy to compare, clients could request them from several servers (run by different people in different locations) to make sure what they are getting hasn't been infiltrated by an attacker who has compromised one of the `asu` instances.