openwrt/packages #17096: Failed to download any source that's from GitHub.

odysseus443, 2021-11-08:
Please make sure that the issue subject starts with `banip `

Maintainer: @\dibdot
Environment: (X86_64, OpenWrt 21.02.1 r16325-88151b8303)

Description:
`uclient-fetch` failed to download any source that's from GitHub.

Error log below:
```
user.debug banIP-0.7.10[14977]: f_down  ::: name: doh_6, url: https://raw.githubusercontent.com/dibdot/DoH-IP-blocklists/master/doh-ipv6.txt, rule: /^(([0-9A-f]{0,4}:){1,7}[0-9A-f]{0,4}:?(\/(1?[0-2][0-8]|[0-9][0-9]))?)([[:space:]]|$)/{print "add doh_6 "$1}, rc: 8, log: Downloading 'https://raw.githubusercontent.com/dibdot/DoH-IP-blocklists/master/doh-ipv6.txt' Connecting to :::443 HTTP error 404
user.debug banIP-0.7.10[14977]: f_down  ::: name: doh_4, url: https://raw.githubusercontent.com/dibdot/DoH-IP-blocklists/master/doh-ipv4.txt, rule: /^(([0-9]{1,3}\.){3}(1?[0-9][0-9]?|2[0-4][0-9]|25[0-5])(\/(1?[0-9]|2?[0-9]|3?[0-2]))?)([[:space:]]|$)/{print "add doh_4 "$1}, rc: 8, log: Downloading 'https://raw.githubusercontent.com/dibdot/DoH-IP-blocklists/master/doh-ipv4.txt' Connecting to :::443 HTTP error 404
f_down  ::: name: proxy_4, url: https://raw.githubusercontent.com/firehol/blocklist-ipsets/master/proxylists.ipset, rule: /^(([0-9]{1,3}\.){3}(1?[0-9][0-9]?|2[0-4][0-9]|25[0-5])(\/(1?[0-9]|2?[0-9]|3?[0-2]))?)([[:space:]]|$)/{print "add proxy_4 "$1}, rc: 8, log: Downloading 'https://raw.githubusercontent.com/firehol/blocklist-ipsets/master/proxylists.ipset' Connecting to :::443 HTTP error 404
```

stangri, 2021-11-08:
Tagging @dibdot

dibdot, 2021-11-08:
Worksforme - not reproducible. Most probably a tempory github outage. Nothing to fix at all.

closed 2021-11-08 by dibdot