openwrt/packages #17097: uw-imap: can't connect to IMAP server using SSL

dahanc, 2021-11-08:
Maintainer: @lucize 
Environment: x86, 64, 21.02.0

Description:

I am trying to use the asterisk-app-voicemail-imap package, which uses the uw-imap package for its IMAP support. However, while the voicemail module can connect to the server without SSL, if I enable SSL (by setting `imapflags=ssl` in Asterisk's `voicemail.conf`), Asterisk would report `app_voicemail_imap.c: IMAP Error: Can't open mailbox {192.168.1.7:993/imap/ssl/user=someuser}INBOX: invalid remote specification`, and tcpdump showed that it didn't even attempt to connect.

Looking through the uw-imap source, I found that the most likely cause of that error is that `ssl_onceonlyinit()` was never called. The uw-imap Makefile creates a `linkage.c` file containing initialization code, and code that uses the package is expected to `#include` that file. If SSL is enabled, one of the targets in the uw-imap Makefile appends a call to `ssl_onceonlyinit()` to `linkage.c`. However, when looking at the generated `linkage.c`, I saw that it didn't include that line.

I tracked the problem down to [this patch](https://github.com/openwrt/packages/blob/212f9fa76c7c79c6eb8897a1fe74cbed30a8254a/libs/uw-imap/patches/001-fix_Makefiles_and_shlib.patch#L120), in particular where it moves `ssl$(SSLTYPE)` to the beginning in `once: onceenv ckp$(PASSWDTYPE) ssl$(SSLTYPE) osdep.c`. The `onceenv` target calls the [drivers](https://github.com/uw-imap/imap/blob/master/src/osdep/unix/drivers) script, which first deletes `linkage.c`, then the `ssl$(SSLTYPE)` target will append the call to `ssl_onceonlyinit()` to `linkage.c`. However, with the order of the targets rearranged, `ssl$(SSLTYPE)` creates `linkage.c` with the call to `ssl_onceonlyinit()`, then `onceenv` deletes the file and replaces it with its own initialization, losing the call to `ssl_onceonlyinit()`.

That patch seems to come from this [Alpine Linux patch](https://git.alpinelinux.org/aports/commit/main/imap/fix-linking.patch?id=6c73adbe1eadb7caf44971bb4b12ec0e61e596f7), with the commit comment "we need to add the libs at the end otherwise will --as-needed remove the ssl libs and we end up with an underlinked c-client library." However, I don't get how reordering the once target would affect that, and that change clearly breaks the generation of a proper `linkage.c`.

I modified the patch to not change the `once` target, rebuilt uw-imap and asterisk-app-voicemail-imap, and Asterisk was able to connect to the IMAP server using SSL. I don't know how to tell if the library is "underlinked", but it works fine for me in this scenario.

Here's a patch for the patch:

```
diff --git a/libs/uw-imap/patches/001-fix_Makefiles_and_shlib.patch b/libs/uw-imap/patches/001-fix_Makefiles_and_shlib.patch
index db83f98d5..8f1dc0f20 100644
--- a/libs/uw-imap/patches/001-fix_Makefiles_and_shlib.patch
+++ b/libs/uw-imap/patches/001-fix_Makefiles_and_shlib.patch
@@ -115,12 +115,7 @@
  
  osdep.c: osdepbas.c osdepckp.c osdeplog.c osdepssl.c
  	$(CAT) osdepbas.c osdepckp.c osdeplog.c osdepssl.c > osdep.c
-@@ -962,21 +972,14 @@ os_sol.h:
- 
- # Once-only environment setup
- 
--once:	onceenv ckp$(PASSWDTYPE) ssl$(SSLTYPE) osdep.c
-+once:	ssl$(SSLTYPE) onceenv ckp$(PASSWDTYPE) osdep.c
+@@ -966,17 +976,10 @@ os_sol.h:
  
  onceenv:
  	@echo Once-only environment setup...

```

mhei, 2022-01-17:
Wow, nice problem analysis, very helpful documentation... I really, really appreciate your work.
This makes following your thoughts really easy, so I think your finding and conclusion is correct.
Next step would be to create a real PR with your suggested patch. I could try to do it during the next evenings and would also try to verify it using PHP's imap module - it should show the same issue since it also uses this library.

dahanc, 2022-09-09:
Hi, just checking in to see if I can help with anything on this.