openwrt/packages #17118: Dakstat : directory permission problem

Philippe-101, 2021-11-12:
Maintainer: @padre-lacroix
Environment: OpenWrt 21.02.0 r16279-5cc0535800 / LuCI openwrt-21.02 branch git-21.314.39479-e1ccb66

Description:

Installed Darkstat package with Luci software, run good. Revise config file and uncomment the 3 option for log file and want to move them on usb disk connected to router. Modify line as follow " /mnt/sda1/darkstat/darkstat_xxxxx.log. Create the directory manualy for that.   Start service, dakstat run OK but no file is created.

After investigation, found darkstat create  directory /var/darkstat as root:root and dont have permission to write these file.

To suit my need, i modify the /etcv/init.d/darkstat script as follow.

Change RUN_D parameter to my usb disk ( /mnt/sda1/dakstat )
add to : set_config_string chown $USER:$GROUP $RUN_D.
Remove modification in the config file.

After these mod, everything run as expected. 

Can we modify config file to add RUN_D parameter and it is ok to chown the directory or i miss something during installation.

Thanks

neheb, 2021-11-13:
ping @zhanhb

zhanhb, 2021-11-13:
ping @simm42 @padre-lacroix for I know nothing about darkstat.

simm42, 2021-11-13:
Sounds like /var/darkstst should have different permissions after the package install. It should be owned by the darkstst user and group. I’d support a patch to fix this in the Luci package. 

I don’t think there should be a chown in the unit script - that smacks of bad practice. 

If a user is creating a different path for log output other than the defaults then the onus should be on them to set the correct permissions. Howe we a comment giving that instruction in the config file at that line might be a good idea.

padre-lacroix, 2021-11-13:
Let me look at all that.

Currently, there is no darkstat user and group in OpenWRT: I guess that they would have to be created.  In the init file there are the setting for USER and GROUP that are set to nobody and nogroup, but they are not used in the init script (not sure why they are there: I guess they were there when I took this orphan package in 2015-2016).

We could set a few more options in the config file and use them in the init sript to allow more flexibility: if somebody wants to do it, the by all mean do it and make a pull request, and if it works, we can approve that.  I currently do not have much time to work on that currently.  Also, I do not have external hard disks in my routers to test that it would work correctly.

Philippe-101, 2021-11-13:
Thanks for your support.

If you need some helps for test, I could do it.