openwrt/packages #17145: nebula: no binaries were built with version 1.5.0

stangri, 2021-11-17:
@jefferyto @aparcar  can you please have a look at this? I'm trying to update the nebula [Makefile](https://github.com/stangri/source.openwrt.melmac.net/blob/90b5ba9972901edae1676630d64cafdbf5fe5100/nebula/Makefile) to build version 1.5.0 and I get the following when building with OpenWrt 21.02.1 x86_64 SDK from downloads:

```
touch -r /home/stangri/development/Builder/SDK_x86-64_21.02.1/build_dir/target-x86_64_musl/linux-x86_64/packages/.built /home/stangri/development/Builder/SDK_x86-64_21.02.1/build_dir/target-x86_64_musl/linux-x86_64/packages/.autoremove 2>/dev/null >/dev/null
find /home/stangri/development/Builder/SDK_x86-64_21.02.1/build_dir/target-x86_64_musl/linux-x86_64/packages -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' -and -not -name '.pkgdir' | xargs -r rm -rf
make[2]: [Makefile:63: compile] Error 1 (ignored)
make[2]: Leaving directory '/home/stangri/development/Builder/SDK_x86-64_21.02.1/package/kernel/linux'
time: package/kernel/linux/compile#1.33#0.11#1.41
make[2]: Entering directory '/home/stangri/development/Builder/SDK_x86-64_21.02.1/feeds/packages/stangri/nebula'
mkdir -p /home/stangri/development/Builder/SDK_x86-64_21.02.1/dl
SHELL= flock /home/stangri/development/Builder/SDK_x86-64_21.02.1/tmp/.nebula-1.5.0.tar.gz.flock -c '  	/home/stangri/development/Builder/SDK_x86-64_21.02.1/scripts/download.pl "/home/stangri/development/Builder/SDK_x86-64_21.02.1/dl" "nebula-1.5.0.tar.gz" "f67684a8eba6da91de3601afc97567fddd0e198973bba950fcf15cded92cdc50" "" "https://codeload.github.com/slackhq/nebula/tar.gz/v1.5.0?"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://codeload.github.com/slackhq/nebula/tar.gz/v1.5.0?/nebula-1.5.0.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 13210    0 13210    0     0  26158      0 --:--:-- --:--:-- --:--:-- 26106
100  920k    0  920k    0     0  1458k      0 --:--:-- --:--:-- --:--:-- 1456k
touch /home/stangri/development/Builder/SDK_x86-64_21.02.1/build_dir/target-x86_64_musl/nebula-1.5.0/.prepared_cae368d1af66a9b284e82c091f4e249d_18f1e190c5d53547fed41a3eaa76e9e9_check
. /home/stangri/development/Builder/SDK_x86-64_21.02.1/include/shell.sh; gzip -dc /home/stangri/development/Builder/SDK_x86-64_21.02.1/dl/nebula-1.5.0.tar.gz | tar -C /home/stangri/development/Builder/SDK_x86-64_21.02.1/build_dir/target-x86_64_musl/nebula-1.5.0/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/stangri/development/Builder/SDK_x86-64_21.02.1/build_dir/target-x86_64_musl/nebula-1.5.0
touch /home/stangri/development/Builder/SDK_x86-64_21.02.1/build_dir/target-x86_64_musl/nebula-1.5.0/.prepared_cae368d1af66a9b284e82c091f4e249d_18f1e190c5d53547fed41a3eaa76e9e9
rm -f /home/stangri/development/Builder/SDK_x86-64_21.02.1/build_dir/target-x86_64_musl/nebula-1.5.0/.configured_*
rm -f /home/stangri/development/Builder/SDK_x86-64_21.02.1/staging_dir/target-x86_64_musl/stamp/.nebula_installed
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/stangri/development/Builder/SDK_x86-64_21.02.1/tmp/go-build" GO_MOD_CACHE_DIR="/home/stangri/development/Builder/SDK_x86-64_21.02.1/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/slackhq/nebula" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/slackhq/nebula" GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/stangri/development/Builder/SDK_x86-64_21.02.1/build_dir/target-x86_64_musl/nebula-1.5.0" GO_BUILD_DIR="/home/stangri/development/Builder/SDK_x86-64_21.02.1/build_dir/target-x86_64_musl/nebula-1.5.0/.go_work/build" GO_BUILD_BIN_DIR="/home/stangri/development/Builder/SDK_x86-64_21.02.1/build_dir/target-x86_64_musl/nebula-1.5.0/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/stangri/development/Builder/SDK_x86-64_21.02.1/staging_dir/target-x86_64_musl/usr/share/gocode/src" /usr/bin/env bash ../../lang/golang//golang-build.sh configure
Copying files from /home/stangri/development/Builder/SDK_x86-64_21.02.1/build_dir/target-x86_64_musl/nebula-1.5.0 into /home/stangri/development/Builder/SDK_x86-64_21.02.1/build_dir/target-x86_64_musl/nebula-1.5.0/.go_work/build/src/github.com/slackhq/nebula
allow_list.go
allow_list_test.go
bits.go
bits_test.go
cert.go
cert/ca.go
cert/cert.go
cert/cert.pb.go
cert/cert.proto
cert/cert_test.go
cidr/parse.go
cidr/tree4.go
cidr/tree4_test.go
cidr/tree6.go
cidr/tree6_test.go
cmd/nebula-cert/ca.go
cmd/nebula-cert/ca_test.go
cmd/nebula-cert/keygen.go
cmd/nebula-cert/keygen_test.go
cmd/nebula-cert/main.go
cmd/nebula-cert/main_test.go
cmd/nebula-cert/print.go
cmd/nebula-cert/print_test.go
cmd/nebula-cert/sign.go
cmd/nebula-cert/sign_test.go
cmd/nebula-cert/test_darwin.go
cmd/nebula-cert/test_linux.go
cmd/nebula-cert/test_windows.go
cmd/nebula-cert/verify.go
cmd/nebula-cert/verify_test.go
cmd/nebula-service/logs_generic.go
cmd/nebula-service/logs_windows.go
cmd/nebula-service/main.go
cmd/nebula-service/service.go
cmd/nebula/main.go
config/config.go
config/config_test.go
connection_manager.go
connection_manager_test.go
connection_state.go
control.go
control_test.go
control_tester.go
dist/windows/wintun/include/wintun.h
dns_server.go
[... 318 lines cut ...]
golang.org/x/crypto/ssh/internal/bcrypt_pbkdf
golang.org/x/crypto/ssh
golang.org/x/crypto/ssh/terminal
github.com/vishvananda/netns
github.com/vishvananda/netlink/nl
github.com/slackhq/nebula/config
expvar
github.com/vishvananda/netlink
github.com/prometheus/common/expfmt
github.com/slackhq/nebula/udp
github.com/prometheus/client_golang/prometheus
github.com/slackhq/nebula/sshd
github.com/nbrownus/go-metrics-prometheus
github.com/prometheus/client_golang/prometheus/promhttp
github.com/slackhq/nebula

Error: No binaries were built
make[2]: *** [Makefile:130: /home/stangri/development/Builder/SDK_x86-64_21.02.1/build_dir/target-x86_64_musl/nebula-1.5.0/.built] Error 1
make[2]: Leaving directory '/home/stangri/development/Builder/SDK_x86-64_21.02.1/feeds/packages/stangri/nebula'
time: package/nebula/compile#57.26#6.27#18.94
    ERROR: package/nebula failed to build.
make[1]: *** [package/Makefile:114: package/nebula/compile] Error 1
make[1]: Leaving directory '/home/stangri/development/Builder/SDK_x86-64_21.02.1'
make: *** [/home/stangri/development/Builder/SDK_x86-64_21.02.1/include/toplevel.mk:223: package/nebula/compile] Error 2
```

jefferyto, 2021-11-18:
Try removing `GO_PKG_BUILD_PKG:=github.com/slackhq/nebula` from your makefile.

stangri, 2021-11-18:
> Try removing `GO_PKG_BUILD_PKG:=github.com/slackhq/nebula` from your makefile.

If I remove it, I get: `Building targets
no Go files in /home/stangri/development/Builder/SDK_x86-64_21.02.1/build_dir/target-x86_64_musl/nebula-1.5.0/.go_work/build` (contrary to the golang makefile comments).

jefferyto, 2021-11-18:
The [wintun](https://github.com/slackhq/nebula/tree/v1.5.0/wintun) package is causing `go list` (on Linux) to throw an error instead of listing packages to build. I'm not sure yet which component is ultimately responsible for this issue, but you can work around this by setting:

```make
GO_PKG_BUILD_PKG:= \
	github.com/slackhq/nebula/cmd/nebula \
	github.com/slackhq/nebula/cmd/nebula-cert
```

closed 2021-11-19 by stangri