openwrt/packages #17147: Dnsmasq 2.86-10 "cannot access directory"

zcy85611, 2021-11-18:
**Package-name: dnsmasq-full 2.86-10
Kernel Ver. : 5.4.158
Openwrt Ver: OpenWrt SNAPSHOT r18107-db34b93331 
Router Model: MT7621 gehua_ghl-r-001**

I compiled a new firmware yesterday, **I just added "conf-dir=/etc/dnsmasq.d" at the end of "/etc/dnsmasq.conf"**

then I restart dnsmasq, it gives me the follwoing err, it seems that the dnsmasq.d folder cannot be accessed

but I've been doing this for years, no problem at all. I've also tried chmod 777 to dnsmasq.d folder and files inside it

Still I get a "cannot access directory", the I tried to modify **"/init.d/dnsmasq"**

I changed config_get dnsmasqconfdir "$cfg" confdir **"/tmp/dnsmasq.d**" to config_get dnsmasqconfdir "$cfg" **confdir "/etc/dnsmasq.d"**

it can be started again, but it's not what I want exactly, I still want the settings in the /etc/dnsmasq.conf works

have anyone met this problem yet? any ideas on what's wrong with dnsmasq? thank you

```
Wed Nov 17 22:37:45 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Nov 17 22:37:45 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Nov 17 22:37:46 2021 daemon.crit dnsmasq[1]: cannot access directory /etc/dnsmasq.d: No such file or directory
Wed Nov 17 22:37:46 2021 daemon.crit dnsmasq[1]: FAILED to start up
Wed Nov 17 22:37:51 2021 daemon.crit dnsmasq[1]: cannot access directory /etc/dnsmasq.d: No such file or directory
Wed Nov 17 22:37:51 2021 daemon.crit dnsmasq[1]: FAILED to start up
Wed Nov 17 22:37:53 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Nov 17 22:37:54 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Nov 17 22:37:56 2021 daemon.crit dnsmasq[1]: cannot access directory /etc/dnsmasq.d: No such file or directory
Wed Nov 17 22:37:56 2021 daemon.crit dnsmasq[1]: FAILED to start up
Wed Nov 17 22:37:58 2021 daemon.crit dnsmasq[1]: cannot access directory /etc/dnsmasq.d: No such file or directory
Wed Nov 17 22:37:58 2021 daemon.crit dnsmasq[1]: FAILED to start up
Wed Nov 17 22:38:00 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Nov 17 22:38:00 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Nov 17 22:38:01 2021 daemon.crit dnsmasq[1]: cannot access directory /etc/dnsmasq.d: No such file or directory
Wed Nov 17 22:38:01 2021 daemon.crit dnsmasq[1]: FAILED to start up
Wed Nov 17 22:38:06 2021 daemon.crit dnsmasq[1]: cannot access directory /etc/dnsmasq.d: No such file or directory
Wed Nov 17 22:38:06 2021 daemon.crit dnsmasq[1]: FAILED to start up
Wed Nov 17 22:38:06 2021 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
```

dave14305, 2021-11-18:
Try adding `/etc/dnsmasq.d` to the init script where the jail mounts are defined.

https://github.com/openwrt/openwrt/blob/db34b93331e91bdb2cbc15d17632aaaab7217a6d/package/network/services/dnsmasq/files/dnsmasq.init#L1162

zcy85611, 2021-11-18:
i

> Try adding `/etc/dnsmasq.d` to the init script where the jail mounts are defined.
> 
> https://github.com/openwrt/openwrt/blob/db34b93331e91bdb2cbc15d17632aaaab7217a6d/package/network/services/dnsmasq/files/dnsmasq.init#L1162

thanks for the replied, I've checked the init script in my other router which has dnsmasq-full, 2.86-2 installed, I've compared the init scripts, it's almost the same, but in 2.86-10, the problem happens, is it because some changes happened in the new versions?

the auto generate config file for dnsmasq in **/var/etc/dnsmasq.conf.cfg01411c** is like the following, it has an entry for customized conf-file, but it actually fails to excute during startup, while the older dnsmasq can execute the customized conf-file

# auto-generated config file from /etc/config/dhcp
conf-file=/etc/dnsmasq.conf
dhcp-authoritative
domain-needed
localise-queries
read-ethers
enable-ubus=dnsmasq
expand-hosts
bind-dynamic
local-service
edns-packet-max=1232
domain=lan
local=/lan/
addn-hosts=/tmp/hosts
dhcp-leasefile=/tmp/dhcp.leases
resolv-file=/tmp/resolv.conf.d/resolv.conf.auto
stop-dns-rebind
rebind-localhost-ok
dhcp-broadcast=tag:needs-broadcast
conf-dir=/tmp/dnsmasq.d
user=dnsmasq
group=dnsmasq

dhcp-ignore-names=tag:dhcp_bogus_hostname
conf-file=/usr/share/dnsmasq/dhcpbogushostname.conf

bogus-priv
conf-file=/usr/share/dnsmasq/rfc6761.conf
dhcp-range=set:lan,192.168.2.100,192.168.2.249,255.255.255.0,12h

closed 2021-11-20 by zcy85611