openwrt/packages #17149: frr: compilation issue since latest update

feckert, 2021-11-18:
Maintainer: @lucize
Environment: lantiq_xrx200, OpenWrt latest master

Description:
Since latest update of the [frr package from 7.5.1 to  8.1.0](https://github.com/openwrt/packages/commit/e965229a7d4d96d9a268488c861b66535fba28b5#diff-4d54ebdae7e02d241e7e97ddfacd268286eb0abf78388c2047659f5c990a9ddd) the compilation of my mips_24kc target failed with the following trace. My x86_64 and x86_geode compiles this new version as expected. I do not know what the problem is! Any help would be great.

```
/bin/bash ./libtool  --tag=CC   --mode=link mips-openwrt-linux-musl-gcc -fno-omit-frame-pointer -funwind-tables -Wall -Wextra -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Wpointer-arith -Wbad-function-cast -Wwrite-strings -Wundef -Wno-unused-result -Wno-unused-parameter -Wno-missing-field-initializers  -I/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/staging_dir/target-mips_24kc_musl/usr/include       -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/build_dir/target-mips_24kc_musl/frr-c18235d5b1e33fad9d3181aa29fefa48b8c6ff3f=frr-c18235d5b1e33fad9d3181aa29fefa48b8c6ff3f -Wformat -Werror=format-security -DPIC -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto  -export-dynamic     -L/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/usr/lib -L/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/lib -DPIC -fpic -specs=/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/include/hardened-ld-pie.specs -znow -zrelro -Wl,--gc-sections,--as-needed -latomic  -o staticd/staticd staticd/static_main.o yang/frr-staticd.yang.o staticd/libstatic.a lib/libfrr.la -lcap -lresolv -lcrypt   -ljson-c -lrt
OpenWrt-libtool: link: mips-openwrt-linux-musl-gcc -fno-omit-frame-pointer -funwind-tables -Wall -Wextra -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Wpointer-arith -Wbad-function-cast -Wwrite-strings -Wundef -Wno-unused-result -Wno-unused-parameter -Wno-missing-field-initializers -I/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/staging_dir/target-mips_24kc_musl/usr/include -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/build_dir/target-mips_24kc_musl/frr-c18235d5b1e33fad9d3181aa29fefa48b8c6ff3f=frr-c18235d5b1e33fad9d3181aa29fefa48b8c6ff3f -Wformat -Werror=format-security -DPIC -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -flto -DPIC -fpic -specs=/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/include/hardened-ld-pie.specs -znow -zrelro -Wl,--gc-sections -Wl,--as-needed -o staticd/.libs/staticd staticd/static_main.o yang/frr-staticd.yang.o -Wl,--export-dynamic  -L/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/usr/lib -L/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/lib staticd/libstatic.a lib/.libs/libfrr.so -L/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/staging_dir/target-mips_24kc_musl/usr/lib /home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/lib/libatomic.so -lyang -lcap -lresolv -lcrypt -ljson-c -lrt -pthread -Wl,-rpath -Wl,/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/lib
/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/staging_dir/hostpkg/bin/clippy ./python/xrelfo.py   -o vtysh/vtysh.xref vtysh/vtysh
/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/staging_dir/hostpkg/bin/clippy ./python/xrelfo.py   -o zebra/zebra.xref zebra/zebra
while processing zebra/zebra:
Traceback (most recent call last):
  File "./python/xrelfo.py", line 377, in _main
    xrelfo.load_file(fn)
  File "./python/xrelfo.py", line 282, in load_file
    self.load_elf(filename, orig_filename)
  File "./python/xrelfo.py", line 324, in load_elf
    if ptr.xref is None:
  File "/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/build_dir/target-mips_24kc_musl/frr-c18235d5b1e33fad9d3181aa29fefa48b8c6ff3f/python/clippy/elf.py", line 351, in __getattr__
    self._fdata[attrname] = self._fdata[attrname]()
  File "/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/build_dir/target-mips_24kc_musl/frr-c18235d5b1e33fad9d3181aa29fefa48b8c6ff3f/python/clippy/elf.py", line 229, in __call__
    return self.cls(self.ptr)
  File "./python/xrelfo.py", line 70, in __init__
    super().__init__(*args, **kwargs)
  File "/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/build_dir/target-mips_24kc_musl/frr-c18235d5b1e33fad9d3181aa29fefa48b8c6ff3f/python/clippy/elf.py", line 313, in __init__
    data = dataptr.get_data(struct.calcsize(pspec))
  File "/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/build_dir/target-mips_24kc_musl/frr-c18235d5b1e33fad9d3181aa29fefa48b8c6ff3f/python/clippy/elf.py", line 119, in get_data
    return self._dstsect[self._dstoffs:self._dstoffs+reflen]
  File "/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/build_dir/target-mips_24kc_musl/frr-c18235d5b1e33fad9d3181aa29fefa48b8c6ff3f/python/clippy/elf.py", line 429, in __getitem__
    return self._obj[k]
_clippy.ELFAccessError: virtual address (4172550656) not found in program headers
make[4]: *** [Makefile:13079: zebra/zebra.xref] Error 1
make[4]: Leaving directory '/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/build_dir/target-mips_24kc_musl/frr-c18235d5b1e33fad9d3181aa29fefa48b8c6ff3f'
make[3]: *** [Makefile:5537: all] Error 2
make[3]: Leaving directory '/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/build_dir/target-mips_24kc_musl/frr-c18235d5b1e33fad9d3181aa29fefa48b8c6ff3f'
make[2]: *** [Makefile:247: /home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/build_dir/target-mips_24kc_musl/frr-c18235d5b1e33fad9d3181aa29fefa48b8c6ff3f/.built] Error 2
make[2]: Leaving directory '/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/feeds/packages/net/frr'
time: package/feeds/packages/frr/compile#148.22#10.66#153.74
    ERROR: package/feeds/packages/frr failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/frr/compile] Error 1
make[1]: Leaving directory '/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt'
make: *** [/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/include/toplevel.mk:230: package/feeds/packages/frr/compile] Error 2
```

BKPepe, 2021-11-18:
Cross-reference: https://github.com/FRRouting/frr/issues/10051

lucize, 2021-11-18:
sorry for this, I didn't catch it it time, despite testing on rampis it seems that is different than mips and also didn't had a mips board with enough flash

lucize, 2022-01-21:
@feckert this should be fixed now, seems that octeonplus and glibc still has issues

feckert, 2022-01-22:
@lucize Thanks for your work to fix this issue.
So the build with my target lantiq_xrx200 is fixed.
Should I close the ticket or should we still let them open for the `octeonplus`.
But I do not have this target.
If not we close this, we have to update the issue header.

PR #17879 (2022-02-17): frr: branch update, mips64 fix

Maintainer: me
build test mips64
Fixes: #17245 
Fixes: #17149
closed 2022-02-17 by neheb