openwrt/packages #17165: tvheadend: advice needed about codecs dependencies and licenses

M95D, 2021-11-20:
Hi. I'm working on updating the tvheadend package to their latest git master. It have troubles with package depends and licenses.

1) There are a few features that I want to make available for tvheadend as menuconfig options. Two of these features, DVB descrambling and H264 transcoding, depend on BUILD_PATENTED packages (libdvbcsa and libx264). I tried simply to create a menuconfig option that selects it's needed library, but that caused a circular dependency.
What is the proper way to add a package feature that depends on a BUILD_PATENTED package?

2) Why is libx264 BUILD_PATENTED? Isn't it free (GPL)?

3) There is no libx265 in OpenWrt, but tvheadend has the option to include a static libx265. Does this make tvheadend a BUILD_PATENTED package?

4) tvheadend can use ffmpeg libraries for transcoding. Does this make tvheadend BUILD_PATENTED too?

5) Should I split tvheadend into a tvheadend-free and a tvheadend-patented variants?

Thanks.
PS: I never thought I would need a lawyer when I started this tvheadend update... :-/

diizzyy, 2021-11-20:
1. Have a look at the ffmpeg package
2. h.264 compression algorithm isn't license free
3. About 99.9% of all devices that OpenWrt targets (ie embedded) doesn't have anything near the processing power to use x265 in a reasonable way and yes, you'll need BUILD_PATENTED
4. Indirectly if ffmpeg is built using said libraries but I don't think every package adheres to this (most likely by mistake)
5. Is it even usable on anything else than x86-64? If not, just submit one package

M95D, 2021-11-21:
1. I already looked at ffmpeg. It's a very complicated package. I didn't understand how/why it works without circular dependecies. I need a very simple example.
2. OK.

3, 4, 5:
I saw in ffmpeg that there are limitations such as no fdk-aac and x264 at the same time. Should I just ignore all that and add libffmpeg without checking what's inside? Is it even possible to check what's inside ffmpeg from another package?

I understand that OpenWrt is mainly for routers and router's can't usually transcode video fast enough, but is there any guide / policy on how to determine the level of functionality that should be included in a package?
I have no ideea how many users run OpenWrt on x86. What is "bloatware" for me might be "crippleware" for someone else. I prefer to let everyone choose with menuconfig options. That's why I'm trying to add options even for patented functionality.
Please understand - I'm a Gentoo user - I don't feel free unless I have tons of build options. :-))

Even if I exclude all transcoding stuff, there's still decrypting DVB which requires patented libdvbcsa. I don't want to drop that too, nor make the entire tvheadend a BUILD_PATENTED package.

diizzyy, 2021-11-21:
1. This might be a bit easier to read and follow, https://github.com/diizzyy/packages/blob/ffmpeg4-pkg/multimedia/ffmpeg4/Makefile although I do have a vague memory of seeing some kind dependency issue which is a limitation of the build framework. This is my one-size-fits-all approach until a moved away completely ;-)
3-5: Sounds pretty pointless if it makes the package essentially unusable? The "only" practical issue you run into with BUILD_PATENTED is that there wont be a precompiled package which you probably want anyway as many hardware capatibilities will be disabled due to the old target for the platform (you'd also need to patch the x86_64 target).

Unless you want to shoehorn this just because for the sake of it you're better off looking at another distro/OS to accomplish this and it'll save to time in the instead of trying to get reasonable optimization out of multimedia-related ports. Why not just run Gentoo or whatever you prefer? I personally found it much more time efficient both administrative and in terms of performance to  move pretty much everything that isn't wireless related elsewhere where you have all this already available and tuned.

M95D, 2021-11-22:
I didn't test, but I think your ffmpeg4 causes circular dependencies too. That's how I wrote the dependencies in tvheadend and it didn't work for me. In the mean time, after much trial-and-error, I have found a (or "the") solution that works:
```
DEPENDS:= +BUILD_PATENTED&&TVHEADEND_CSA:libdvbcsa
(and check CONFIG_BUILD_PATENTED before adding config args)
```

About the licenses:
I decided to make libffmpeg dependency only available with BUILD_PATENTED, so that I won't have to check if ffmpeg contains patented code or not.

Thanks. I'm closing this issue.

closed 2021-11-22 by M95D