openwrt/packages #17172: python: Concurrency issue between Meson and host Python

jefferyto, 2021-11-21:
Maintainer: me, @commodo (for Python), also ping @dhewg, @neheb (for Meson)
Environment: 2021-11-14 snapshot sdk

Description:
(I'm not sure which component is ultimately responsible for this issue or where this report should actually be filed, but I thought I would open an issue here first.)

While test compiling python-pycparser (for #17154), I encountered an error that I could reproduce a few times. If I run `make package/python-pycparser/compile -j3 V=s` with a newly-extracted sdk (in this case armvirt-32), I would see this:

```
...
Packaged contents of /media/jeff/Jekyll/Downloads/openwrt/testing/a32/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armvirt_32/packages/ipkg-arm_cortex-a15_neon-vfpv4/kmod-usb-serial-qualcomm into /media/jeff/Jekyll/Downloads/openwrt/testing/a32/bin/targets/armvirt/32/packages/kmod-usb-serial-qualcomm_5.10.79-1_arm_cortex-a15_neon-vfpv4.ipk
(cd /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/share/man/man1; ln -s python3.9.1 python3.1)
gcc -pthread -L/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/host/lib -L/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib -L/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/lib -Wl,-rpath,/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt -L/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/host/lib -L/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib -L/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/lib -Wl,-rpath,/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt   -Xlinker -export-dynamic -o Programs/_testembed Programs/_testembed.o libpython3.9.a -lcrypt -lpthread -ldl  -lutil -lm -lz   -lm
rstrip.sh: /media/jeff/Jekyll/Downloads/openwrt/testing/a32/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armvirt_32/packages/ipkg-arm_cortex-a15_neon-vfpv4/kmod-usb-storage/lib/modules/5.10.79/usb-storage.ko: relocatable
Packaged contents of /media/jeff/Jekyll/Downloads/openwrt/testing/a32/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armvirt_32/packages/ipkg-arm_cortex-a15_neon-vfpv4/kmod-usb-storage into /media/jeff/Jekyll/Downloads/openwrt/testing/a32/bin/targets/armvirt/32/packages/kmod-usb-storage_5.10.79-1_arm_cortex-a15_neon-vfpv4.ipk
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/tkinter
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/tkinter/test
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/tkinter/test/test_tkinter
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/tkinter/test/test_ttk
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/site-packages
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/audiodata
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/capath
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/data
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/cjkencodings
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/decimaltestdata
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/xmltestdata
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/xmltestdata/c14n-20
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/dtracedata
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/eintrdata
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/imghdrdata
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/libregrtest
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/subprocessdata
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/sndhdrdata
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/support
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/tracedmodules
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/encoded_modules
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/test_import
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/test_import/data
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/test_import/data/circular_imports
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/test_import/data/circular_imports/subpkg
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/test_import/data/package
[101/464] Generating symbol file glib/libglib-2.0.so.0.7000.0.p/libglib-2.0.so.0.7000.0.symbolsCreating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/test_import/data/package2
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/test_import/data/unwritable
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/importlib
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/test_importlib
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/test_importlib/builtin
Creating directory /media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/test/test_importlib/data
[102/464] Generating symbol file glib/libglib-2.0.so.0.7000.0.p/libglib-2.0.so.0.7000.0.symbols
FAILED: glib/libglib-2.0.so.0.7000.0.p/libglib-2.0.so.0.7000.0.symbols
/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/host/bin/meson --internal symbolextractor /media/jeff/Jekyll/Downloads/openwrt/testing/a32/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/glib-2.70.0/openwrt-build glib/libglib-2.0.so.0.7000.0 glib/libglib-2.0.so.0.7000.0 glib/libglib-2.0.so.0.7000.0.p/libglib-2.0.so.0.7000.0.symbols --cross-host=linux
Could not find platform independent libraries <prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = (not set)
  program name = 'python3'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = '/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/bin/python3'
  sys.base_prefix = '/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg'
  sys.base_exec_prefix = '/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg'
  sys.platlibdir = 'lib'
  sys.executable = '/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/bin/python3'
  sys.prefix = '/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg'
  sys.exec_prefix = '/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg'
  sys.path = [
    '/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python39.zip',
[... 59 lines cut ...]
      '/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python39.zip',
      '/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9',
      '/media/jeff/Jekyll/Downloads/openwrt/testing/a32/staging_dir/hostpkg/lib/python3.9/lib-dynload',
    ]
[... cut ...]
If I restart the build (using the same command) or start again (with a new sdk instance) but without `-j3`, then the build will succeed.

I think what is happening is, at some point during the host Python install (when the main Python binary has been installed but not all of the standard library), Meson (or Ninja) detects that host Python is "available" and tries to use it (instead of system Python). This fails as host Python isn't installed completely yet.

Does this sound correct? What can/should be done to avoid this situation?

dhewg, 2021-11-21:
`symbolextractor` is a meson python script. It seems that meson itself is using the python host package here to run that script, while we're explicitly running meson using the build host python (see #4659).

So it seems that meson is running python scripts with a different interpreter than itself is running under (likely because the "wrong" one is now found in the path)? That'll be a problem, but I'm not sure why that is happening. Maybe it's related to meson's zipapp deployment (see #4635) and @eli-schwartz has an idea?

eli-schwartz, 2021-11-21:
The build command definition is defined here:

https://github.com/mesonbuild/meson/blob/680c6bb012c4566935d2589565ef4d2031442307/mesonbuild/backend/ninjabackend.py#L2006

It is supposed to run `meson --internal symbolextractor` but with the meson command defined in some way that's supposed to reproduce how you initially ran meson. When invoked as an executable called `meson` it will be the full path to the executable, when invoked as a .py script it assumes it is running in-tree fr a git checkout and uses `sys.executable` followed by the script path.

Is the meson zipapp deployment pointing to the correct python in its shebang, or is it using `/usr/bin/env python`?

neheb, 2021-11-21:
The latter.

eli-schwartz, 2021-11-21:
Well, I recommended the former in my initial review comments back when zipapp support was first implemented. What happened?

What do you do for other host programs that happen to be implemented in python?

dhewg, 2021-11-21:
It's using the absolute path of the build host python interpreter with the full path to meson as argv[1]: https://github.com/openwrt/openwrt/blob/e9610794fd978175702b715a17f7925915fd1479/include/meson.mk#L59
And that's because of https://github.com/openwrt/openwrt/pull/4659

But with that we have meson python interpreter != symbolextractor python interpreter. It would work with its own argv[0] I guess.

eli-schwartz, 2021-11-21:
I had originally mentioned: 
> This should work flawlessly unless those Make variables get changed to point elsewhere, but my assumption is you'd probably want to recreate your environment in that case.

Apparently that assumption didn't hold true?

> What do you do for other "build host" programs that happen to be implemented in python?

It's still not clear to me what you do here.

> But with that we have meson python interpreter != symbolextractor python interpreter. It would work with its own argv[0] I guess.

`argv[0]` in python is always the name of the file being executed, it is rewritten for "convenience" so that doing `python foo.py` has the UI of running a program called `foo.py` without "leaking python internals". There is `sys.orig_argv`, originally requested in 2015 and finally added in python 3.10, which handles the difference between `python foo/main.py` and `python -m foo.main` but unfortunately doesn't distinguish whether the original command actually invoked the path to python.

An actual working implementation which is unfortunately linux-specific is:
```python
import shlex

with open(f'/proc/{os.getpid()}/cmdline', 'r') as f:
    cmdline = f.read()
print(shlex.join(cmdline.rstrip('\0').split('\0')))
```

Another option is always adding sys.executable, so that when running `/usr/bin/meson setup builddir/` the build.ninja is generated with commands that call into `/usr/bin/python3 /usr/bin/meson` which seems... extraneous? This is ostensibly why shebangs exist.

Another option which is a workaround for openwrt, would be to install the zipapp as `$(STAGING_DIR_HOST)/bin/meson.py` in order to trick meson's heuristics into thinking it needs to run using the "running from git repo" logic, which takes special care to preserve the original python sys.executable because there is no expectation that the shebang contains the correct path.

dhewg, 2021-11-21:
> Apparently that assumption didn't hold true?

Apparently yes ;) Because of the OpenWrt SDK, where paths may be different. That fact wasn't considered, or, well... forgotten, back then (not everyone is using that though, myself included)

> It's still not clear to me what you do here.

Most stuff is about python packages for the target, and all that is using the python host package anyway. Only a few a in the same boat as meson wrt build host interpreter, but I guess meson is the most complex of those and maybe even the single one spawning additional python scripts

> Another option which is a workaround for openwrt, would be to install the zipapp as $(STAGING_DIR_HOST)/bin/meson.py in order to trick meson's heuristics into thinking it needs to run using the [...]

That's very subtle, but may indeed be an option, thanks!
@jefferyto complete untested, but should be as simple as [meson.txt](https://github.com/openwrt/packages/files/7576846/meson.txt). Does that fix it?

jefferyto, 2021-11-22:
> @jefferyto complete untested, but should be as simple as meson.txt. Does that fix it?

That seems to work - I've done a few test builds and the error did not occur with this fix.

dhewg, 2021-11-22:
Nice, thanks everybody!
FYI the fix is already included in https://github.com/openwrt/openwrt/pull/4789

jefferyto, 2021-12-17:
The fix was merged in https://github.com/openwrt/openwrt/commit/75093d1b1de7af82acd3b8bcc61f308c37493400 - closing.

closed 2021-12-17 by jefferyto