openwrt/packages #17192: Softethervpn 5.02.5180 doesn't work - how to downgrade to 5.01.9674

desigabri, 2021-11-24:
MAINTAINER:=Andy Walsh <andy.walsh44+github@gmail.com>

Environment:
ipq806x
TP-Link Archer c2600 
master , openwrt-21.02

I can get working only old package version v5.01.9674
compiling the newer v5.02.5180, firmware loose configuration, and VPN interfaces couldn't work.

If I reconfigure them from scratch they seem working but after some time the interfaces stops and show error so I loose the vpn connections

I compiled some firmwares using older package, manually putting it into /package/network/ directory. 
Starting from some time until now I can't finish compiling firmware getting 
""ninja: Entering directory `/home/gabriele/openwrt/mastermod/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eab/SoftEtherVPN-5.01.9674'
ninja: error: loading 'build.ninja': No such file or directory""

Selecting actual version the compiling ends right (but doesn't work).

HOW CAN I SELECT THE OLDER (PREVIOUS) SOFTETHERVPN5 VERSION FOR COMPILING?  THANKS

Andy2244, 2021-11-26:
Just edit your local makefile to the older version:
https://github.com/openwrt/packages/commit/9c756cf618d501baaf309e1054024a546d123bb1#diff-a8968a252cd93a55e975e1d5dffa594fdf5e4a2c210bdfb63fa3fb653fc168b2

desigabri, 2021-11-26:
I think it's not sufficient: the package has differents patches and files. I tried changing the package but It doesn't work, so I tried adding the package into another places. It worked for older revision, I have to check which  ones, but it worked for older master. At the moment, for my esperience, master doesn't accept older softethervpn package as for right position so for alternative positions.

desigabri, 2021-12-19:
-- Build files have been written to: /home/gabriele/openwrt/mastermod/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/SoftEtherVPN-5.01.9674
touch /home/gabriele/openwrt/mastermod/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/SoftEtherVPN-5.01.9674/.configured_a17fb5ef857664f03cd0ce37cc5ea591
rm -f /home/gabriele/openwrt/mastermod/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/SoftEtherVPN-5.01.9674/.built
touch /home/gabriele/openwrt/mastermod/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/SoftEtherVPN-5.01.9674/.built_check
MAKEFLAGS="" /home/gabriele/openwrt/mastermod/staging_dir/host/bin/ninja -v -j1 -C /home/gabriele/openwrt/mastermod/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/SoftEtherVPN-5.01.9674 vpnserver vpnbridge vpnclient vpncmd hamcore-archive-build
ninja: Entering directory `/home/gabriele/openwrt/mastermod/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/SoftEtherVPN-5.01.9674'
ninja: error: loading 'build.ninja': No such file or directory
make[3]: *** [Makefile:169: /home/gabriele/openwrt/mastermod/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/SoftEtherVPN-5.01.9674/.built] Error 1
make[3]: Leaving directory '/home/gabriele/openwrt/mastermod/package/network/services/softethervpn5'
time: package/network/services/softethervpn5/compile#6.59#6.09#11.88
    ERROR: package/network/services/softethervpn5 failed to build.
make[2]: *** [package/Makefile:116: package/network/services/softethervpn5/compile] Error 1
make[2]: Leaving directory '/home/gabriele/openwrt/mastermod'
make[1]: *** [package/Makefile:110: /home/gabriele/openwrt/mastermod/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gabriele/openwrt/mastermod'
make: *** [/home/gabriele/openwrt/mastermod/include/toplevel.mk:230: world] Error 2

closed 2025-11-15 by GeorgeSapkin