openwrt/packages #17200: unbound: lan contain nothing?

youxiaojie, 2021-11-25:
Maintainer: Eric Luehrsen <ericluehrsen@gmail.com>
Environment: 21.02.1

Description:
my lan shows contain nothing, correct?
lan (no interface attached)

[image]

EricLuehrsen, 2021-12-06:
That is the standard network widget in LuCI. What appears there is dependent on your network configuration (i.e. /etc/config/network). This not technically part of Unbound or luci-app-unbound.

youxiaojie, 2021-12-15:
but my /etc/config/network define lan lan-br, but nothing here.

luisimasd, 2023-02-08:
Same issue here, any solution?

luisimasd, 2023-02-08:
something is not ok, I added a new interface to the system, as "LAN2", unmanaged, just to try.
Now all the interfaces under unbound are noted as "no interface attached"