openwrt/packages #17208: build failed on package minicom

ptpt52, 2021-11-26:
```
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether mips-openwrt-linux-musl-gcc accepts -g... yes
checking for mips-openwrt-linux-musl-gcc option to accept ISO C89... none needed
checking whether mips-openwrt-linux-musl-gcc understands -c and -o together... yes
checking dependency style of mips-openwrt-linux-musl-gcc... gcc3
checking how to run the C preprocessor... mips-openwrt-linux-musl-gcc -E
checking for grep that handles long lines and -e... /home/runner/work/build-release/build-release/x-wrt/staging_dir/host/bin/grep
checking for egrep... /home/runner/work/build-release/build-release/x-wrt/staging_dir/host/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking for mips-openwrt-linux-gcc... (cached) mips-openwrt-linux-musl-gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether mips-openwrt-linux-musl-gcc accepts -g... (cached) yes
checking for mips-openwrt-linux-musl-gcc option to accept ISO C89... (cached) none needed
checking whether mips-openwrt-linux-musl-gcc understands -c and -o together... (cached) yes
checking dependency style of mips-openwrt-linux-musl-gcc... (cached) gcc3
checking for mips-openwrt-linux-ar... mips-openwrt-linux-musl-gcc-ar
checking the archiver (mips-openwrt-linux-musl-gcc-ar) interface... ar
checking for mips-openwrt-linux-ranlib... mips-openwrt-linux-musl-gcc-ranlib
checking for library containing strerror... none required
checking build system type... x86_64-pc-linux-gnu
checking host system type... mips-openwrt-linux-gnu
checking for ld used by GCC... mips-openwrt-linux-musl-ld
checking if the linker (mips-openwrt-linux-musl-ld) is GNU ld... yes
checking for shared library run path origin... done
checking for iconv... yes
checking for working iconv... guessing yes
checking how to link with libiconv... -liconv
checking pkg-config is at least version 0.9.0... yes
checking for LOCKDEV... no
./configure: line 6725: syntax error near unexpected token `external'
./configure: line 6725: `AM_GNU_GETTEXT(external)'
Makefile:47: recipe for target '/home/runner/work/build-release/build-release/x-wrt/build_dir/target-mips_24kc_musl/minicom-2.8/.configured_a17fb5ef857664f03cd0ce37cc5ea591' failed
make[3]: *** [/home/runner/work/build-release/build-release/x-wrt/build_dir/target-mips_24kc_musl/minicom-2.8/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 2
make[3]: Leaving directory '/home/runner/work/build-release/build-release/x-wrt/feeds/packages/utils/minicom'
time: package/feeds/packages/minicom/compile#3.60#0.48#5.89
    ERROR: package/feeds/packages/minicom failed to build.
package/Makefile:114: recipe for target 'package/feeds/packages/minicom/compile' failed
make[2]: *** [package/feeds/packages/minicom/compile] Error 1
make[2]: Leaving directory '/home/runner/work/build-release/build-release/x-wrt'
package/Makefile:110: recipe for target '/home/runner/work/build-release/build-release/x-wrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/runner/work/build-release/build-release/x-wrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/build-release/build-release/x-wrt'
/home/runner/work/build-release/build-release/x-wrt/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

```

ptpt52, 2021-11-26:
ping @nbd168  any help on this `minicom` build error

neheb, 2021-11-26:
does make package/gettext-full/host/compile fix the issue?

feckert, 2021-11-26:
I noticed that in my build too. I don't know what triggered it. The following change fixed it in my build. The targets are x86_64 and lantiq_xrx200. I wanted to evaluate this and would have created a pullrequest. 

```
From: Florian Eckert <fe@dev.tdt.de>
Date: Wed, 24 Nov 2021 13:08:09 +0100
Subject: [PATCH] minicom: remove PKG_FIXUP:autoreconf

Redmine-patch-id: 6608
The definition PKG_FIXUP=:autorenconf is no longer necessary. If this
option is not removed, I get the following compilation error warning.

configure.ac:125: warning: macro 'AM_GNU_GETTEXT' not found in library
configure.ac:126: warning: macro 'AM_GNU_GETTEXT_VERSION' not found in library

And the compilation stops with the following message.

./configure: line 6690: syntax error near unexpected token `external'
./configure: line 6690: `AM_GNU_GETTEXT(external)'

Removeing the PKG_FIXUP solves this.

---
 utils/minicom/Makefile | 2 --
 1 file changed, 2 deletions(-)

diff --git a/utils/minicom/Makefile b/utils/minicom/Makefile
index bc40b4e05edb949bbd25c37b4f2857086b0d58ca..4972d4a02a950c995d33b2f5284a87a5cad0b9c1 100644
--- a/utils/minicom/Makefile
+++ b/utils/minicom/Makefile
@@ -20,8 +20,6 @@ PKG_LICENSE:=GPL-2.0-or-later
 PKG_LICENSE_FILES:=COPYING
 PKG_CPE_ID:=cpe:/a:minicom:minicom
 
-PKG_FIXUP:=autoreconf
-
 include $(INCLUDE_DIR)/nls.mk
 include $(INCLUDE_DIR)/package.mk
 
-- 

```

Could you confirm that this fixes your builds?

cotequeiroz, 2021-11-26:
Before you test @feckert's patch, or use @neheb's suggestion, post the results of getting the first line of each staging-dir-installed `gettext.m4`, when you encounter the build error: 
```
find staging_dir -name gettext.m4 -exec head -n1 {} +
```
Let's see if a package installed an older copy of the file along the way.

I have `# gettext.m4 serial 71 (gettext-0.20.2)` in both `staging_dir/{hostpkg,target*}` and minicom builds fine.

ptpt52, 2021-11-26:
@cotequeiroz 
I try `find staging_dir -name gettext.m4 -exec head -n1 {} +` and get nothing

you could reproduce it via :
```
rm -rf staging_dir build_dir
# and build it again
```

cotequeiroz, 2021-11-26:
I build it fine here, though.  What's the output of `./scripts/diffconfig.sh`?

ptpt52, 2021-11-26:
> find staging_dir -name gettext.m4 -exec head -n1 {} +

`makes package/libs/gettext-full/host/compile`
and then rebuild fix the issue?
what to do?

feckert, 2021-11-26:
The question is why do we have a `PKG_FIXUP`?
Maybe that is no longer necessary

ptpt52, 2021-11-26:
maybe?
```
diff --git a/utils/minicom/Makefile b/utils/minicom/Makefile
index bc40b4e05..fd945cd04 100644
--- a/utils/minicom/Makefile
+++ b/utils/minicom/Makefile
@@ -20,6 +20,7 @@ PKG_LICENSE:=GPL-2.0-or-later
 PKG_LICENSE_FILES:=COPYING
 PKG_CPE_ID:=cpe:/a:minicom:minicom
 
+PKG_BUILD_DEPENDS:=gettext-full/host
 PKG_FIXUP:=autoreconf
 
 include $(INCLUDE_DIR)/nls.mk

```

cotequeiroz, 2021-11-26:
> > find staging_dir -name gettext.m4 -exec head -n1 {} +
> 
> `makes package/libs/gettext-full/host/compile` and then rebuild fix the issue? what to do?

Try to build the package.  It will fail.  Don't remove anything, and try to run my command again (I added a -v so it will always print the filename) paste the results if you got any:
`find staging_dir -name gettext.m4 -exec head -vn1 {} +`
Now, try @neheb's fix:
`make package/libs/gettext-full/host/compile`, and then try to build just the package (don't remove anything else):
`make package/minicom/{clean,compile}`

ptpt52, 2021-11-26:
There is not gettext.m4 in stage_dir if you do not run the `make package/libs/gettext-full/host/compile`

This may be the point.

cotequeiroz, 2021-11-26:
Then either one of the fixes will work.  I'd go with removing `PKG_FIXUP:=autoreconf`.

ptpt52, 2021-11-26:
> maybe?
>
> ```
> diff --git a/utils/minicom/Makefile b/utils/minicom/Makefile
> index bc40b4e05..fd945cd04 100644
> --- a/utils/minicom/Makefile
> +++ b/utils/minicom/Makefile
> @@ -20,6 +20,7 @@ [...]

likely this change would fix? wait, I am doing an clean build

ptpt52, 2021-11-26:
notice simlilar to `multimedia/gstreamer1/Makefile`
```
PKG_BUILD_DEPENDS:=gettext-full/host
```

cotequeiroz, 2021-11-26:
Either one of the proposed changes will fix it.
- If you don't rebuild `configure`, then you won't need `gettext.m4`, and the package will build faster.  The fixup was already present in 2014 when the package was imported to the repo, so it was probably added earlier than that.  It was updated twice since, and perhaps the fix is no longer needed.

- If you need or want to rebuild `configure`, then you must ensure `gettext/host` has already been built.

ptpt52, 2021-11-26:
I prefer PKG_BUILD_DEPENDS

feckert, 2021-11-26:
Maybe we should add the two amendments.

> Then either one of the fixes will work. I'd go with removing `PKG_FIXUP:=autoreconf`.

I don't think PKG_FIXUP is necessary anymore. However, I am not the maintainer.

The main thing is that it builds and works again :-)

cotequeiroz, 2021-11-26:
> Maybe we should add the two amendments.

That's about the only thing we should not do.  We'll increase the build time (of the package by itself, not if gettext-full/host is already present), while not using anything from the gettext that was force-built, and still risking incompatibilities by using the shipped `configure`.

Here's some numbers about the build time:
#### `+PKG_BUILD_DEPENDS:=gettext-full/host`
```
$ make -j12 package/minicom/clean
 make[1] package/minicom/clean
 make[2] -C feeds/packages/utils/minicom clean
$ package/minicom/compile
 make[2] -C package/libs/gettext-full host-compile
 make[2] -C package/libs/ncurses host-compile
 make[2] -C package/libs/toolchain clean-build
 make[2] -C package/libs/toolchain compile
 make[2] -C package/libs/zlib clean-build
 make[2] -C package/libs/zlib compile
 make[2] -C package/libs/libiconv clean-build
 make[2] -C package/libs/libiconv compile
 make[2] -C package/libs/ncurses clean-build
 make[2] -C package/libs/ncurses compile
 make[2] -C feeds/packages/utils/minicom clean-build
 make[2] -C feeds/packages/utils/minicom compile

real   	3m8.000s
user   	3m51.924s
sys    	0m23.341s

```

#### `-PKG_FIXUP:=autoreconf`
```
$ make package/minicom/clean
Collecting package info: done
 make[1] package/minicom/clean
 make[2] -C feeds/packages/utils/minicom clean
time make -j12 package/minicom/compile
time: target/linux/prereq#0.61#0.06#0.66
 make[1] package/minicom/compile
 make[2] -C package/libs/ncurses host-compile
 make[2] -C package/libs/toolchain clean-build
 make[2] -C package/libs/toolchain compile
 make[2] -C package/libs/zlib clean-build
 make[2] -C package/libs/zlib compile
 make[2] -C package/libs/libiconv clean-build
 make[2] -C package/libs/libiconv compile
 make[2] -C package/libs/ncurses clean-build
 make[2] -C package/libs/ncurses compile
 make[2] -C feeds/packages/utils/minicom clean-build
 make[2] -C feeds/packages/utils/minicom compile

real   	0m53.733s
user   	1m20.728s
sys    	0m9.022s
```

The biggest gain is if you don't need to build `gettext-full/host`.  If you already have to build it because of something else, then the difference is just ~3.8s.

feckert, 2021-12-02:
How do we continue here now
I'm not the package manager, so I can't say which is better or worse.

The PKG_FIXUP was simply taken over during the import and there has only been one package bump so far.
The question is do we still need the PKG_FIXUP?

@ptpt52 Could you please explain why we should use `PKG_BUILD_DEPENDS:=gettext-full/host`
If I were to add the package now, I would also start without PKG_FIXUP for the time being, wouldn't I?

I can live with both suggestions.
The main thing is that the package builds again and works.

ptpt52, 2021-12-02:
okay, if all works, i have no preference.

I thought `PKG_BUILD_DEPENDS:=gettext-full/host` and `PKG_FIXUP ` would be more general

PR #17262 (2021-12-06): minicom: remove PKG_FIXUP:autoreconf
Maintainer: @neheb 
Compile tested: x86_64 APU3 and lantiq_xrx200, latest openwrt master
Run tested: x86_64 APU3 and lantiq_xrx200, latest openwrt master, start minicom

Description:
```
root@st-dev-07 ~ # minicom -v
minicom version 2.8
Copyright (C) Miquel van Smoorenburg.

This program is free software; you can redistribute it and/or
modify it under the terms of the GNU General Public License
as published by the Free Software Foundation; either version
2 of the License, or (at your option) any later version.
```

The definition PKG_FIXUP=:autorenconf is n [... cut]
closed 2021-12-06 by feckert