openwrt/packages #17211: domoticz: fails to compile on arm_cortex-a15_neon-vfpv4

PolynomialDivision, 2021-11-26:
Maintainer: @stintel 

```

Applying ./patches/012-minizip-overflow.patch using plaintext: 
patching file main/unzip_stream.h
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc
-- Check for working C compiler: /builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc - broken
CMake Error at /builder/shared-workdir/build/sdk/staging_dir/host/share/cmake-3.21/Modules/CMakeTestCCompiler.cmake:69 (message):
  The C compiler

    "/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/domoticz-2021.1/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/builder/shared-workdir/build/sdk/staging_dir/host/bin/ninja cmTC_42f89 && [1/2] Building C object CMakeFiles/cmTC_42f89.dir/testCCompiler.c.o
    [2/2] Linking C executable cmTC_42f89
    FAILED: cmTC_42f89 
    : && /builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/domoticz-2021.1=domoticz-2021.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -L/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/lib -L/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/lib -znow -zrelro -lpython3.9 CMakeFiles/cmTC_42f89.dir/testCCompiler.c.o -o cmTC_42f89   && :
    /builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/bin/../lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: cannot find -lpython3.9
    collect2: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!
See also "/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/domoticz-2021.1/CMakeFiles/CMakeOutput.log".
See also "/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/domoticz-2021.1/CMakeFiles/CMakeError.log".
make[3]: *** [Makefile:134: /builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/domoticz-2021.1/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
time: package/feeds/packages/domoticz/compile#1.36#0.54#2.39

```

https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a15_neon-vfpv4/packages/domoticz/compile.txt

stintel, 2021-11-26:
I stopped maintaining and using Domoticz long ago.

PolynomialDivision, 2021-11-26:
Okay, @neheb did the last changes. ;)

PolynomialDivision, 2021-11-26:
https://github.com/openwrt/packages/pull/17205

closed 2021-12-12 by BKPepe