openwrt/packages #17215: php7 Makefile missing libsasl2

manageryzy, 2021-11-26:
Maintainer: Michael Heimpold <mhei@heimpold.de>
Environment: x86_64 

Description:

if `CONFIG_PACKAGE_php7-mod-ldap` is selected,  while executing `$(eval $(call BuildPackage,php7))`,  at `Build/CoreTargets`  no libsasl2 leads to `./configure` fails with
```
checking for ldap_refresh_s... yes
checking for libsasl2... no
configure: error: Package requirements (libsasl2) were not met:

No package 'libsasl2' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables SASL_CFLAGS
and SASL_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

```

mhei, 2021-11-28:
Could you please provide more details? I tried but I cannot reproduce, so I guess I'm doing other steps as you do. Please tell us which distribution you use (host system), which release you try (snapshot aka master branch, or one of the release branches...). Does this problem still occurs when doing a "make dirclean" before you build?

manageryzy, 2021-12-03:
I found this issue when I was built it on a forked OpenWrt. I was using [https://github.com/Lienol/openwrt/tree/19.07](https://github.com/Lienol/openwrt/tree/19.07). The main difference between this fork with mainline is Lienol is using Lede 18

mhei, 2021-12-03:
Thanks for your feedback. So if I understand correctly, we can close this here.

closed 2021-12-03 by mhei