openwrt/packages #17217: python3: build failed for mipsel_24kc_24kf snapshots

MoetaYuko, 2021-11-27:
Maintainer: @jefferyto
Environment: https://downloads.openwrt.org/snapshots/targets/pistachio/generic/openwrt-sdk-pistachio_gcc-11.2.0_musl.Linux-x86_64.tar.xz

Description:

```
2021-11-26T23:34:02.4940943Z mipsel-openwrt-linux-musl-gcc -L/tmp/tmp.56YaDbl1Fv/staging_dir/toolchain-mipsel_24kc+24kf_gcc-11.2.0_musl/usr/lib -L/tmp/tmp.56YaDbl1Fv/staging_dir/toolchain-mipsel_24kc+24kf_gcc-11.2.0_musl/lib -znow -zrelro -L/tmp/tmp.56YaDbl1Fv/build_dir/target-mipsel_24kc+24kf_musl/Python-3.10.0 -L/tmp/tmp.56YaDbl1Fv/staging_dir/target-mipsel_24kc+24kf_musl/usr/lib -L/tmp/tmp.56YaDbl1Fv/staging_dir/target-mipsel_24kc+24kf_musl/lib -L/tmp/tmp.56YaDbl1Fv/staging_dir/toolchain-mipsel_24kc+24kf_gcc-11.2.0_musl/usr/lib -L/tmp/tmp.56YaDbl1Fv/staging_dir/toolchain-mipsel_24kc+24kf_gcc-11.2.0_musl/lib -znow -zrelro -L/tmp/tmp.56YaDbl1Fv/build_dir/target-mipsel_24kc+24kf_musl/Python-3.10.0 -L/tmp/tmp.56YaDbl1Fv/staging_dir/target-mipsel_24kc+24kf_musl/usr/lib -L/tmp/tmp.56YaDbl1Fv/staging_dir/target-mipsel_24kc+24kf_musl/lib -fno-semantic-interposition -fprofile-generate -Xlinker -export-dynamic -o python Programs/python.o -L. -lpython3.10 -ldl  -lpthread -lm -lz   -lm 
2021-11-26T23:34:02.5282224Z /tmp/tmp.56YaDbl1Fv/staging_dir/toolchain-mipsel_24kc+24kf_gcc-11.2.0_musl/bin/../lib/gcc/mipsel-openwrt-linux-musl/11.2.0/../../../../mipsel-openwrt-linux-musl/bin/ld: /tmp/tmp.56YaDbl1Fv/build_dir/target-mipsel_24kc+24kf_musl/Python-3.10.0/libpython3.10.so: undefined reference to `__fn_local_PyFloat_FromDouble.localalias'
2021-11-26T23:34:02.5295381Z collect2: error: ld returned 1 exit status
2021-11-26T23:34:02.5296804Z make[7]: Leaving directory '/tmp/tmp.56YaDbl1Fv/build_dir/target-mipsel_24kc+24kf_musl/Python-3.10.0'
2021-11-26T23:34:02.5298573Z make[7]: *** [Makefile:602: python] Error 1
2021-11-26T23:34:02.5299669Z make[6]: Leaving directory '/tmp/tmp.56YaDbl1Fv/build_dir/target-mipsel_24kc+24kf_musl/Python-3.10.0'
2021-11-26T23:34:02.5301298Z make[6]: *** [Makefile:532: build_all_generate_profile] Error 2
2021-11-26T23:34:02.5302447Z make[5]: Leaving directory '/tmp/tmp.56YaDbl1Fv/build_dir/target-mipsel_24kc+24kf_musl/Python-3.10.0'
2021-11-26T23:34:02.5304253Z make[5]: *** [Makefile:508: profile-gen-stamp] Error 2
2021-11-26T23:34:02.5305379Z make[4]: Leaving directory '/tmp/tmp.56YaDbl1Fv/build_dir/target-mipsel_24kc+24kf_musl/Python-3.10.0'
2021-11-26T23:34:02.5315419Z make[4]: *** [Makefile:520: profile-run-stamp] Error 2
2021-11-26T23:34:02.5316467Z make[3]: Leaving directory '/home/runner/cache/feeds/packages/lang/python/python3'
2021-11-26T23:34:02.5319446Z make[3]: *** [Makefile:388: /tmp/tmp.56YaDbl1Fv/build_dir/target-mipsel_24kc+24kf_musl/Python-3.10.0/.built] Error 2
2021-11-26T23:34:02.5320435Z time: package/feeds/packages/python3/compile#199.17#17.53#219.82
2021-11-26T23:34:02.5322700Z     ERROR: package/feeds/packages/python3 failed to build.
2021-11-26T23:34:02.5327403Z make[2]: *** [package/Makefile:116: package/feeds/packages/python3/compile] Error 1
2021-11-26T23:34:02.5328868Z make[2]: *** Waiting for unfinished jobs....
```

jefferyto, 2021-12-17:
Buildbots are also seeing the same error (https://downloads.openwrt.org/snapshots/faillogs/mipsel_24kc_24kf/packages/python3/compile.txt):

```
mipsel_24kc+24kf_gcc-11.2.0_musl/lib -znow -zrelro -L/builder/shared-workdir/build/sdk/build_dir/target-mipsel_24kc+24kf_musl/Python-3.10.0 -L/builder/shared-workdir/build/sdk/staging_dir/target-mipsel_24kc+24kf_musl/usr/lib -L/builder/shared-workdir/build/sdk/staging_dir/target-mipsel_24kc+24kf_musl/lib -fno-semantic-interposition -fprofile-generate -Xlinker -export-dynamic -o python Programs/python.o -L. -lpython3.10 -ldl  -lpthread -lm -lz   -lm 
/builder/shared-workdir/build/sdk/staging_dir/toolchain-mipsel_24kc+24kf_gcc-11.2.0_musl/bin/../lib/gcc/mipsel-openwrt-linux-musl/11.2.0/../../../../mipsel-openwrt-linux-musl/bin/ld: /builder/shared-workdir/build/sdk/build_dir/target-mipsel_24kc+24kf_musl/Python-3.10.0/libpython3.10.so: undefined reference to `__fn_local_PyFloat_FromDouble.localalias'
collect2: error: ld returned 1 exit status
make[7]: *** [Makefile:602: python] Error 1
make[7]: Leaving directory '/builder/shared-workdir/build/sdk/build_dir/target-mipsel_24kc+24kf_musl/Python-3.10.0'
make[6]: *** [Makefile:532: build_all_generate_profile] Error 2
make[6]: Leaving directory '/builder/shared-workdir/build/sdk/build_dir/target-mipsel_24kc+24kf_musl/Python-3.10.0'
make[5]: *** [Makefile:508: profile-gen-stamp] Error 2
make[5]: Leaving directory '/builder/shared-workdir/build/sdk/build_dir/target-mipsel_24kc+24kf_musl/Python-3.10.0'
make[4]: *** [Makefile:520: profile-run-stamp] Error 2
make[4]: Leaving directory '/builder/shared-workdir/build/sdk/build_dir/target-mipsel_24kc+24kf_musl/Python-3.10.0'
make[3]: *** [Makefile:388: /builder/shared-workdir/build/sdk/build_dir/target-mipsel_24kc+24kf_musl/Python-3.10.0/.built] Error 2
time: package/feeds/packages/python3/compile#398.97#27.43#405.06
```

Curious that mipsel_24kc_24kf is the only arch with this error.

jefferyto, 2022-01-05:
I have opened a ticket upstream: https://bugs.python.org/issue46265

closed 2022-01-06 by BKPepe