openwrt/packages #17225: radicale2: can't make it run

wehkah, 2021-11-28:
I am trying to install radicale2 onto my fb4040, running OpenWrt 21.02.1 but it simply does not start. I am sorry for not being able to provide better error description, but there is none. I repeated the installation several times from scratch, each time removing everything related to this package. I tried manual installation, I tried installation via GUI. There is also no output to any logfile.

After configuration, I call "/etc/init.d/radicale2 start", but nothing happens at all: no running service, no error message.

How can I make the app more verbose? How can I at least produce some logfile output?

This is frustrating. Thanks in advance for any help!

jefferyto, 2022-01-05:
What do you see when you run `logread` at the command line?

wehkah, 2022-01-08:
There's a missing dependency, as I found in this thread:

https://github.com/openwrt/packages/issues/11062

> Radicale does not start since the module urllib is missing.

After installing urllig, radicale2 is working now!

jefferyto, 2022-01-08:
radicale2 [depends](https://github.com/openwrt/packages/blob/032d0157c30bca400d058087e920c8a6b94851e5/net/radicale2/Makefile#L32) on python3, and python3 already depends on python3-urllib. Not sure why it wasn't already installed on your system.

closed 2022-01-08 by jefferyto

wehkah, 2022-01-09:
> Not sure why it wasn't already installed on your system.

The reason is more than obvious, just look at the other thread I quoted: the package management is missing a dependency. That has not been changed since OpenWRT 19.07, the version that the user in the other thread is talking about.

jefferyto, 2022-01-09:
If you actually read the thread you quoted, you would see that the issue was already fixed in master (which 21.02 is based on) and fixed in 19.07 in #11078.

wehkah, 2022-01-09:
What about the second part of the sentence you quoted?

"...,however, branch 19.07 does not have python3-urllib as a dependecy."

I did not remove the missing package manually, it simply was not installed automatically when I installed radicale2. 

The thread  #11078 is about radicale, while I tried to install radicale2. Is it possible that the maintainer fixed the dependency for radicale, but forget to fix it for radicale2 as well? Just a thought...

jefferyto, 2022-01-09:
> Is it possible that the maintainers fixed the dependency for radicale, but forgot to fix it for radicale2 as well?

No.

wehkah, 2022-01-15:
I am sorry to bother you, but I need your help again. Today I installed the available upgrade and now radicale2 once again won't start. :(

Radicale2 is now version 2.1.12-1. Again, there is no kind of reaction when I call

`/etc/init.d/radicale2 start`

No error, no log file, nothing. And of course: no running radicale.

But, when I call

`python3 -m radicale -C /var/etc/radicale2/config `

the server works as expected. Clients and users can connect.

The package python3-urllib is installed, so this can not be the cause of failure this time.

Why does the start script cause the server to fail, while a direct command makes it work?

wehkah, 2022-01-16:
This time logread tells me this:

```

Sun Jan 16 12:02:53 2022 daemon.err radicale2[7577]: Traceback (most recent call last):
Sun Jan 16 12:02:53 2022 daemon.err radicale2[7577]:   File "/usr/bin/radicale2", line 33, in <module>
Sun Jan 16 12:02:53 2022 daemon.err radicale2[7577]:     sys.exit(load_entry_point('Radicale==2.1.12', 'console_scripts', 'radicale')())
Sun Jan 16 12:02:53 2022 daemon.err radicale2[7577]:   File "/usr/bin/radicale2", line 25, in importlib_load_entry_point
Sun Jan 16 12:02:53 2022 daemon.err radicale2[7577]:     return next(matches).load()
Sun Jan 16 12:02:53 2022 daemon.err radicale2[7577]: StopIteration
Sun Jan 16 12:02:54 2022 daemon.info procd: Instance radicale2::radicale2 s in a crash loop 17 crashes, 1 seconds since last crash

```

jefferyto, 2022-01-17:
What packages do you have installed (`opkg list-installed`)?

wehkah, 2022-01-17:
Here you are:

```
apcupsd - 3.14.14-3
at - 3.2.1-1
ath10k-board-qca4019 - 20201118-3
ath10k-firmware-qca4019-ct - 2020-11-08-1
base-files - 1434-r16325-88151b8303
block-mount - 2021-01-04-c53b1882-1
busybox - 1.33.1-6
ca-bundle - 20210119-1
ca-certificates - 20210119-1
cgi-io - 2021-09-08-98cef9dd-20
collectd - 5.12.0-10
coreutils - 8.32-6
coreutils-date - 8.32-6
coreutils-md5sum - 8.32-6
curl - 7.80.0-1
dejavu-fonts-ttf-DejaVuSansMono-Bold - 2.37-1
dnsmasq - 2.85-8
dropbear - 2020.81-2
e2fsprogs - 1.45.6-2
fdisk - 2.36.1-2
firewall - 2021-03-23-61db17ed-1
fritz-caldata - 1
fritz-tffs - 1
fstools - 2021-01-04-c53b1882-1
fwtool - 2019-11-12-8f7fe925-1
getrandom - 2020-10-25-9ef88681-2
hashdeep - 4.4-1
hostapd-common - 2020-06-08-5a8b3662-35
imagemagick - 7.0.9-2
ip6tables - 1.8.7-1
iptables - 1.8.7-1
iw - 5.9-8fab0c9e-1
iwinfo - 2021-04-30-c45f0b58-2.1
jshn - 2021-05-16-b14c4688-2
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 5.4.154-1-d04d8d4d6f3bad7588be7c9e49004fda
kmod-ath - 5.4.154+5.10.68-1-1
kmod-ath10k-ct - 5.4.154+2021-09-22-e6a7d5b5-1
kmod-cfg80211 - 5.4.154+5.10.68-1-1
kmod-crypto-crc32c - 5.4.154-1
kmod-crypto-hash - 5.4.154-1
kmod-crypto-md5 - 5.4.154-1
kmod-crypto-sha1 - 5.4.154-1
kmod-crypto-sha256 - 5.4.154-1
kmod-crypto-sha512 - 5.4.154-1
kmod-fs-ext4 - 5.4.154-1
kmod-gpio-button-hotplug - 5.4.154-3
kmod-hid - 5.4.154-1
kmod-hid-generic - 5.4.154-1
kmod-hwmon-core - 5.4.154-1
kmod-input-core - 5.4.154-1
kmod-input-evdev - 5.4.154-1
kmod-ip6tables - 5.4.154-1
kmod-ipt-conntrack - 5.4.154-1
kmod-ipt-core - 5.4.154-1
kmod-ipt-nat - 5.4.154-1
kmod-ipt-offload - 5.4.154-1
kmod-leds-gpio - 5.4.154-1
kmod-lib-crc-ccitt - 5.4.154-1
kmod-lib-crc16 - 5.4.154-1
kmod-lib-crc32c - 5.4.154-1
kmod-mac80211 - 5.4.154+5.10.68-1-1
kmod-nf-conntrack - 5.4.154-1
kmod-nf-conntrack6 - 5.4.154-1
kmod-nf-flow - 5.4.154-1
kmod-nf-ipt - 5.4.154-1
kmod-nf-ipt6 - 5.4.154-1
[... 165 lines cut ...]
rpcd-mod-rad2-enc - 20190109
rpcd-mod-rrdns - 20170710
swconfig - 12
terminfo - 6.2-3
ubi-utils - 2.1.2-1
uboot-envtools - 2021.01-10
ubox - 2020-10-25-9ef88681-2
ubus - 2021-06-30-4fc532c8-2
ubusd - 2021-06-30-4fc532c8-2
uci - 2020-10-06-52bbc99f-5
uclibcxx - 0.2.5-3
uclient-fetch - 2021-05-14-6a6011df-1
uhttpd - 2021-03-21-15346de8-2
uhttpd-mod-ubus - 2021-03-21-15346de8-2
urandom-seed - 3
urngd - 2020-01-21-c7f7b6b6-1
usbids - 0.354-1
usbutils - 013-2
usign - 2020-05-23-f1f65026-1
wifischedule - 1-3
wireless-regdb - 2021.04.21-1
wpad-basic-wolfssl - 2020-06-08-5a8b3662-35
zlib - 1.2.11-3

```

jefferyto, 2022-01-17:
Can you try removing and reinstalling the radicale2 package?

wehkah, 2022-01-18:
Been there, done that... That was also my first thought, and I even tried two different ways of deinstalling and reinstalling: via web GUI and via command line. But neither way doesn't make any difference: the start script fails, while starting the service by 

`/usr/bin/python3 -m radicale -C /var/etc/radicale2/config &`

does work. Until I will find the solution to this, I put the above command into /etc/rc.local, so the service is started when the router starts. 

I took a look into /etc/init.d/radicale2 and according to my limited knowledge of scripting, I think that this script creates the config file  `/var/etc/radicale2/config`, using the contents of `/etc/config/radicale2` and then it starts the server. Correct my if I am wrong there!

But if I understood that correctly, the command line should not work either, because it uses the same config file. So I guess there's some deeper magic going on.

jefferyto, 2022-01-19:
I have installed radicale2 on armvirt-32 (in qemu) and verified that it starts with no issue. The problem isn't the init script, something is wrong with your Python installation.

wehkah, 2022-01-19:
I removed radicale2 and every python package that was installed. After that, I re-installed radicale2, and it automatically installed these packages:

```
root@fb4040:~# opkg install radicale2
Installing radicale2 (2.1.12-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/radicale2_2.1.12-1_arm_cortex-a7_neon-vfpv4.ipk
Installing libpython3-3.9 (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/libpython3-3.9_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-base (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-base_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing libffi (3.3-2) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/libffi_3.3-2_arm_cortex-a7_neon-vfpv4.ipk
Installing libbz2-1.0 (1.0.8-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/base/libbz2-1.0_1.0.8-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-light (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-light_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-unittest (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-unittest_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-ncurses (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-ncurses_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-ctypes (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-ctypes_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-pydoc (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-pydoc_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-decimal (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-decimal_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-multiprocessing (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-multiprocessing_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-codecs (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-codecs_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing libreadline8 (8.1-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/base/libreadline8_8.1-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-readline (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-readline_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing libgdbm (1.18.1-2) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/libgdbm_1.18.1-2_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-gdbm (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-gdbm_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-email (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-email_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-urllib (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-urllib_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-xml (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-xml_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing libsqlite3-0 (3330000-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/libsqlite3-0_3330000-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-sqlite3 (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-sqlite3_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-logging (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-logging_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-distutils (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-distutils_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-openssl (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-openssl_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-cgi (3.9.9-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-cgi_3.9.9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-cgitb (3.9.9-1) to root...
[... cut ...]
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-pycparser_2.20-5_arm_cortex-a7_neon-vfpv4.ipk
Installing python3-cffi (1.14.6-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/arm_cortex-a7_neon-vfpv4/packages/python3-cffi_1.14.6-1_arm_cortex-a7_neon-vfpv4.ipk
Configuring python3-ply.
Configuring python3-pycparser.
Configuring python3-cffi.
Configuring python3-bcrypt.

```
But still, radicale2 will only start when I use the command line.

wehkah, 2022-01-21:
Some additional information:

Although the init script fails, it tells me that radicale2 is running:

```
root@fb4040:~# /etc/init.d/radicale2 status
running
```

That statement is not true:

```
root@fb4040:~# ps | grep radical
24415 root      1100 R    grep radical
```

Where does the init script that information? I can not find any pid file for this.
 
If I use the argument "trace" I get this error message:

```
root@fb4040:~# logread
...
Fri Jan 21 08:06:50 2022 daemon.warn procd: Seccomp support for radicale2::radicale2 not available
```

Followed by the same error messages as shown above ("traceback...") about the crash loop. But still, the init script insists that radicale2 is running:

```
root@fb4040:~# /etc/init.d/radicale2 status
running
```