openwrt/packages #17232: boringssl: Failing to build for arm_fa526

martinschneider, 2021-11-29:
Maintainer: @martinschneider
Environment: mt7621

Description:
Reported [here](https://github.com/openwrt/packages/commit/4fde0fdad558db07b168fcf8828f9c609c517b50#commitcomment-58839415).

```
make[2]: Entering directory '/home/vladdrako/openwrt/feeds/packages/libs/boringssl'
rm -f /home/vladdrako/openwrt/build_dir/target-mipsel_24kc_musl/boringssl-20210608/.built
touch /home/vladdrako/openwrt/build_dir/target-mipsel_24kc_musl/boringssl-20210608/.built_check
MAKEFLAGS="" /home/vladdrako/openwrt/staging_dir/host/bin/ninja -v -j1 -C /home/vladdrako/openwrt/build_dir/target-mipsel_24kc_musl/boringssl-20210608 
ninja: Entering directory `/home/vladdrako/openwrt/build_dir/target-mipsel_24kc_musl/boringssl-20210608'
[1/398] /home/vladdrako/openwrt/staging_dir/host/bin/ccache /home/vladdrako/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/bin/mipsel-openwrt-linux-musl-gcc -DBORINGSSL_IMPLEMENTATION -DBORINGSSL_SHARED_LIBRARY -I/home/vladdrako/openwrt/build_dir/target-mipsel_24kc_musl/boringssl-20210608/third_party/googletest/include -I/home/vladdrako/openwrt/build_dir/target-mipsel_24kc_musl/boringssl-20210608/crypto/../include -I/home/vladdrako/openwrt/build_dir/target-mipsel_24kc_musl/boringssl-20210608/crypto/fipsmodule/../../include -O3 -pipe -march=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/home/vladdrako/openwrt/build_dir/target-mipsel_24kc_musl/boringssl-20210608=boringssl-20210608 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Werror -Wformat=2 -Wsign-compare -Wmissing-field-initializers -Wwrite-strings -Wvla -ggdb -Wall -fvisibility=hidden -fno-common -Wno-free-nonheap-object -Wimplicit-fallthrough -Wmissing-prototypes -Wold-style-definition -Wstrict-prototypes -Wshadow -std=c11 -D_XOPEN_SOURCE=700 -DNDEBUG -fPIC -MD -MT crypto/fipsmodule/CMakeFiles/fipsmodule.dir/bcm.c.o -MF crypto/fipsmodule/CMakeFiles/fipsmodule.dir/bcm.c.o.d -o crypto/fipsmodule/CMakeFiles/fipsmodule.dir/bcm.c.o -c /home/vladdrako/openwrt/build_dir/target-mipsel_24kc_musl/boringssl-20210608/crypto/fipsmodule/bcm.c
FAILED: crypto/fipsmodule/CMakeFiles/fipsmodule.dir/bcm.c.o 
/home/vladdrako/openwrt/staging_dir/host/bin/ccache /home/vladdrako/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/bin/mipsel-openwrt-linux-musl-gcc -DBORINGSSL_IMPLEMENTATION -DBORINGSSL_SHARED_LIBRARY -I/home/vladdrako/openwrt/build_dir/target-mipsel_24kc_musl/boringssl-20210608/third_party/googletest/include -I/home/vladdrako/openwrt/build_dir/target-mipsel_24kc_musl/boringssl-20210608/crypto/../include -I/home/vladdrako/openwrt/build_dir/target-mipsel_24kc_musl/boringssl-20210608/crypto/fipsmodule/../../include -O3 -pipe -march=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/home/vladdrako/openwrt/build_dir/target-mipsel_24kc_musl/boringssl-20210608=boringssl-20210608 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Werror -Wformat=2 -Wsign-compare -Wmissing-field-initializers -Wwrite-strings -Wvla -ggdb -Wall -fvisibility=hidden -fno-common -Wno-free-nonheap-object -Wimplicit-fallthrough -Wmissing-prototypes -Wold-style-definition -Wstrict-prototypes -Wshadow -std=c11 -D_XOPEN_SOURCE=700 -DNDEBUG -fPIC -MD -MT crypto/fipsmodule/CMakeFiles/fipsmodule.dir/bcm.c.o -MF crypto/fipsmodule/CMakeFiles/fipsmodule.dir/bcm.c.o.d -o crypto/fipsmodule/CMakeFiles/fipsmodule.dir/bcm.c.o -c /home/vladdrako/openwrt/build_dir/target-mipsel_24kc_musl/boringssl-20210608/crypto/fipsmodule/bcm.c
In file included from crypto/fipsmodule/bcm.c:86:
crypto/fipsmodule/modes/cfb.c: In function 'CRYPTO_cfb128_encrypt':
crypto/fipsmodule/modes/cfb.c:120:17: error: writing 1 byte into a region of size 0 [-Werror=stringop-overflow=]
  120 |         ivec[n] = c;
      |         ~~~~~~~~^~~
crypto/fipsmodule/modes/cfb.c:61:56: note: at offset 16 into destination object 'ivec' of size [0, 16]
   61 |                            const AES_KEY *key, uint8_t ivec[16], unsigned *num,
      |                                                ~~~~~~~~^~~~~~~~
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.
make[2]: *** [Makefile:55: /home/vladdrako/openwrt/build_dir/target-mipsel_24kc_musl/boringssl-20210608/.built] Error 1
```

martinschneider, 2022-03-09:
I cannot reproduce this. My steps are:

```
make menuconfig # configure Target System "MediaTek Ralink MIPS" and Subtarget "MT7621 based boards"
make tools/install
make toolchain/install
make package/boringssl/compile -j1 V=s
```

This successfully creates `bin/packages/mipsel_24kc/packages/boringssl_20210608-1_mipsel_24kc.ipk`.

Is there anything I'm missing?

neheb, 2022-03-09:
sounds like adding volatile will fix it.

GeorgeSapkin, 2025-11-15:
@martinschneider still relevant? Considering the package hasn't been updated in 4 years.