openwrt/packages #17234: nfs-kernel-server: broken on 21.02.x

salim-b, 2021-11-29:
Maintainer: @neheb @dangowrt
Environment: `arm_cortex-a15_neon-vfpv4`, `ipq806x`, TP-Link Archer C2600, OpenWrt 21.02.1 stable

### Description:

I'm unable to get my NFS server config to work after upgrading from OpenWrt 19.07 to 21.02:

- When I try to mount an NFS share on my desktop, **the mount command always times out without success – or hangs indefinitively in case of no timeout**. I tried various different mount settings, both `nfsvers=3` and `nfsvers=4` (and `nfsvers=4.2`), all without luck.
- `showmount -a` on the router shows a connected client (although the mount never succeeded). Same command on 19.07 doesn't show any client regardless of successful mounts.
- `showmount -e` shows all my NFS shares fine, same as with 19.07.
- The ["NFS Service Setup" mentioned in the docs](https://openwrt.org/docs/guide-user/services/nas/nfs.server#nfs_service_setup) that exports a directory under `/tmp/` to a different directory under `/tmp/`, i.e. both on the server device, works fine.
- `cat /proc/fs/nfsd/versions` gives me `-2 +3 +4 +4.1 +4.2`, same as on 19.07.

Unfortunately, I lack the knowledge on how to further debug this issue. Would be glad if someone more proficient could provide some guidance. Or maybe this is even a known issue?

### OpenWrt builds I've tried so far

The following info is not really related to the above problem, I think. But nevertheless I've written down my little 21.02 build journey:

1. First I tried with [@ACwifidude's NSS build](https://forum.openwrt.org/t/ipq806x-nss-build-netgear-r7800-tp-link-c2600-linksys-ea8500/82525) (the [`ath10k-ct` build based on 21.02 stable](https://github.com/ACwifidude/openwrt/blob/openwrt-21.02-nss-qsdk10.0/bin/targets/ipq806x/generic/C2600-20211106-Stable2012NSS-sysupgrade.bin)).

   The three packages `kmod-crypto-cts`, `kmod-fs-nfs-common-rpcsec` and `kmod-fs-nfsd` are not part of the image but required by `nfs-kernel-server`. Since `kmod-*` packages need to match the exact build version, @ACwifidude kindly offers [some more prebuilt packages](https://github.com/ACwifidude/openwrt/tree/openwrt-21.02-nss-qsdk10.0/bin/targets/ipq806x/generic/packages)) from where I've installed the mentioned `kmod-*` packages. Then I've installed `nfs-kernel-server` from OpenWrt's 21.02's regular package repository.

   The above NFS server issue is present.

2. I assumed the NFS server problem was caused by some remaining/obscure package/dependency incompatibilities. Thus I compiled my own image that preincluded all the required packages.

   Exactly the same result, above NFS server issue is present.

3. Then I tried with latest OpenWrt 21.02.1 stable.

   Exactly the same result, above NFS server issue is present.

4. Finally I reverted back to OpenWrt 19.07.08 stable since I wanted to have back a working NFS setup.

   Everything works fine with the *exact* same NFS configuration as I used on 21.02.

KolonelKustard, 2022-01-01:
I can confirm I'm seeing the same behaviour, here's my setup:

Model: `Linksys WRT3200ACM`
Architecture: `ARMv7 Processor rev 1 (v7l)`
Firmware Version: `OpenWrt 21.02.1 r16325-88151b8303 / LuCI openwrt-21.02 branch git-21.295.67054-13df80d`
Kernel Version: `5.4.154`

I'm upgrading from `LEDE Reboot 17.01.4 (Linux 4.4.92)` which worked fine as an NFS server.

I'm planning to downgrade to 19.07.08 as per @salim-b in due course, will update this comment with how that goes.

hb9xcg, 2022-01-01:
Hi,
I got NFS4 server export working on OpenWrt 21.02.1 with help of https://www.adrian.idv.hk/2021-05-05-nfsv4/
The overlay filesystem seems indeed to make the kernelmode NFS server unhappy. At least I'm able to export from /tmp.

salim-b, 2022-01-01:
> I got NFS4 server export working on OpenWrt 21.02.1 with help of https://www.adrian.idv.hk/2021-05-05-nfsv4/

@hb9xcg I just read that blog post. The following part from section [*OpenWRT example*](https://www.adrian.idv.hk/2021-05-05-nfsv4/#openwrt-example) is interesting:

> The second is the `fsid` option for NFSv4. It is required to say `fsid=0` for the root directory in NFSv4 but the other should be automatically detected. **However, I found that we still need to specify `fsid` in OpenWRT. So my `/etc/exports` now looks like this**:
>
> ```
> /tmp	*(rw,all_squash,insecure,no_subtree_check,sync,fsid=0)
> /tmp/dir1	*(rw,all_squash,insecure,no_subtree_check,sync,fsid=1)
> /tmp/dir2	*(rw,all_squash,insecure,no_subtree_check,sync,fsid=2)
> ```

Is the whole secret to make the NFS server work on OpenWrt 21.02 simply specifying `fsid=1`, `fsid=2` etc. for all the non-root NFSv4 exports?

KolonelKustard, 2022-01-02:
> Is the whole secret to make the NFS server work on OpenWrt 21.02 simply specifying `fsid=1`, `fsid=2` etc. for all the non-root NFSv4 exports?

Haha! Bingo!! I thought I'd give this a try before downgrading and all is good. This `/etc/exports` caused NFS clients to hang (even on the router itself):

```
/mnt/manilow/ *(rw,root_squash,insecure,no_subtree_check)
```

`fsid` for the win, this `/etc/exports` works:

```
/mnt/manilow/ *(rw,root_squash,insecure,no_subtree_check,fsid=0)
```

Note I also hacked around with my firewall rules and did a few router reboots. So there's a chance it was some other hack I did that fixed things. :shrug:

> The overlay filesystem seems indeed to make the kernelmode NFS server unhappy

FYI this isn't the problem I was having. I was getting clients hanging when exporting under `/tmp`.

salim-b, 2022-01-02:
@KolonelKustard Thanks for the update!

I can confirm that after explicitly setting `fsid` *to a unique integer between 0-255* for all exports[^1], the NFS server from OpenWrt 21.02 package feeds (`nfs-kernel-server 2.5.2-1`) actually works! 🥳

I've also tried to set `fsid` to the UUID of the underlying filesystem[^2] of each export. While this works fine with `nfs-kernel-server 2.3.4-3` from OpenWrt 19.07, with `nfs-kernel-server 2.5.2-1` from OpenWrt 21.02 it resulted in the same indefinite hang when mounting NFS exports as descibed in my initial post.

[^1]: Note [that](https://man.cx/exports) 

    > for NFSv4, there is a distinguished filesystem which is the root of all exported filesystem. This is specified with `fsid=root` or `fsid=0` both of which mean exactly the same thing.

[^2]: UUIDs can be determined using `lsblk -f` for example.

salim-b, 2022-01-04:
I've added a paragraph about this issue and the workaround (specifying explicit `fsid`s for all shares) to the relevant OpenWrt docs [here](https://openwrt.org/docs/guide-user/services/nas/nfs.server#server_configuration). Should be removed again once the underlying problem is fixed.

topiolli, 2022-01-19:
Another data point on OpenWrt 21.02.1. Mounts hang even with an explicit fsid if ip-based access control is used. This doesn't work:

```
/mnt/usbdrive 10.10.10.30(fsid=0,ro,sync,no_subtree_check) 10.10.10.3(fsid=0,rw,sync,no_root_squash,no_subtree_check)
```

This works

```
/mnt/usbdrive *(fsid=0,rw,sync,no_subtree_check)
```

The problem is I cannot specify client-specific rules.

salim-b, 2022-01-19:
@topiolli I guess you're missing to specify the **netmask** besides the IP address. See the [manpage of `/etc/exports`](https://man.cx/exports).

I'm successfully using export entries like

```
/path/to/nfs4_root/              192.168.1.0/25(fsid=root,ro,async,secure,no_subtree_check,pnfs,root_squash,anonuid=1000,anongid=1000)
/path/to/nfs4_root/some_subdir   192.168.1.0/26(fsid=1,rw,wdelay,async,secure,no_subtree_check,pnfs,root_squash,anonuid=1000,anongid=1000)
```

topiolli, 2022-01-19:
@salim-b Thanks for the suggestion, but "single host" is the first supported format listed on the man page. I however got it work with my original /etc/exports by downgrading to an older nfs-kernel-server version as suggested above. Exact steps in my case:

```
$ opkg print-architecture
arch all 1
arch noarch 1
arch arm_cortex-a9_vfpv3-d16 10
$ wget https://downloads.openwrt.org/releases/19.07.8/packages/arm_cortex-a9_vfpv3-d16/packages/nfs-kernel-server_2.3.4-3_arm_cortex-a9_vfpv3-d16.ipk
$ opgk update
$ opkg remove nfs-kernel-server
$ opkg install ./nfs-kernel-server_2.3.4-3_arm_cortex-a9_vfpv3-d16.ipk
```

salim-b, 2022-01-19:
> Thanks for the suggestion, but "single host" is the first supported format listed on the man page.

My fault, you're right (those who can read have a clear advantage 🙈).

Happy to hear that downgrading at least helped.

zhanglongqi, 2022-11-20:
`fsid=0` works for macOS also