openwrt/packages #17235: modemmanager: 1.16.6 not works in latest OpenWrt 21.02

zokl, 2021-11-29:
Maintainer: @nickberry17
Environment: (x86_64, OpenWrt 21.02)

Description:

The final release of the network infrastructure in openwrt 21.02 (https://github.com/openwrt/openwrt/tree/openwrt-21.02) is not compatible with Modemmanager 1.16.6 and its dependencies. Modemmanager is not able to use Telit FN980 cdc-wdm modem. New application and libraries from master branch works fine.

feckert, 2021-12-03:
From my point of view, this is not necessary.
Have look at https://openwrt.org/docs/guide-developer/package-policies#dependencies

Since we define the dependency as runtime DEPENDS and therefore gets build before this package.
https://github.com/openwrt/packages/blob/4be2a5b73757eb0037d41da962518eaac4e4d3ac/net/modemmanager/Makefile#L43-L44
Or is that not a runtime dependency?

zokl, 2021-12-03:
> From my point of view, this is not necessary. Have look at https://openwrt.org/docs/guide-developer/package-policies#dependencies
>
> Since we define the dependency as runtime DEPENDS and therefore [...]

I think there must be a problem because without the declaration of PKG_BUILD_DEPENDS I am not able to build the system with working QMI communication with modem. I attached a log from the system which was built without mentioned variable.

[modem.log](https://github.com/openwrt/packages/files/7648134/modem.log)

professor-jonny, 2021-12-06:
Are you sure your modem is in qmi mode and you have the qmi related stuff installed?
Quote:

```
 [plugin/telit] could not grab port cdc-wdm0: Cannot add port 'usbmisc/cdc-wdm0', unhandled port type
 [plugin/telit] ignoring QMI net port wwan0 as QMI support isn't available
```

what is the output of dmesg ?

zokl, 2021-12-06:
> Are you sure your modem is in qmi mode and you have the qmi related stuff installed? Quote:
>
> ```
>  [plugin/telit] could not grab port cdc-wdm0: Cannot add port 'usbmisc/cdc-wdm0', unhandled port [...]

Yes I am sure. The mentioned output is compilation of OpenWRt from sources without  `PKG_BUILD_DEPENDS:=MODEMMANAGER_WITH_QMI:libqmi`. If I put this variable into modemmanager makefile the modemmanager telit's detection works fine. The modemmanager works too without PKG_BUILD_DEPENDS definition if I make single thread compilation.

zokl, 2021-12-08:
Here is a snippet from my workable Makefile:

```
#
# Copyright (C) 2016 Velocloud Inc.
# Copyright (C) 2016 Aleksander Morgado <aleksander@aleksander.es>
#
# This is free software, licensed under the GNU General Public License v2.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=modemmanager
PKG_VERSION:=1.16.6
PKG_RELEASE:=1

PKG_SOURCE:=ModemManager-$(PKG_VERSION).tar.xz
PKG_SOURCE_URL:=https://www.freedesktop.org/software/ModemManager
PKG_HASH:=2a90b6260f66d3135609d62667ada73416694d717e7fd9b73223e3703a499617
PKG_BUILD_DIR:=$(BUILD_DIR)/ModemManager-$(PKG_VERSION)

PKG_MAINTAINER:=Nicholas Smith <nicholas.smith@telcoantennas.com.au>
PKG_LICENSE:=GPL-2.0-or-later
PKG_LICENSE_FILES:=COPYING

PKG_INSTALL:=1
PKG_BUILD_PARALLEL:=1

include $(INCLUDE_DIR)/package.mk
include $(INCLUDE_DIR)/nls.mk

define Package/modemmanager/config
  source "$(SOURCE)/Config.in"
endef

PKG_BUILD_DEPENDS:=MODEMMANAGER_WITH_MBIM:libmbim MODEMMANAGER_WITH_QMI:libqmi

define Package/modemmanager
  SECTION:=net
  CATEGORY:=Network
  TITLE:=Control utility for any kind of mobile broadband modem
  URL:=https://www.freedesktop.org/wiki/Software/ModemManager
  DEPENDS:= \
        $(INTL_DEPENDS) \
        +glib2 \
        +dbus \
        +ppp \
        +MODEMMANAGER_WITH_MBIM:libmbim \
        +MODEMMANAGER_WITH_QMI:libqmi
endef

...
```

zokl, 2021-12-24:
The problem is in parallel build betwen modemmanager and libqmi. I need prepare better description.

closed 2021-12-24 by zokl