openwrt/packages #17237: kmod-ata-ahci won't recognize HDD connected in SATA ports (Non-USB connected)

pbrena, 2021-11-29:
Environment: (SoC Type: MediaTek MT7621 ver:1 eco:3, MIPS: machine MQmaker WiTi, OpenWrt 21.02.1 r16325-88151b8303 / LuCI openwrt-21.02 branch git-21.295.67054-13df80d)

Description:

OpenWrt 21 fails to recognize Disk Drives directly connected to the SATA ports (WITI hardware include 2 ports), please notice that the HDD are not connected thru the USB port.

As far as I understand kmod-ata-ahci is the one responsible.

If I go back to OpenWrt 19 (same HW Involved), the drive immediately  shows up in the log

Please advice

[LogWiti19ok.txt](https://github.com/openwrt/packages/files/7621145/LogWiti19ok.txt)
[LogWiti21Fails.txt](https://github.com/openwrt/packages/files/7621146/LogWiti21Fails.txt)

brada4, 2021-11-30:
Both kernel logs get sda disk behind pci 0000:02:00.0
Maybe you miss filesystem drivers ?

pbrena, 2021-11-30:
I missed them actually, but...  Now, I installed ext4 support along with all what is listed in the "Using storage devices" Guide, although it seems focused in USB attached Drives, unfortunately I still can't use the drive.

This is what I expected to see in the new log (and is still absent):

 scsi 1:0:0:0: Direct-Access     ATA      ST9320325AS      SDM1 PQ: 0 ANSI: 5
 sd 1:0:0:0: [sda] 625142448 512-byte logical blocks: (320 GB/298 GiB)
 sd 1:0:0:0: [sda] Write Protect is off
 sd 1:0:0:0: [sda] Mode Sense: 00 3a 00 00
 sd 1:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
 sda: sda1 sda2 sda4

See the responses of commands

root@OpenWrt:~# ls -al /dev/sd*
ls: /dev/sd*: No such file or directory

root@OpenWrt:~# block info 
/dev/mtdblock5: UUID="617d7fdf-f79626d5-329864ea-bd8562a7" VERSION="4.0" MOUNT="/rom" TYPE="squashfs"
/dev/mtdblock6: MOUNT="/overlay" TYPE="jffs2"

Are there something missed to support Sata ports in the board (vs USB attached ports)?

I include the new Log withe the packages installed

[2ndLogWiti21.txt](https://github.com/openwrt/packages/files/7629294/2ndLogWiti21.txt)

pbrena, 2021-11-30:
closed by mistake

BKPepe, 2021-12-02:
Closing. This bug does not belong here.

Package kmod-ata-ahci is provided by core repository. Take a look at this:
https://openwrt.org/bugs

closed 2021-12-02 by BKPepe