openwrt/packages #17245: FRR: on mips64_octeonplus fails to link (ld)

kalkocz, 2021-12-01:
Maintainer: @lucize
Environment: ( mips64_octeonplus, OpenWrt SNAPSHOT)

Description: 

error output from compile.txt in ../snapshot/faillogs/mips64_octeonplus/packages/frr
 CC       yang/ietf/ietf-bgp-types.yang.lo
CC       yang/frr-module-translator.yang.lo
CCLD     lib/libfrrcares.la
OpenWrt-libtool: link: warning: library `/builder/shared-workdir/build/sdk/staging_dir/toolchain-mips64_octeonplus_64_gcc-11.2.0_musl/lib/libatomic.la' was moved.
/builder/shared-workdir/build/sdk/tmp/ccgPNQVD.s: Assembler messages:
/builder/shared-workdir/build/sdk/tmp/ccgPNQVD.s:19: Error: PC-relative reference to a different section
/builder/shared-workdir/build/sdk/tmp/ccgPNQVD.s:20: Error: PC-relative reference to a different section
lto-wrapper: fatal error: /builder/shared-workdir/build/sdk/staging_dir/toolchain-mips64_octeonplus_64_gcc-11.2.0_musl/bin/mips64-openwrt-linux-musl-gcc returned 1 exit status
compilation terminated.
/builder/shared-workdir/build/sdk/staging_dir/toolchain-mips64_octeonplus_64_gcc-11.2.0_musl/bin/../lib/gcc/mips64-openwrt-linux-musl/11.2.0/../../../../mips64-openwrt-linux-musl/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make[5]: *** [Makefile:6939: lib/libfrrcares.la] Error 1
make[5]: Leaving directory '/builder/shared-workdir/build/sdk/build_dir/target-mips64_octeonplus_64_musl/frr-c18235d5b1e33fad9d3181aa29fefa48b8c6ff3f'
make[4]: *** [Makefile:5542: all] Error 2
make[4]: Leaving directory '/builder/shared-workdir/build/sdk/build_dir/target-mips64_octeonplus_64_musl/frr-c18235d5b1e33fad9d3181aa29fefa48b8c6ff3f'
make[3]: *** [Makefile:247: /builder/shared-workdir/build/sdk/build_dir/target-mips64_octeonplus_64_musl/frr-c18235d5b1e33fad9d3181aa29fefa48b8c6ff3f/.built] Error 2
time: package/feeds/packages/frr/compile#286.01#52.63#420.59

BrainSlayer, 2022-02-15:
`--- lib/xref.h  (revision 48380)
+++ lib/xref.h  (working copy)
@@ -122,7 +122,7 @@
 extern void xref_block_add(struct xref_block *block);
 extern void xref_gcc_workaround(const struct xref *xref);

-#ifndef HAVE_SECTION_SYMS
+#if !defined(HAVE_SECTION_SYMS) || defined(__mips64)
 /* we have a build system patch to use GNU ld on Solaris;  if that doesn't
  * work we end up on Solaris ld which doesn't support the section start/end
  * symbols.
`

lucize, 2022-02-15:
thanks !
can you open an issue about this on frr ?

lucize, 2022-02-17:
made a PR also opened an issue with frr

closed 2022-02-17 by neheb

BrainSlayer, 2022-02-17:
> thanks ! can you open an issue about this on frr ?

consider that i'm a the dd-wrt developer. i have no time to maintain yet another patch discussion. i just want to give a hint how a possible solution could look like. so please be so kind and manage this by yourself

PR #17879 (2022-02-17): frr: branch update, mips64 fix

Maintainer: me
build test mips64
Fixes: #17245 
Fixes: #17149