openwrt/packages #17246: ERROR: package/system/ca-certificates failed to build.

openips, 2021-12-01:
branch                  master
target x64

error log below
_____________________________________________________
Ignoring certificate "D-TRUST Root CA 3 2013".  SAUTH=CKT_NSS_MUST_VERIFY_TRUST, EPROT=CKT_NSS_TRUSTED_DELEGATOR
Ignoring certificate "GlobalSign Secure Mail Root R45".  SAUTH=CKT_NSS_MUST_VERIFY_TRUST, EPROT=CKT_NSS_TRUSTED_DELEGATOR
Ignoring certificate "GlobalSign Secure Mail Root E45".  SAUTH=CKT_NSS_MUST_VERIFY_TRUST, EPROT=CKT_NSS_TRUSTED_DELEGATOR
Traceback (most recent call last):
  File "certdata2pem.py", line 125, in <module>
    cert = x509.load_der_x509_certificate(obj['CKA_VALUE'])
TypeError: load_der_x509_certificate() missing 1 required positional argument: 'backend'
make[5]: *** [Makefile:6: all] Error 1
make[5]: Leaving directory '/LEDE/openwrt-master/build_dir/target-x86_64_musl/ca-certificates-20211016/work/mozilla'
make[4]: *** [Makefile:9: all] Error 2
make[4]: Leaving directory '/LEDE/openwrt-master/build_dir/target-x86_64_musl/ca-certificates-20211016/work'
make[3]: *** [Makefile:73: /LEDE/openwrt-master/build_dir/target-x86_64_musl/ca-certificates-20211016/.built] Error 2
make[3]: Leaving directory '/LEDE/openwrt-master/package/system/ca-certificates'
time: package/system/ca-certificates/compile#0.45#0.08#0.58
    ERROR: package/system/ca-certificates failed to build.
make[2]: *** [package/Makefile:116: package/system/ca-certificates/compile] Error 1
make[2]: Leaving directory '/LEDE/openwrt-master'
make[1]: *** [package/Makefile:110: /LEDE/openwrt-master/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/LEDE/openwrt-master'
make: *** [/LEDE/openwrt-master/include/toplevel.mk:230: world] Error 2

siwind, 2021-12-01:
ME to ~

```
build_dir/target-mipsel_24kc_musl/ca-certificates-20211016/work/mozilla/certdata2pem.py", line 31, in <module>
    from cryptography import x509
ModuleNotFoundError: No module named 'cryptography'
make[5]: *** [Makefile:6: all] Error 1
```

openips, 2021-12-02:
fixed

closed 2021-12-02 by openips