openwrt/packages #17252: stubby: Use standard notation for IPv6 local loopback address ( ::1 instead of 0::1 )

shoulders, 2021-12-02:
# Issue template (remove lines from top till here)

Maintainer: Not sure how to do this
Environment: 
<html>
<body>

Item | Value
-- | --
Model | BT Home Hub 5A
Architecture | xRX200 rev 1.2
Firmware Version | OpenWrt 21.02.1 r16325-88151b8303 / LuCI openwrt-21.02 branch git-21.295.67054-13df80d
Kernel Version | 5.4.154

</body>
</html>

Description:

Within the documentation and config files of stubby  and the OpenWrt instructions you use the address **0::1** for the IPv6 local loopback address (which does work) but I cannot find anywhere else that uses this form, everyone uses **::1**

I think these instances should be changed to keep things easier to read unless there is a technical reason for this.

These are a few that I found

README.md (several occurances)
https://github.com/openwrt/packages/blob/master/net/stubby/files/README.md

https://github.com/openwrt/packages/blob/04b82e9da01e8c067702867e3ad2a604f481b169/net/stubby/files/stubby.conf#L19

https://github.com/openwrt/packages/blob/04b82e9da01e8c067702867e3ad2a604f481b169/net/stubby/files/stubby.yml#L14

I have posted this over at **getdnsapi/stubby** aswell [****](https://github.com/getdnsapi/stubby/issues/304)

jamesmacwhite, 2021-12-06:
Hi. Not the maintainer I'm afraid, but submit a PR and someone with push access can help get it merged.

msvamp, 2022-07-05:
This seems like a limitation in the upstream stubby:
* [stubby.yml.example (line 17)](https://github.com/getdnsapi/stubby/blob/d0f5b97c8311aaa3cefef939f0ade36dc0d1ba7b/stubby.yml.example#L17)
* [stubby.yml.example (lines 126-127)](https://github.com/getdnsapi/stubby/blob/d0f5b97c8311aaa3cefef939f0ade36dc0d1ba7b/stubby.yml.example#L126:L127)

However, no reason for this limitation is specified.

In my recent PR, I have added a note in the README about this: https://github.com/openwrt/packages/pull/18863/files#diff-19dbc41aac11b9f660a125da07966a249ce1a0407f1072ea23ac7c28efc35113R232-R234