openwrt/packages #17266: iwlwifi 5.4.162+5.15-rc6-1-1 pvm firmware ax210 timeout

jpsollie, 2021-12-03:
Maintainer: Felix Fietkau <nbd@nbd.name>
Environment: (mips24kc, mikrotik rbm33g, snapshot)

Description:
I haven't been able to find a working firmware for my ax210 m.2 card.
According to a few sites, 5.11 introduced an error which was fixed in 5.12 and later:
https://bugzilla.kernel.org/show_bug.cgi?id=212371#c13
tried versions 66,64,63,62 (with PVM file) and 59 (without PVM file), but the error is always the same:
```
Fri Dec  3 11:41:43 2021 kern.warn kernel: [   38.324535] iwlwifi 0000:02:00.0: api flags index 2 larger than supported by driver
Fri Dec  3 11:41:43 2021 kern.info kernel: [   38.332471] iwlwifi 0000:02:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 0.59.2.22
Fri Dec  3 11:41:43 2021 kern.info kernel: [   38.350148] iwlwifi 0000:02:00.0: loaded firmware version 62.49eeb572.0 ty-a0-gf-a0-62.ucode op_mode iwlmvm
Fri Dec  3 11:41:43 2021 kern.info kernel: [   38.609505] iwlwifi 0000:02:00.0: Detected Intel(R) Wi-Fi 6 AX210 160MHz, REV=0x420
Fri Dec  3 11:41:43 2021 kern.warn kernel: [   38.632143] thermal thermal_zone0: failed to read out thermal zone (-61)
Fri Dec  3 11:41:43 2021 kern.info kernel: [   38.842530] iwlwifi 0000:02:00.0: loaded PNVM version 0x324cd670
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.101542] iwlwifi 0000:02:00.0: Timeout waiting for PNVM load!
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.107532] iwlwifi 0000:02:00.0: Failed to start RT ucode: -145
Fri Dec  3 11:41:43 2021 kern.warn kernel: [   39.113551] iwlwifi 0000:02:00.0: WRT: Collecting data: ini trigger 13 fired (delay=0ms).
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.122906] iwlwifi 0000:02:00.0: Start IWL Error Log Dump:
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.128461] iwlwifi 0000:02:00.0: Transport status: 0x00000042, valid: 6
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.135156] iwlwifi 0000:02:00.0: Loaded firmware version: 62.49eeb572.0 ty-a0-gf-a0-62.ucode
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.143675] iwlwifi 0000:02:00.0: 0x00000084 | NMI_INTERRUPT_UNKNOWN
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.150608] iwlwifi 0000:02:00.0: 0x002002F0 | trm_hw_status0
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.156350] iwlwifi 0000:02:00.0: 0x00000000 | trm_hw_status1
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.162096] iwlwifi 0000:02:00.0: 0x004DA02C | branchlink2
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.167561] iwlwifi 0000:02:00.0: 0x004D070E | interruptlink1
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.173303] iwlwifi 0000:02:00.0: 0x004D070E | interruptlink2
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.179027] iwlwifi 0000:02:00.0: 0x004D8E06 | data1
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.183993] iwlwifi 0000:02:00.0: 0x01000000 | data2
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.188940] iwlwifi 0000:02:00.0: 0x00000000 | data3
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.193903] iwlwifi 0000:02:00.0: 0x00000000 | beacon time
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.199369] iwlwifi 0000:02:00.0: 0x0005930E | tsf low
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.204508] iwlwifi 0000:02:00.0: 0x00000000 | tsf hi
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.209538] iwlwifi 0000:02:00.0: 0x00000000 | time gp1
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.214763] iwlwifi 0000:02:00.0: 0x0006E2C5 | time gp2
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.219969] iwlwifi 0000:02:00.0: 0x00000001 | uCode revision type
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.226145] iwlwifi 0000:02:00.0: 0x0000003E | uCode version major
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.232331] iwlwifi 0000:02:00.0: 0x49EEB572 | uCode version minor
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.238487] iwlwifi 0000:02:00.0: 0x00000420 | hw version
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.243887] iwlwifi 0000:02:00.0: 0x00489002 | board version
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.249525] iwlwifi 0000:02:00.0: 0x8006FC03 | hcmd
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.254406] iwlwifi 0000:02:00.0: 0x00020000 | isr0
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.259265] iwlwifi 0000:02:00.0: 0x00000000 | isr1
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.264145] iwlwifi 0000:02:00.0: 0x48F00002 | isr2
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.269003] iwlwifi 0000:02:00.0: 0x00C0001C | isr3
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.273882] iwlwifi 0000:02:00.0: 0x00000000 | isr4
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.278743] iwlwifi 0000:02:00.0: 0x00000000 | last cmd Id
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.284228] iwlwifi 0000:02:00.0: 0x004D8E06 | wait_event
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.289606] iwlwifi 0000:02:00.0: 0x00000000 | l2p_control
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.295091] iwlwifi 0000:02:00.0: 0x00000020 | l2p_duration
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.300642] iwlwifi 0000:02:00.0: 0x00000000 | l2p_mhvalid
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.306127] iwlwifi 0000:02:00.0: 0x00000000 | l2p_addr_match
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.311874] iwlwifi 0000:02:00.0: 0x00000009 | lmpm_pmg_sel
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.317426] iwlwifi 0000:02:00.0: 0x00000000 | timestamp
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.322736] iwlwifi 0000:02:00.0: 0x00000020 | flow_handler
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.328350] iwlwifi 0000:02:00.0: Start IWL Error Log Dump:
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.333923] iwlwifi 0000:02:00.0: Transport status: 0x00000042, valid: 7
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.340599] iwlwifi 0000:02:00.0: 0x20000066 | NMI_INTERRUPT_HOST
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.346690] iwlwifi 0000:02:00.0: 0x00000000 | umac branchlink1
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.352614] iwlwifi 0000:02:00.0: 0x8045C7E4 | umac branchlink2
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.358512] iwlwifi 0000:02:00.0: 0x8047E400 | umac interruptlink1
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.364695] iwlwifi 0000:02:00.0: 0x8047E400 | umac interruptlink2
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.370850] iwlwifi 0000:02:00.0: 0x01000000 | umac data1
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.376249] iwlwifi 0000:02:00.0: 0x8047E400 | umac data2
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.381650] iwlwifi 0000:02:00.0: 0x00000000 | umac data3
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.387028] iwlwifi 0000:02:00.0: 0x0000003E | umac major
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.392428] iwlwifi 0000:02:00.0: 0x49EEB572 | umac minor
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.397808] iwlwifi 0000:02:00.0: 0x0006E3FC | frame pointer
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.403477] iwlwifi 0000:02:00.0: 0xC0886264 | stack pointer
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.409116] iwlwifi 0000:02:00.0: 0x00000000 | last host cmd
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.414772] iwlwifi 0000:02:00.0: 0x00000400 | isr status reg
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.420514] iwlwifi 0000:02:00.0: IML/ROM dump:
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.425049] iwlwifi 0000:02:00.0: 0x00000B03 | IML/ROM error/state
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.431221] iwlwifi 0000:02:00.0: 0x00008818 | IML/ROM data1
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.436898] iwlwifi 0000:02:00.0: 0x00000080 | IML/ROM WFPM_AUTH_KEY_0
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.443429] iwlwifi 0000:02:00.0: Fseq Registers:
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.448123] iwlwifi 0000:02:00.0: 0x20000000 | FSEQ_ERROR_CODE
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.453959] iwlwifi 0000:02:00.0: 0x80440002 | FSEQ_TOP_INIT_VERSION
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.460292] iwlwifi 0000:02:00.0: 0x00070008 | FSEQ_CNVIO_INIT_VERSION
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.466818] iwlwifi 0000:02:00.0: 0x0000A652 | FSEQ_OTP_VERSION
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.472741] iwlwifi 0000:02:00.0: 0x00000002 | FSEQ_TOP_CONTENT_VERSION
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.479333] iwlwifi 0000:02:00.0: 0x4552414E | FSEQ_ALIVE_TOKEN
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.485259] iwlwifi 0000:02:00.0: 0x00400410 | FSEQ_CNVI_ID
[... cut ...]
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.496418] iwlwifi 0000:02:00.0: 0x00400410 | CNVI_AUX_MISC_CHIP
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.502522] iwlwifi 0000:02:00.0: 0x00400410 | CNVR_AUX_MISC_CHIP
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.508597] iwlwifi 0000:02:00.0: 0x00009061 | CNVR_SCU_SD_REGS_SD_REG_DIG_DCDC_VTRIM
Fri Dec  3 11:41:43 2021 kern.err kernel: [   39.516426] iwlwifi 0000:02:00.0: 0x00000061 | CNVR_SCU_SD_REGS_SD_REG_ACTIVE_VDIG_MIRROR
Fri Dec  3 11:41:43 2021 kern.err kernel: [   40.581616] iwlwifi 0000:02:00.0: Failed to run INIT ucode: -145
```

BKPepe, 2021-12-04:
Dear reporter,

You created an issue in the  ``packages``  repository, and from your description, this behavior should be reported elsewhere.

Would you please follow the guideline for [reporting bugs](https://openwrt.org/bugs)? It would be highly appreciated as kernel and iwlwifi come from the core repository. This could be checked by browsing the source code here.

Don't hesitate to reach our community on [the forum](https://forum.openwrt.org/).

In any case, I need to close this report.

closed 2021-12-04 by BKPepe

jpsollie, 2021-12-04:
sorry for that, my username at bugs.openwrt.org is linked to my github account, but I do not get access when trying to log in there.
Could anyone please push this issue to bugs.openwrt.org?