openwrt/packages #17269: miniupnpd: not working on master

coolmaxter, 2021-12-04:
Hi， I cannot mapping port anymore on the master after I upgrade from 19.08, here is my config and log
### Version
```
Linux OpenWrt 5.4.162 #0 SMP Wed Sep 2 02:42:00 2020 mips GNU/Linux
root@OpenWrt:~# miniupnpd --version
miniupnpd 2.2.2 master-2a002c7 Sep  2 2020
using netfilter(iptables) backend
```
### miniupnp config:
```
firewall.miniupnpd=include
firewall.miniupnpd.type='script'
firewall.miniupnpd.path='/usr/share/miniupnpd/firewall.include'
firewall.miniupnpd.family='any'
firewall.miniupnpd.reload='1'
ucitrack.@firewall[0].affects='luci-splash' 'qos' 'miniupnpd'
ucitrack.@upnpd[0]=upnpd
ucitrack.@upnpd[0].init='miniupnpd'
upnpd.config=upnpd
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.enabled='1'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
```
### Log:
```
Sat Dec  4 22:49:48 2021 user.notice firewall: Reloading firewall due to ifup of wan (wan)
Sat Dec  4 22:49:49 2021 daemon.notice miniupnpd[3440]: shutting down MiniUPnPd
Sat Dec  4 22:49:49 2021 daemon.notice miniupnpd[5242]: HTTP listening on port 5000
Sat Dec  4 22:49:49 2021 daemon.warn miniupnpd[5242]: no HTTP IPv6 address, disabling IPv6
Sat Dec  4 22:49:49 2021 daemon.notice miniupnpd[5242]: Listening for NAT-PMP/PCP traffic on port 5351
Sat Dec  4 23:20:24 2021 daemon.err miniupnpd[6265]: PCP MAP: failed to add mapping UDP 5354->192.168.100.180:5353 'PCP MAP 87b46ec752910b1f6f368a35'
Sat Dec  4 23:20:57 2021 daemon.err miniupnpd[6265]: Failed to add NAT-PMP 43100 tcp->192.168.100.100:43100 'NAT-PMP 43100 tcp'
Sat Dec  4 23:20:57 2021 daemon.err miniupnpd[6265]: Failed to add NAT-PMP 43100 udp->192.168.100.100:43100 'NAT-PMP 43100 udp'
Sat Dec  4 23:21:24 2021 daemon.err miniupnpd[6265]: Failed to add NAT-PMP 43100 tcp->192.168.100.100:43100 'NAT-PMP 43100 tcp'
Sat Dec  4 23:21:24 2021 daemon.err miniupnpd[6265]: Failed to add NAT-PMP 43100 udp->192.168.100.100:43100 'NAT-PMP 43100 udp'
```

professor-jonny, 2021-12-05:
there is a v1 and v2 to support idgv1 are you building the correct version in 21.02 ?

v1 works with windows the xbox etc... where the newer version wont

stintel, 2022-01-07:
Try version 2.2.3. If the problem persists there, run miniupnpd with -vv and post the output.

tiagogaspar8, 2022-08-25:
Hi @coolmaxter, can you test with the latest version recently committed?

closed 2025-11-15 by GeorgeSapkin