openwrt/packages #17280: busybox: inetd compile requires rpc/rpc.h header

jpsollie, 2021-12-07:
# Issue template (remove lines from top till here)

Maintainer: (not found in Makefile)
Environment: mips, rbm33g, snapshot
Description:
when compiling busybox with inetd remote support, it needs a rpc/rpc.h header which is not installed by default.
In order to fix this, libtirpc dependency is required, as well as adding -I$(STAGING_DIR)/usr/include/tirpc to the busybox CFLAGS, as well as adding -ltirpc to LDLIBS

mhei, 2022-01-17:
Difficult. The busybox package is part of [openwrt,](https://github.com/openwrt/openwrt) not packages. libtirpc, on the other hand, is part of packages. One would have to add a dependency for busybox to libtirpc, at least a conditional one. Please report this issue on [openwrt,](https://github.com/openwrt/openwrt) and ask howto proceed. Please expect that people will ask for your actual use-case, so you should elaborate on it.
I tend to mark this here as invalid or wontfix and to close this issue since we cannot do here very much.

nanfengnan1, 2024-09-04:
so? how to resolve this issue in redhat?