openwrt/packages #17284: modemmanager: unable to start service on mips_24kc

feckert, 2021-12-07:
Maintainer: @aleksander0m  @nickberry17
Environment: lantiq_xrx200, latest OpenWrt master

Description:

I do not know exactly what the problem here is but I have two targets an lantiq_xrx200 device and an x86_64 (APU3) device.
I am using the same ModemManager version 1.18.2 but on my lantiq_xrx200 (this is mips_24kc target) the ModemManger could not start. I get the following messages in the logs

```
Tue Dec  7 17:02:19 2021 daemon.info [17068]: <info>  [1638892939.268189] ModemManager (version 1.18.2) starting in system bus...
Tue Dec  7 17:02:19 2021 daemon.debug [17068]: <debug> [1638892939.269515] [charsets] detecting platform iconv() support...
Tue Dec  7 17:02:19 2021 daemon.debug [17068]: <debug> [1638892939.270638] [charsets]   UTF-8: iconv conversion to/from charset is supported
Tue Dec  7 17:02:19 2021 daemon.debug [17068]: <debug> [1638892939.271902] [charsets]   UCS-2BE: iconv conversion to charset not supported
Tue Dec  7 17:02:19 2021 daemon.debug [17068]: <debug> [1638892939.273176] [charsets]   ASCII: iconv conversion to/from charset is supported
Tue Dec  7 17:02:19 2021 daemon.debug [17068]: <debug> [1638892939.274376] [charsets]   ISO8859-1: iconv conversion to charset not supported
Tue Dec  7 17:02:19 2021 daemon.debug [17068]: <debug> [1638892939.275581] [charsets]   CP437: iconv conversion to charset not supported
Tue Dec  7 17:02:19 2021 daemon.debug [17068]: <debug> [1638892939.276763] [charsets]   CP850: iconv conversion to charset not supported
Tue Dec  7 17:02:19 2021 daemon.debug [17068]: <debug> [1638892939.277928] [charsets]   UTF-16BE: iconv conversion to/from charset is supported
Tue Dec  7 17:02:19 2021 daemon.warn [17068]: <warn>  [1638892939.310430] could not acquire the 'org.freedesktop.ModemManager1' service name
Tue Dec  7 17:02:19 2021 daemon.info [17068]: <info>  [1638892939.311374] ModemManager is shut down
```
Can anyone confirm this? Could it be due to the different endianness of the processors?

- x86_64 little endieanness
- mips_24kc big endianness

That would be the only difference I can think of at the moment from a software point of view why the modem manager behaves differently?

aleksander0m, 2021-12-09:
Not sure I have a BE system around to test this myself :/
We usually see that "could not acquire" message when there is already another process owning the same well-known DBus name. In this case, though, it could be a different problem.

nickberry17, 2021-12-09:
Sorry, I don't know; works on all my devices (fresh OpenWrt).  Has your local installation replaced any files in the /files directory? If so, could you please replace them with the current ones?

feckert, 2021-12-10:
> Sorry, I don't know; works on all my devices (fresh OpenWrt). Has your local installation replaced any files in the /files directory? If so, could you please replace them with the current ones?

@nickberry17 I have compiled the last master and added the modem manager to my image via the image build. I have not changed anything.

> Not sure I have a BE system around to test this myself :/ We usually see that "could not acquire" message when there is already another process owning the same well-known DBus name. In this case, though, it could be a different problem.

 @aleksander0m When I stop the dbus service on the working target x86_64 APU3, I get the same error.
So I think this has something todo with the dbus. I checked again to see if there is another service running, but this is not the case. The modemmanager I want to start is the only.

The error message seems to come from this line https://github.com/freedesktop/ModemManager/blob/1.18.2/src/main.c#L132

Output of my lantiq_xrx200 target
```
root@VR2-106149 ~ # dbus-send --system --dest=org.freedesktop.DBus --type=method_call --print-reply /org/freedesktop/DBus org.freedeskt
op.DBus.ListNames
method return time=1639125179.137379 sender=org.freedesktop.DBus -> destination=:1.1 serial=3 reply_serial=2
   array [
      string "org.freedesktop.DBus"
      string ":1.1"
   ]
```
As you can see there is no modemmanager object listed but the modemamanger complains with this error:
`Fri Dec 10 08:57:16 2021 daemon.warn [11782]: <warn>  could not acquire the 'org.freedesktop.ModemManager1' service `

Output of my x86_64 target:
```
root@st-dev-07 ~ # dbus-send --system --dest=org.freedesktop.DBus --type=method_call --print-reply /org/freedesktop/DBus org.freedesktop.DBus.ListNames
method return time=1639125222.123352 sender=org.freedesktop.DBus -> destination=:1.120 serial=3 reply_serial=2
   array [
      string "org.freedesktop.DBus"
      string ":1.120"
      string ":1.0"
      string "org.freedesktop.ModemManager1"
   ]

```

aleksander0m, 2021-12-10:
> As you can see there is no modemmanager object listed but the modemmanager complains with this error

Yeah, that's why I said that this could be a completely different problem. The info we get back on failures trying to acquire the name is just "you couldn't", without explanation of why it really failed... I'll try to test this myself in a BE system, I think I have one somewhere.

feckert, 2021-12-10:
@aleksander0m I have now compiled the dbus with the compile option `DBUS_ENABLE_VERBOSE_MODE=ON` on my xrx200_lantiq (mips_24kc) target and saved the log when I start the `ModemManager`.

There you can see the output of ModemManager and the log output of the dbus.
[ModemManager-dbus.log](https://github.com/openwrt/packages/files/7693072/ModemManager-dbus.log)

In the log I find these lines interesting:
```
Fri Dec 10 14:19:25 2021 daemon.err dbus-daemon[24852]: 24852: 0x77e61d64: 1639142365.720398 [dbus/dbus-message.c(4711):_dbus_message_loader_queue_messages] Initial peek at header says we don't have a whole message yet, or data broken with invalid code 17
Fri Dec 10 14:19:25 2021 daemon.err dbus-daemon[24852]: 24852: 0x77e61d64: 1639142365.720514 [dbus/dbus-transport.c(1205):_dbus_transport_queue_messages] Corrupted message stream, disconnecting
```

feckert, 2021-12-16:
@aleksander0m @nickberry17 I just can't get the ModemManager to work on my lantiq_xrx200 target!
I suspect that something is wrong with glib2 abstraction.
Because when I use the native dbus api in my test program, it can get the name and it works.

I am not a glib2 expert, but is there a way or does anyone have a hint or link how to compile a small test program with glib2 and dbus and then see if it works! I would like to narrow down the problem whether it is the glib2 or the ModemManger itself.

aleksander0m, 2021-12-16:
> am not a glib2 expert, but is there a way or does anyone have a hint or link how to compile a small test program with glib2 and dbus and then see if it works 

You can get the "service-alarm.c" program along with the "es.aleksander.Alarm.xml" file from here:
https://aleksander.es/data/GNOMEASIA2014-Examples/

It is an extremely minimal example of a DBus service using glib2 and GDBus, which is what ModemManager does.

And set it up to compile just like this:
```
gdbus-codegen \
    --interface-prefix es.aleksander \
    --generate-c-code alarm \
    es.aleksander.Alarm.xml

gcc -o service-alarm \
    service-alarm.c alarm.h alarm.c \
    `pkg-config --cflags --libs gio-2.0 gio-unix-2.0`
```

gdbus-codegen is a tool that gets installed along with glib. It's a python program, so you can run the one installed in staging, same as the MM compilation does.

feckert, 2021-12-17:
@aleksander0m Thank you for this small snipped.
I have now [packaged](https://github.com/TDT-AG/packages/tree/example/20211217-service-alarm) it for Openwrt and run it on my xrx200 and on my x86_64 (APU3)

It looks like it's not the ModemManager but either the glib2 or the dbus.
But since the dbus works with another small programm, I think it can only be glib2!

Enclosed is my test summary. I think that's all I can do.
I can now run it again on another big-endiann system.
I just find it strange that I am the only one who has this problem.
Others should have the same problem as well.

**lantiq_xrx200 not working:**

Terminal with dbus-monitor log:
```
root@VR2-106149 ~ # dbus-monitor --system
signal time=1639752484.770872 sender=org.freedesktop.DBus -> destination=:1.0 serial=2 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameAcquired
   string ":1.0"
signal time=1639752484.771295 sender=org.freedesktop.DBus -> destination=:1.0 serial=4 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameLost
   string ":1.0"
```
 **Terminal to start the service (returns immediately):**
```
root@VR2-106149 ~ # ./service-alarm
root@VR2-106149 ~ #
```
**Terminal with syslog:**
```
Fri Dec 17 15:48:04 2021 daemon.info dbus-daemon[449]: [system] Connection :1.0 (uid=0 pid=632 comm="dbus-monitor --system") became a monitor.
Fri Dec 17 15:48:04 2021 daemon.err dbus-daemon[449]: dbus-daemon[449]: [system] Connection :1.0 (uid=0 pid=632 comm="dbus-monitor --system") became a monitor.
Fri Dec 17 15:48:15 2021 user.crit : Could not acquire service name
```
### x86_64  working:
**Terminal with dbus-monitor log:**
```
root@st-dev-07 ~ # dbus-monitor --system
signal time=1639752459.729337 sender=org.freedesktop.DBus -> destination=:1.3 serial=2 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameAcquired
   string ":1.3"
signal time=1639752459.729447 sender=org.freedesktop.DBus -> destination=:1.3 serial=4 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameLost
   string ":1.3"
method call time=1639752463.034993 sender=:1.4 -> destination=org.freedesktop.DBus serial=1 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=Hello
method return time=1639752463.035084 sender=org.freedesktop.DBus -> destination=:1.4 serial=1 reply_serial=1
   string ":1.4"
signal time=1639752463.035159 sender=org.freedesktop.DBus -> destination=(null destination) serial=5 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameOwnerChanged
   string ":1.4"
   string ""
   string ":1.4"
signal time=1639752463.035245 sender=org.freedesktop.DBus -> destination=:1.4 serial=2 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameAcquired
   string ":1.4"
method call time=1639752463.037329 sender=:1.4 -> destination=org.freedesktop.DBus serial=2 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=RequestName
   string "es.aleksander.Alarm"
   uint32 0
signal time=1639752463.037408 sender=org.freedesktop.DBus -> destination=(null destination) serial=6 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameOwnerChanged
   string "es.aleksander.Alarm"
   string ""
   string ":1.4"
signal time=1639752463.037470 sender=org.freedesktop.DBus -> destination=:1.4 serial=3 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameAcquired
   string "es.aleksander.Alarm"
method return time=1639752463.037540 sender=org.freedesktop.DBus -> destination=:1.4 serial=4 reply_serial=2
   uint32 1
```
 **Terminal to start the service (running):**
```
root@st-dev-07 ~ # ./service-alarm

```

**Terminal with syslog:**
```
Fri Dec 17 16:04:02 2021 daemon.info dbus-daemon[2411]: [system] Connection :1.5 (uid=0 pid=2740 comm="dbus-monitor --system") became a monitor.
Fri Dec 17 16:04:02 2021 daemon.err dbus-daemon[2411]: dbus-daemon[2411]: [system] Connection :1.5 (uid=0 pid=2740 comm="dbus-monitor --system") became a monitor.
```

arkanoid87, 2022-02-01:
I'm facing same issue with glinet_gl-x300b

firmware just built snapshot with `openwrt-imagebuilder-ath79-generic.Linux-x86_64`:
```
make image PROFILE="glinet_gl-x300b" PACKAGES="kmod-usb2 modemmanager"                                                                                                                                                           
```

```
# service modemmanager stop
# ModemManager --debug
[9533]: <info>  [1643689422.807563] ModemManager (version 1.18.2) starting in system bus...
[9533]: <debug> [1643689422.811767] [charsets] detecting platform iconv() support...
[9533]: <debug> [1643689422.815685] [charsets]   UTF-8: iconv conversion to/from charset is supported
[9533]: <debug> [1643689422.820474] [charsets]   UCS-2BE: iconv conversion to charset not supported
[9533]: <debug> [1643689422.823622] [charsets]   ASCII: iconv conversion to/from charset is supported
[9533]: <debug> [1643689422.824706] [charsets]   ISO8859-1: iconv conversion to charset not supported
[9533]: <debug> [1643689422.825771] [charsets]   CP437: iconv conversion to charset not supported
[9533]: <debug> [1643689422.826895] [charsets]   CP850: iconv conversion to charset not supported
[9533]: <debug> [1643689422.831176] [charsets]   UTF-16BE: iconv conversion to/from charset is supported
[9533]: <warn>  [1643689422.856645] could not acquire the 'org.freedesktop.ModemManager1' service name
[9533]: <info>  [1643689422.857113] ModemManager is shut down
```

arkanoid87, 2022-02-02:
Thanks to @lkundrak we've reduced the issue to:
- dbus-daemon answers little endian messages
- dbus-daemon closes the connection upon getting a big-endian Hello right away
- dbus-send utility sends little endian messages on big endian device
- ModemManager sends big endian messages

AFAIK there is an endianess compatibility layer in glib, but apparently something is not working correctly

How to reproduce:
```
opkg update
opkg install modemmanager dbus-utils strace
service modemmanager stop
```

in terminal 1 run `dbus-monitor --system`
in terminal 2 run `ModemManager --debug`
see that no output is generated in dbus-monitor

### diving in by listening messages sent to dbus socket via strace:

note that if the message starts with **l** is little endian, with **B** is big endian

#### test case with dbus-send

terminal 1 run
```
strace -s4096 -f -e trace=sendmsg,recvmsg -p `pidof dbus-daemon`
```

terminal 2 run:
```
strace -f -e sendmsg -e trace=sendmsg,recvmsg -s4096 dbus-send --system --print-reply --type=method_call --dest=org.freedesktop.DBus /org/freedesktop/DBus org.freedesktop.DBus.Hello
```

terminal 1 output:
```
recvmsg(8, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0", iov_len=1}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 1
recvmsg(8, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="l\1\0\1\0\0\0\0\0\0\0\1\0\0\0n\1\1o\0\0\0\0\25/org/freedesktop/DBus\0\0\0\6\1s\0\0\0\0\24org.freedesktop.DBus\0\0\0\0\2\1s\0\0\0\0\24org.freedesktop.DBus\0\0\0\0\3\1s\0\0\0\0\5Hello\0\0\0", iov_len=2048}], msg_iovlen=1, msg_controllen=0, msg_flags=MSG_CMSG_CLOEXEC}, MSG_CMSG_CLOEXEC) = 128
recvmsg(8, {msg_namelen=0}, MSG_CMSG_CLOEXEC) = -1 EAGAIN (Resource temporarily unavailable)
sendmsg(8, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="l\2\1\1\0\0\0\n\0\0\0\1\0\0\0=\6\1s\0\0\0\0\5:1.12\0\0\0\5\1u\0\0\0\0\1\10\1g\0\1s\0\0\7\1s\0\0\0\0\24org.freedesktop.DBus\0\0\0\0", iov_len=80}, {iov_base="\0\0\0\5:1.12\0", iov_len=10}], msg_iovlen=2, msg_controllen=0, msg_flags=0}, 0) = 90
sendmsg(8, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="l\4\1\1\0\0\0\n\0\0\0\2\0\0\0\215\1\1o\0\0\0\0\25/org/freedesktop/DBus\0\0\0\2\1s\0\0\0\0\24org.freedesktop.DBus\0\0\0\0\3\1s\0\0\0\0\fNameAcquired\0\0\0\0\6\1s\0\0\0\0\5:1.12\0\0\0\10\1g\0\1s\0\0\7\1s\0\0\0\0\24org.freedesktop.DBus\0\0\0\0", iov_len=160}, {iov_base="\0\0\0\5:1.12\0", iov_len=10}], msg_iovlen=2, msg_controllen=0, msg_flags=0}, 0) = 170
recvmsg(8, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="l\1\0\1\0\0\0\0\0\0\0\2\0\0\0m\1\1o\0\0\0\0\25/org/freedesktop/DBus\0\0\0\2\1s\0\0\0\0\24org.freedesktop.DBus\0\0\0\0\3\1s\0\0\0\0\5Hello\0\0\0\6\1s\0\0\0\0\24org.freedesktop.DBus\0\0\0\0", iov_len=2048}], msg_iovlen=1, msg_controllen=0, msg_flags=MSG_CMSG_CLOEXEC}, MSG_CMSG_CLOEXEC) = 128
recvmsg(8, {msg_namelen=0}, MSG_CMSG_CLOEXEC) = -1 EAGAIN (Resource temporarily unavailable)
sendmsg(8, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="l\3\1\1\0\0\0%\0\0\0\3\0\0\0m\6\1s\0\0\0\0\5:1.12\0\0\0\4\1s\0\0\0\0!org.freedesktop.DBus.Error.Failed\0\0\0\0\0\0\0\5\1u\0\0\0\0\2\10\1g\0\1s\0\0\7\1s\0\0\0\0\24org.freedesktop.DBus\0\0\0\0", iov_len=128}, {iov_base="\0\0\0 Already handled an Hello message\0", iov_len=37}], msg_iovlen=2, msg_controllen=0, msg_flags=0}, 0) = 165
recvmsg(8, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="", iov_len=2048}], msg_iovlen=1, msg_controllen=0, msg_flags=MSG_CMSG_CLOEXEC}, MSG_CMSG_CLOEXEC) = 0
```

terminal 2 output:
```
sendmsg(3, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="l\1\0\1\0\0\0\0\0\0\0\1\0\0\0n\1\1o\0\0\0\0\25/org/freedesktop/DBus\0\0\0\6\1s\0\0\0\0\24org.freedesktop.DBus\0\0\0\0\2\1s\0\0\0\0\24org.freedesktop.DBus\0\0\0\0\3\1s\0\0\0\0\5Hello\0\0\0", iov_len=128}, {iov_base="", iov_len=0}], msg_iovlen=2, msg_controllen=0, msg_flags=0}, 0) = 128
recvmsg(3, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="l\2\1\1\0\0\0\n\0\0\0\1\0\0\0=\6\1s\0\0\0\0\5:1.12\0\0\0\5\1u\0\0\0\0\1\10\1g\0\1s\0\0\7\1s\0\0\0\0\24org.freedesktop.DBus\0\0\0\0\0\0\0\5:1.12\0", iov_len=2048}], msg_iovlen=1, msg_controllen=0, msg_flags=MSG_CMSG_CLOEXEC}, MSG_CMSG_CLOEXEC) = 90
recvmsg(3, {msg_namelen=0}, MSG_CMSG_CLOEXEC) = -1 EAGAIN (Resource temporarily unavailable)
sendmsg(3, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="l\1\0\1\0\0\0\0\0\0\0\2\0\0\0m\1\1o\0\0\0\0\25/org/freedesktop/DBus\0\0\0\2\1s\0\0\0\0\24org.freedesktop.DBus\0\0\0\0\3\1s\0\0\0\0\5Hello\0\0\0\6\1s\0\0\0\0\24org.freedesktop.DBus\0\0\0\0", iov_len=128}, {iov_base="", iov_len=0}], msg_iovlen=2, msg_controllen=0, msg_flags=0}, 0) = 128
recvmsg(3, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="l\4\1\1\0\0\0\n\0\0\0\2\0\0\0\215\1\1o\0\0\0\0\25/org/freedesktop/DBus\0\0\0\2\1s\0\0\0\0\24org.freedesktop.DBus\0\0\0\0\3\1s\0\0\0\0\fNameAcquired\0\0\0\0\6\1s\0\0\0\0\5:1.12\0\0\0\10\1g\0\1s\0\0\7\1s\0\0\0\0\24org.freedesktop.DBus\0\0\0\0\0\0\0\5:1.12\0", iov_len=2048}], msg_iovlen=1, msg_controllen=0, msg_flags=MSG_CMSG_CLOEXEC}, MSG_CMSG_CLOEXEC) = 170
recvmsg(3, {msg_namelen=0}, MSG_CMSG_CLOEXEC) = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(3, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="l\3\1\1\0\0\0%\0\0\0\3\0\0\0m\6\1s\0\0\0\0\5:1.12\0\0\0\4\1s\0\0\0\0!org.freedesktop.DBus.Error.Failed\0\0\0\0\0\0\0\5\1u\0\0\0\0\2\10\1g\0\1s\0\0\7\1s\0\0\0\0\24org.freedesktop.DBus\0\0\0\0\0\0\0 Already handled an Hello message\0", iov_len=2048}], msg_iovlen=1, msg_controllen=0, msg_flags=MSG_CMSG_CLOEXEC}, MSG_CMSG_CLOEXEC) = 165
[... cut ...]
[pid 24752] sendmsg(6, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="B\1\0\1\0\0\0\0\0\0\0\1\0\0\0n\1\1o\0\0\0\0\25/org/freedesktop/DBus\0\0\0\2\1s\0\0\0\0\24org.freedesktop.DBus\0\0\0\0\6\1s\0\0\0\0\24org.freedesktop.DBus\0\0\0\0\3\1s\0\0\0\0\5Hello\0\0\0", iov_len=128}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, MSG_NOSIGNAL) = 128
[pid 24752] recvmsg(6, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="", iov_len=16}], msg_iovlen=1, msg_controllen=0, msg_flags=MSG_CMSG_CLOEXEC}, MSG_CMSG_CLOEXEC) = 0
```

arkanoid87, 2022-02-02:
I've tried compiling glib2 with the proposed (quick & dirty) patch from @lkundrak

```patch
diff --git a/gio/gdbusmessage.c b/gio/gdbusmessage.c
index cc8f37e5d..5d9610534 100644
--- a/gio/gdbusmessage.c
+++ b/gio/gdbusmessage.c
@@ -587,21 +587,21 @@ g_dbus_message_class_init (GDBusMessageClass *klass)
 
 static void
 g_dbus_message_init (GDBusMessage *message)
 {
   /* Any D-Bus implementation is supposed to handle both Big and
    * Little Endian encodings and the Endianness is part of the D-Bus
    * message - we prefer to use Big Endian (since it's Network Byte
    * Order and just easier to read for humans) but if the machine is
    * Little Endian we use that for performance reasons.
    */
-#if G_BYTE_ORDER == G_LITTLE_ENDIAN
+#if 1 || G_BYTE_ORDER == G_LITTLE_ENDIAN
   message->byte_order = G_DBUS_MESSAGE_BYTE_ORDER_LITTLE_ENDIAN;
 #else
   /* this could also be G_PDP_ENDIAN */
   message->byte_order = G_DBUS_MESSAGE_BYTE_ORDER_BIG_ENDIAN;
 #endif
   message->headers = g_hash_table_new_full (g_direct_hash,
                                             g_direct_equal,
                                             NULL,
                                             (GDestroyNotify) g_variant_unref);
 }
```

Now modemmanager sends little endian `Hello` message instead of big endian, but dbus refuses to answer and closes the socket just like previous big endian case.

```
strace -s4096 -f -e trace=sendmsg,recvmsg -p `pidof dbus-daemon`
```
```
recvmsg(8, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0", iov_len=1}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 1
recvmsg(8, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="l\1\0\1\0\0\0\0\1\0\0\0n\0\0\0\1\1o\0\25\0\0\0/org/freedesktop/DBus\0\0\0\2\1s\0\24\0\0\0org.freedesktop.DBus\0\0\0\0\6\1s\0\24\0\0\0org.freedesktop.DBus\0\0\0\0\3\1s\0\5\0\0\0Hello\0\0\0", iov_len=2048}], msg_iovlen=1, msg_controllen=0, msg_flags=MSG_CMSG_CLOEXEC}, MSG_CMSG_CLOEXEC) = 128
```

```
strace -f -e trace=sendmsg,recvmsg -s4096 ModemManager
```

```
[pid  3920] sendmsg(6, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0", iov_len=1}], msg_iovlen=1, msg_control=[{cmsg_len=24, cmsg_level=SOL_SOCKET, cmsg_type=SCM_CREDENTIALS, cmsg_data={pid=3918, uid=0, gid=0}}], msg_controllen=24, msg_flags=0}, MSG_NOSIGNAL) = 1
strace: Process 3921 attached
[pid  3921] sendmsg(6, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="l\1\0\1\0\0\0\0\1\0\0\0n\0\0\0\1\1o\0\25\0\0\0/org/freedesktop/DBus\0\0\0\2\1s\0\24\0\0\0org.freedesktop.DBus\0\0\0\0\6\1s\0\24\0\0\0org.freedesktop.DBus\0\0\0\0\3\1s\0\5\0\0\0Hello\0\0\0", iov_len=128}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, MSG_NOSIGNAL) = 128
[pid  3921] recvmsg(6, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="", iov_len=16}], msg_iovlen=1, msg_controllen=0, msg_flags=MSG_CMSG_CLOEXEC}, MSG_CMSG_CLOEXEC) = 0
```

aleksander0m, 2022-02-02:
> dbus-daemon answers little endian messages
>     dbus-daemon closes the connection upon getting a big-endian Hello right away
>     dbus-send utility sends little endian messages on big endian [...]

If the system is BE, shouldn't dbus-daemon expect BE and dbus-send utility send BE? I think the error could be in how the dbus daemon, libs and utils are built, maybe they're not correctly detecting endianness during the build for some reason?

If the system is BE, and MM/glib sends BE, that looks fine to me.

Or am I missing something? Is this system not BE?

aleksander0m, 2022-02-02:
>  But since the dbus works with another small programm, I think it can only be glib2!

What other small program makes it work? please note that dbus-send is likely built along the dbus-daemon itself, they both link to libdbus. If there's an issue here with the endianess expected by both those tools, it may very well be the same single issue.

arkanoid87, 2022-02-02:
> If the system is BE, shouldn't dbus-daemon expect BE and dbus-send utility send BE? I think the error could be in how the dbus daemon, libs and utils are built, maybe they're not correctly detecting [...]

The system I'm testing on is BE
according to https://serverfault.com/questions/163487/how-to-tell-if-a-linux-system-is-big-endian-or-little-endian
I get:
```
# xxd -c 1 -l 6 /bin/ls
00000000: 7f  .
00000001: 45  E
00000002: 4c  L
00000003: 46  F
00000004: 01  .
00000005: 02  .
```

arkanoid87, 2022-02-02:
> What other small program makes it work? please note that dbus-send is likely built along the dbus-daemon itself, they both link to libdbus. If there's an issue here with the endianess expected by [...]

I agree that there should be an all BE or all LE message passing here, but @lkundrak told me there's a compatibility layer inside gnome, but the but may well be there

arkanoid87, 2022-02-02:
We need a gnome dev here

dependency tree is:

modemmanager -> [libc, librt, libpthread, **glib2**, **dbus**, ppp, libmbim, libqmi]
dbus -> [libc, librt, libpthread, libexpat, **libdbus**]
dbus-utils -> [libc, librt, libpthread, **dbus**]
libdbus -> [libc, libpthread]
glib2 -> [libc, zlib, libpthread, libffi, libattr]

follows that if dbus-send sending a *LE* Hello on a *BE* system is wrong, so the problem is in **dbus** or **libdbus**, otherwise if that is accepted the problem may be in **ModemManager** or **glib2**. Yet, still wild guessing here.

aleksander0m, 2022-02-02:
> modemmanager -> [libc, librt, libpthread, glib2, dbus, ppp, libmbim, libqmi]

The dbus dependency for ModemManager is really a runtime dependency, not a build dependency per se. The ModemManager implementation relies exclusively on glib2's GDBus implementation.

arkanoid87, 2022-02-02:
> > if dbus-send sending a LE Hello on a BE system is wrong, so the problem is in dbus or libdbus, otherwise if that is accepted the problem may be in ModemManager or glib2. Yet, still wild guessing here.
> 
> Do you have the verbose configure output of the dbus package in the build log?

I've to enable CONFIG_BUILD_LOG in make config first and rebuild everything. Doing right now.

arkanoid87, 2022-02-03:
After returning to the unpatched glib2, I decoded/unmarshalled manually some message buffers according to the [dbus specification](https://dbus.freedesktop.org/doc/dbus-specification.html).

TL;DR; conclusions at the bottom.

### Ubuntu 20.04 on x86_64 - dbus-send
The result does fit perfectly with specification

`# dbus-send --system --print-reply --type=method_call --dest=org.freedesktop.DBus /org/freedesktop/DBus org.freedesktop.DBus.Hello`
```
# \x6c\x01\x00\x01\x00\x00\x00\x00\x01\x00\x00\x00\x6e\x00\x00\x00\x01\x01\x6f\x00\x15\x00\x00\x00\x2f\x6f\x72\x67\x2f\x66\x72\x65\x65\x64\x65\x73\x6b\x74\x6f\x70\x2f\x44\x42\x75\x73\x00\x00\x00\x06\x01\x73\x00\x14\x00\x00\x00\x6f\x72\x67\x2e\x66\x72\x65\x65\x64\x65\x73\x6b\x74\x6f\x70\x2e\x44\x42\x75\x73\x00\x00\x00\x00\x02\x01\x73\x00\x14\x00\x00\x00\x6f\x72\x67\x2e\x66\x72\x65\x65\x64\x65\x73\x6b\x74\x6f\x70\x2e\x44\x42\x75\x73\x00\x00\x00\x00\x03\x01\x73\x00\x05\x00\x00\x00\x48\x65\x6c\x6c\x6f\x00\x00\x00
\x6c    l [endianness: BYTE]
\x01    METHOD_CALL [message type: BYTE]
\x00    0 [flags: BYTE]
\x01    1 [major protocol version: BYTE]
\x00\x00\x00\x00    0 [body length: UINT32]
\x01\x00\x00\x00    1 [serial: UINT32]
ARRAY
    \x6e\x00\x00\x00    110 [array len: UINT32]
    STRUCT: (BYTE, VARIANT)
        \x01    path [field code: BYTE]
        VARIANT
            \x01                1 [signature len: BYTE]
            \x6f\x00            o [signature: OBJECT_PATH]
            \x15\x00\x00\x00    21 [object path len: UINT32]
            \x2f\x6f\x72\x67\x2f\x66\x72\x65\x65\x64\x65\x73\x6b\x74\x6f\x70\x2f\x44\x42\x75\x73\x00    /org/freedesktop/DBus [object path: OBJECT PATH]
            \x00\x00            padding
    STRUCT: (BYTE, VARIANT)    
        \x06    destination [field code: BYTE]
        VARIANT
            \x01                1 [signature len: BYTE]
            \x73\x00            s [signature: STRING]
            \x14\x00\x00\x00    20 [string len: UINT32]
            \x6f\x72\x67\x2e\x66\x72\x65\x65\x64\x65\x73\x6b\x74\x6f\x70\x2e\x44\x42\x75\x73\x00        org.freedesktop.DBus [string: STRING]
            \x00\x00\x00        padding
    STRUCT: (BYTE, VARIANT)
        \x02    interface [field code: BYTE]
        VARIANT
            \x01                1 [signature len: BYTE]
            \x73\x00            s [signature: STRING]
            \x14\x00\x00\x00    20 [string len: UINT32]
            \x6f\x72\x67\x2e\x66\x72\x65\x65\x64\x65\x73\x6b\x74\x6f\x70\x2e\x44\x42\x75\x73\x00        org.freedesktop.DBus [string: STRING]
            \x00\x00\x00        padding 
    STRUCT: (BYTE, VARIANT)
        \x03    member [field code: BYTE]
        VARIANT
            \x01                1 [signature len: BYTE]
            \x73\x00            s [signature: STRING]
            \x05\x00\x00\x00    5 [string len: UINT32]
            \x48\x65\x6c\x6c\x6f\x00    Hello [string: STRING]
            \x00\x00            padding
```

### OpenWrt on gl-x300b - dbus-send
The first byte of the header says **l** as *little endian*, but all the involved types (only UINT32 in this context) in the buffer are actually laid *big endian*

Somehow, dbus responds correctly to this message with `Error org.freedesktop.DBus.Error.Failed: Already handled an Hello message`

`# dbus-send --system --print-reply --type=method_call --dest=org.freedesktop.DBus /org/freedesktop/DBus org.freedesktop.DBus.Hello`
```
# \x6c\x01\x00\x01\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x6e\x01\x01\x6f\x00\x00\x00\x00\x15\x2f\x6f\x72\x67\x2f\x66\x72\x65\x65\x64\x65\x73\x6b\x74\x6f\x70\x2f\x44\x42\x75\x73\x00\x00\x00\x06\x01\x73\x00\x00\x00\x00\x14\x6f\x72\x67\x2e\x66\x72\x65\x65\x64\x65\x73\x6b\x74\x6f\x70\x2e\x44\x42\x75\x73\x00\x00\x00\x00\x02\x01\x73\x00\x00\x00\x00\x14\x6f\x72\x67\x2e\x66\x72\x65\x65\x64\x65\x73\x6b\x74\x6f\x70\x2e\x44\x42\x75\x73\x00\x00\x00\x00\x03\x01\x73\x00\x00\x00\x00\x05\x48\x65\x6c\x6c\x6f\x00\x00\x00
\x6c    l [endianness: BYTE]
\x01    METHOD_CALL [message type: BYTE]
\x00    0 [flags: BYTE]
\x01    1 [major protocol version: BYTE]
\x00\x00\x00\x00    0 [body length: UINT32]
\x00\x00\x00\x01    1 [serial: UINT32]
ARRAY
    \x00\x00\x00\x6e    110 [array len: UINT32]
    STRUCT: (BYTE, VARIANT)
        \x01    path [field code: BYTE]
[... 137 lines cut ...]
- on gl-x300b dbus-send generates malformed messages and dbus-deamon accepts malformed messages. According to what Mr. @aleksander0m says and what the patch does, ModemManager relies on glib2 to generate dbus messages but dbus-utils uses libdbus, but only the first is formally correct.

raw data:
```
# Ubuntu 20.04 - dbus-send --system --print-reply --type=method_call --dest=org.freedesktop.DBus /org/freedesktop/DBus org.freedesktop.DBus.Hello
l\1\0\1\0\0\0\0\1\0\0\0n\0\0\0\1\1o\0\25\0\0\0/org/freedesktop/DBus\0\0\0\6\1s\0\24\0\0\0org.freedesktop.DBus\0\0\0\0\2\1s\0\24\0\0\0org.freedesktop.DBus\0\0\0\0\3\1s\0\5\0\0\0Hello\0\0\0
\x6c\x01\x00\x01\x00\x00\x00\x00\x01\x00\x00\x00\x6e\x00\x00\x00\x01\x01\x6f\x00\x15\x00\x00\x00\x2f\x6f\x72\x67\x2f\x66\x72\x65\x65\x64\x65\x73\x6b\x74\x6f\x70\x2f\x44\x42\x75\x73\x00\x00\x00\x06\x01\x73\x00\x14\x00\x00\x00\x6f\x72\x67\x2e\x66\x72\x65\x65\x64\x65\x73\x6b\x74\x6f\x70\x2e\x44\x42\x75\x73\x00\x00\x00\x00\x02\x01\x73\x00\x14\x00\x00\x00\x6f\x72\x67\x2e\x66\x72\x65\x65\x64\x65\x73\x6b\x74\x6f\x70\x2e\x44\x42\x75\x73\x00\x00\x00\x00\x03\x01\x73\x00\x05\x00\x00\x00\x48\x65\x6c\x6c\x6f\x00\x00\x00

# Ubuntu 20.04 - ModemManager
[... cut ...]
\x6c\x01\x00\x01\x00\x00\x00\x00\x01\x00\x00\x00\x6e\x00\x00\x00\x01\x01\x6f\x00\x15\x00\x00\x00\x2f\x6f\x72\x67\x2f\x66\x72\x65\x65\x64\x65\x73\x6b\x74\x6f\x70\x2f\x44\x42\x75\x73\x00\x00\x00\x02\x01\x73\x00\x14\x00\x00\x00\x6f\x72\x67\x2e\x66\x72\x65\x65\x64\x65\x73\x6b\x74\x6f\x70\x2e\x44\x42\x75\x73\x00\x00\x00\x00\x06\x01\x73\x00\x14\x00\x00\x00\x6f\x72\x67\x2e\x66\x72\x65\x65\x64\x65\x73\x6b\x74\x6f\x70\x2e\x44\x42\x75\x73\x00\x00\x00\x00\x03\x01\x73\x00\x05\x00\x00\x00\x48\x65\x6c\x6c\x6f\x00\x00\x00

```

arkanoid87, 2022-02-03:
In the meanwhile, I got suggestion from IRC to try compiling dbus adding this line in Makefile:
```
TARGET_CFLAGS += $(if $(CONFIG_BIG_ENDIAN),-DWORDS_BIGENDIAN)
```
just before `CMAKE_OPTIONS`

and I got ModemManager booting on gl-x300b!

```
[5514]: <info>  [1643831166.260534] ModemManager (version 1.18.2) starting in system bus...
[5514]: <debug> [1643831166.265234] [charsets] detecting platform iconv() support...
[5514]: <debug> [1643831166.269787] [charsets]   UTF-8: iconv conversion to/from charset is supported
[5514]: <debug> [1643831166.274431] [charsets]   UCS-2BE: iconv conversion to charset not supported
[5514]: <debug> [1643831166.280683] [charsets]   ASCII: iconv conversion to/from charset is supported
[5514]: <debug> [1643831166.285420] [charsets]   ISO8859-1: iconv conversion to charset not supported
[5514]: <debug> [1643831166.290071] [charsets]   CP437: iconv conversion to charset not supported
[5514]: <debug> [1643831166.294664] [charsets]   CP850: iconv conversion to charset not supported
[5514]: <debug> [1643831166.299296] [charsets]   UTF-16BE: iconv conversion to/from charset is supported
[5514]: <debug> [1643831166.330939] bus acquired, creating manager...
[5514]: <debug> [1643831166.339970] [auth-provider] singleton created
[5514]: <debug> [1643831166.348653] [filter] created
[5514]: <debug> [1643831166.353044] [filter]   explicit whitelist:         yes
[5514]: <debug> [1643831166.357775] [filter]   explicit blacklist:         yes
[5514]: <debug> [1643831166.362374] [filter]   plugin whitelist:           yes
[5514]: <debug> [1643831166.366916] [filter]   qrtr devices allowed:       yes
[5514]: <debug> [1643831166.371395] [filter]   virtual devices forbidden:  yes
[5514]: <debug> [1643831166.375897] [filter]   net devices allowed:        yes
[5514]: <debug> [1643831166.380349] [filter]   usbmisc devices allowed:    yes
[5514]: <debug> [1643831166.384850] [filter]   rpmsg devices allowed:      yes
[5514]: <debug> [1643831166.389375] [filter]   wwan devices allowed:       yes
[5514]: <debug> [1643831166.393877] [filter]   tty devices:
[5514]: <debug> [1643831166.399467] [filter]       platform driver check:    yes
[5514]: <debug> [1643831166.403982] [filter]       driver check:             yes
[5514]: <debug> [1643831166.408551] [filter]       cdc-acm interface check:  yes
[5514]: <debug> [1643831166.413018] [filter]       with net check:           yes
[5514]: <debug> [1643831166.417652] [filter]       default:                  forbidden
[5514]: <debug> [1643831166.422793] [plugin-manager] looking for plugins in '/usr/lib/ModemManager'
[5514]: <debug> [1643831166.431665] [plugin-manager] loaded shared 'Icera' utils from '/usr/lib/ModemManager/libmm-shared-icera.so'
[5514]: <debug> [1643831166.438864] [plugin-manager] loaded shared 'Novatel' utils from '/usr/lib/ModemManager/libmm-shared-novatel.so'
[5514]: <debug> [1643831166.446068] [plugin-manager] loaded shared 'Option' utils from '/usr/lib/ModemManager/libmm-shared-option.so'
[5514]: <debug> [1643831166.454038] [plugin-manager] loaded shared 'Sierra' utils from '/usr/lib/ModemManager/libmm-shared-sierra.so'
[5514]: <debug> [1643831166.462091] [plugin-manager] loaded shared 'Telit' utils from '/usr/lib/ModemManager/libmm-shared-telit.so'
[5514]: <debug> [1643831166.468644] [plugin-manager] loaded shared 'Foxconn' utils from '/usr/lib/ModemManager/libmm-shared-foxconn.so'
[5514]: <debug> [1643831166.478609] [plugin-manager] loaded shared 'Xmm' utils from '/usr/lib/ModemManager/libmm-shared-xmm.so'
[5514]: <debug> [1643831166.487731] [plugin-manager] loaded plugin 'motorola' from '/usr/lib/ModemManager/libmm-plugin-motorola.so'
[5514]: <debug> [1643831166.492477] [filter] registered plugin whitelist product id: 22b8:3802
[5514]: <debug> [1643831166.497076] [filter] registered plugin whitelist product id: 22b8:4902
[5514]: <debug> [1643831166.503798] [plugin-manager] loaded plugin 'generic' from '/usr/lib/ModemManager/libmm-plugin-generic.so'
[5514]: <debug> [1643831166.510911] [plugin-manager] loaded plugin 'telit' from '/usr/lib/ModemManager/libmm-plugin-telit.so'
[5514]: <debug> [1643831166.515884] [filter] registered plugin whitelist vendor id: 1bc7
[5514]: <debug> [1643831166.522929] [plugin-manager] loaded plugin 'haier' from '/usr/lib/ModemManager/libmm-plugin-haier.so'
[5514]: <debug> [1643831166.527690] [filter] registered plugin whitelist vendor id: 201e
[5514]: <debug> [1643831166.552129] [plugin-manager] loaded plugin 'nokia-icera' from '/usr/lib/ModemManager/libmm-plugin-nokia-icera.so'
[5514]: <debug> [1643831166.558125] [filter] registered plugin whitelist vendor id: 0421
[5514]: <debug> [1643831166.565543] [plugin-manager] loaded plugin 'x22x' from '/usr/lib/ModemManager/libmm-plugin-x22x.so'
[5514]: <debug> [1643831166.570375] [filter] registered plugin whitelist tag: ID_MM_X22X_TAGGED
[5514]: <debug> [1643831166.574886] [filter] registered plugin whitelist vendor id: 1bbb
[5514]: <debug> [1643831166.579456] [filter] registered plugin whitelist vendor id: 0b3c
[5514]: <debug> [1643831166.588080] [plugin-manager] loaded plugin 'cinterion' from '/usr/lib/ModemManager/libmm-plugin-cinterion.so'
[5514]: <debug> [1643831166.592801] [filter] registered plugin whitelist vendor id: 1e2d
[5514]: <debug> [1643831166.597379] [filter] registered plugin whitelist vendor id: 0681
[5514]: <debug> [1643831166.604521] [plugin-manager] loaded plugin 'mtk' from '/usr/lib/ModemManager/libmm-plugin-mtk.so'
[... cut ...]
[5514]: <debug> [1643831167.019209] [plugin-manager] successfully loaded 38 plugins registering 4 subsystems: tty, net, usbmisc, wwan
[5514]: <debug> [1643831167.061318] service name 'org.freedesktop.ModemManager1' was acquired
[5514]: <debug> [1643831167.067600] [base-manager] unsupported automatic device scan...
[5514]: <info>  [1643831170.388699] caught signal, shutting down...
[5514]: <debug> [1643831170.394785] [base-manager] stopping connection in object manager server
[5514]: <info>  [1643831170.409813] ModemManager is shut down
[5514]: <debug> [1643831170.414468] [auth-provider] singleton disposed
```

aleksander0m, 2022-02-03:
Yey! So this is an issue in the openwrt packaging of dbus, which uses CMAKE, and likely broken. Can you report this as a new bug against the dbus package?

aleksander0m, 2022-02-03:
I think this issue can be closed now.
@arkanoid87 thanks for all the debugging done, you nailed it!

feckert, 2022-02-04:
@arkanoid87 Thank you for taking on this matter.
I almost suspected something like this.
But I am not a glib2 expert.

If we port this back to Autotools, then we also need to port it back to 21.02, then it will work again in this branch.

closed 2022-02-04 by feckert