openwrt/packages #17288: v4l2tools: recompile with -fPIC

BKPepe, 2021-12-08:
Maintainer: @mpromonet 
Environment: Turris Omnia, mvebu - cortex-a9+vfpv3-d16_musl_eabi, OpenWrt master

Description:

```
FAILED: v4l2tools-0.1.7 
: && /openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-g++ -Wall -DNDEBUG -L/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/usr/lib -L/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib -DPIC -fpic -specs=/openwrt-master/build/include/hardened-ld-pie.specs -znow -zrelro   -rdynamic CMakeFiles/v4l2tools-0.1.7.dir/src/v4l2compress.cpp.o -o v4l2tools-0.1.7  v4l2wrapper/libv4l2wrapper.a  -ljpeg  libyuv/libyuv.a  -lvpx  -lm && :
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: CMakeFiles/v4l2tools-0.1.7.dir/src/v4l2compress.cpp.o: relocation R_ARM_MOVW_ABS_NC against `LogLevel' can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: v4l2wrapper/libv4l2wrapper.a(V4l2Capture.cpp.o): relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: v4l2wrapper/libv4l2wrapper.a(V4l2Device.cpp.o): relocation R_ARM_MOVW_ABS_NC against `LogLevel' can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: v4l2wrapper/libv4l2wrapper.a(V4l2MmapDevice.cpp.o): relocation R_ARM_MOVW_ABS_NC against `LogLevel' can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: v4l2wrapper/libv4l2wrapper.a(V4l2Output.cpp.o): relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: v4l2wrapper/libv4l2wrapper.a(logger.cpp.o): relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: libyuv/libyuv.a(convert_from.cc.o): relocation R_ARM_MOVW_ABS_NC against `I422ToYUY2Row_C' can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: libyuv/libyuv.a(format_conversion.cc.o): relocation R_ARM_MOVW_ABS_NC against `ARGBToBayerRow_C' can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: libyuv/libyuv.a(planar_functions.cc.o): relocation R_ARM_MOVW_ABS_NC against `CopyRow_C' can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: libyuv/libyuv.a(rotate.cc.o): relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: libyuv/libyuv.a(row_common.cc.o): relocation R_ARM_MOVW_ABS_NC against `fixed_invtbl8' can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: libyuv/libyuv.a(scale.cc.o): relocation R_ARM_MOVW_ABS_NC against `ScaleRowDown2Linear_C' can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: libyuv/libyuv.a(scale_common.cc.o): relocation R_ARM_MOVW_ABS_NC against `InterpolateRow_C' can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: libyuv/libyuv.a(video_common.cc.o): relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: libyuv/libyuv.a(convert.cc.o): relocation R_ARM_MOVW_ABS_NC against `CopyRow_C' can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: libyuv/libyuv.a(convert_jpeg.cc.o): relocation R_ARM_MOVW_ABS_NC against `_ZN6libyuv12MJpegDecoder16kColorSpaceYCbCrE' can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: libyuv/libyuv.a(mjpeg_decoder.cc.o): relocation R_ARM_MOVW_ABS_NC against `_ZN6libyuv12ErrorHandlerEP18jpeg_common_struct' can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: libyuv/libyuv.a(convert_argb.cc.o): relocation R_ARM_MOVW_ABS_NC against `I444ToARGBRow_C' can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: CMakeFiles/v4l2tools-0.1.7.dir/src/v4l2compress.cpp.o(.text+0xf80): unresolvable R_ARM_MOVW_ABS_NC relocation against symbol `_ZSt4cout@@GLIBCXX_3.4'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make[2]: *** [Makefile:47: /openwrt-master/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/v4l2tools-0.1.7/.built] Error 1
```

mpromonet, 2021-12-11:
Hi @BKPepe 

There is no really problem to set `CMAKE_POSITION_INDEPENDENT_CODE` to `ON`, but I don't understand where come from the options in your build command `-DPIC -fpic`. This is usually interesting to link shared library, but not for executable ?

Best Regards,
Michel.

micmac1, 2021-12-31:
Looks like these flags come from include/hardening.mk (uses $FPIC var, which is set in rules.mk). Maybe just set PKG_ASLR_PIE:=0 as long as it's not supported.

BKPepe, 2022-02-09:
Maybe @neheb might know what to do with this.

neheb, 2022-02-09:
My suggestion is to replace the build system with meson.

mpromonet, 2022-02-09:
v0.1.8 add -fPIC, maybe upgrading to v0.1.8 solve the problem ?

BKPepe, 2022-02-09:
Hmm, I will try it and let you know if it works. Otherwise, I will disable ASLR_PIE as suggested.

closed 2022-02-11 by neheb