openwrt/packages #17308: meson issues with glib2?

OUZYcn, 2021-12-11:
Name:meson
Maintainer: @Andre Heider
Openwrt branch : 21.02
host: Ubuntu 20.04

Question: may i know how to fix this issue? I tried "make clean" after error happened first time then "make -j1 V=sc" again. 
same issue, so that i tried :
```
make -j1 package/feeds/packages/glib2/compile V=sc
```

same error copied below:

```
make[2]: Entering directory '/home/ouzy/kiddin9/openwrt/feeds/packages/libs/glib2'
mkdir -p /home/ouzy/kiddin9/openwrt/dl
SHELL= flock /home/ouzy/kiddin9/openwrt/tmp/.glib-2.66.4.tar.xz.flock -c '      /home/ouzy/kiddin9/openwrt/scripts/download.pl "/home/ouzy/kiddin9/openwrt/dl" "glib-2.66.4.tar.xz" "97df8670e32f9fd4f7392b0980e661dd625012015d58350da1e58e343f4af984" "" "@GNOME/glib/2.66"    '
rm -f /home/ouzy/kiddin9/openwrt/build_dir/target-x86_64_musl/glib-2.66.4/.configured_*
rm -f /home/ouzy/kiddin9/openwrt/staging_dir/target-x86_64_musl/stamp/.glib2_installed
/home/ouzy/kiddin9/openwrt/staging_dir/host/bin/sed -e "s|@CC@|'ccache', 'gcc',|" -e "s|@CXX@|'ccache', 'g++',|" -e "s|@PKGCONFIG@|/home/ouzy/kiddin9/openwrt/staging_dir/host/bin/pkg-config|" -e "s|@CFLAGS@|'-O2', '-I/home/ouzy/kiddin9/openwrt/staging_dir/host/include', '-I/home/ouzy/kiddin9/openwrt/staging_dir/hostpkg/include', '-I/home/ouzy/kiddin9/openwrt/staging_dir/target-x86_64_musl/host/include', '-I/home/ouzy/kiddin9/openwrt/staging_dir/host/include', '-I/home/ouzy/kiddin9/openwrt/staging_dir/hostpkg/include', '-I/home/ouzy/kiddin9/openwrt/staging_dir/target-x86_64_musl/host/include',|" -e "s|@CXXFLAGS@|'-I/home/ouzy/kiddin9/openwrt/staging_dir/host/include', '-I/home/ouzy/kiddin9/openwrt/staging_dir/hostpkg/include', '-I/home/ouzy/kiddin9/openwrt/staging_dir/target-x86_64_musl/host/include',|" -e "s|@LDFLAGS@|'-L/home/ouzy/kiddin9/openwrt/staging_dir/host/lib', '-L/home/ouzy/kiddin9/openwrt/staging_dir/hostpkg/lib', '-L/home/ouzy/kiddin9/openwrt/staging_dir/target-x86_64_musl/host/lib', '-liconv', '-Wl,-rpath,/home/ouzy/kiddin9/openwrt/staging_dir/hostpkg/lib',|" -e "s|@PREFIX@|/home/ouzy/kiddin9/openwrt/staging_dir/hostpkg|" < /home/ouzy/kiddin9/openwrt/staging_dir/hostpkg/lib/meson/openwrt-native.txt.in > /home/ouzy/kiddin9/openwrt/build_dir/target-x86_64_musl/glib-2.66.4/openwrt-native.txt
/home/ouzy/kiddin9/openwrt/staging_dir/host/bin/sed -e "s|@CC@|'ccache_cc',|" -e "s|@CXX@|'ccache_cxx',|" -e "s|@AR@|x86_64-openwrt-linux-musl-gcc-ar|" -e "s|@STRIP@|x86_64-openwrt-linux-musl-strip|" -e "s|@NM@|x86_64-openwrt-linux-musl-gcc-nm|" -e "s|@PKGCONFIG@|/home/ouzy/kiddin9/openwrt/staging_dir/host/bin/pkg-config|" -e "s|@CFLAGS@|'-O2', '-pipe', '-fno-caller-saves', '-fno-plt', '-fhonour-copts', '-Wno-error=unused-but-set-variable', '-Wno-error=unused-result', '-fmacro-prefix-map=/home/ouzy/kiddin9/openwrt/build_dir/target-x86_64_musl/glib-2.66.4=glib-2.66.4', '-Wformat', '-Werror=format-security', '-fstack-protector', '-D_FORTIFY_SOURCE=1', '-Wl,-z,now', '-Wl,-z,relro', '-I/home/ouzy/kiddin9/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include', '-I/home/ouzy/kiddin9/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include', '-ffunction-sections', '-fdata-sections', '-I/home/ouzy/kiddin9/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include', '-I/home/ouzy/kiddin9/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify', '-I/home/ouzy/kiddin9/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include', '-I/home/ouzy/kiddin9/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include', '-I/home/ouzy/kiddin9/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include',|" -e "s|@CXXFLAGS@|'-O2', '-pipe', '-fno-caller-saves', '-fno-plt', '-fhonour-copts', '-Wno-error=unused-but-set-variable', '-Wno-error=unused-result', '-fmacro-prefix-map=/home/ouzy/kiddin9/openwrt/build_dir/target-x86_64_musl/glib-2.66.4=glib-2.66.4', '-Wformat', '-Werror=format-security', '-fstack-protector', '-D_FORTIFY_SOURCE=1', '-Wl,-z,now', '-Wl,-z,relro', '-I/home/ouzy/kiddin9/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include', '-I/home/ouzy/kiddin9/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include', '-ffunction-sections', '-fdata-sections', '-I/home/ouzy/kiddin9/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include', '-I/home/ouzy/kiddin9/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify', '-I/home/ouzy/kiddin9/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include', '-I/home/ouzy/kiddin9/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include', '-I/home/ouzy/kiddin9/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include',|" -e "s|@LDFLAGS@|'-L/home/ouzy/kiddin9/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib', '-L/home/ouzy/kiddin9/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib', '-znow', '-zrelro', '-L/home/ouzy/kiddin9/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib', '-Wl,-rpath-link=/home/ouzy/kiddin9/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib', '-L/home/ouzy/kiddin9/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib', '-Wl,-rpath-link=/home/ouzy/kiddin9/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib', '-Wl,--gc-sections',|" -e "s|@ARCH@|"x86_64"|" -e "s|@CPU@|"64"|" -e "s|@ENDIAN@|little|" < /home/ouzy/kiddin9/openwrt/staging_dir/hostpkg/lib/meson/openwrt-cross.txt.in > /home/ouzy/kiddin9/openwrt/build_dir/target-x86_64_musl/glib-2.66.4/openwrt-cross.txt
/home/ouzy/kiddin9/openwrt/staging_dir/host/bin/python /home/ouzy/kiddin9/openwrt/staging_dir/hostpkg/lib/meson/meson.py  --buildtype plain --native-file /home/ouzy/kiddin9/openwrt/build_dir/target-x86_64_musl/glib-2.66.4/openwrt-native.txt --cross-file /home/ouzy/kiddin9/openwrt/build_dir/target-x86_64_musl/glib-2.66.4/openwrt-cross.txt -Ddefault_library=both -Diconv=external -Dselinux=disabled -Dlibmount=disabled -Dinternal_pcre=true -Dman=false -Ddtrace=false -Dsystemtap=false -Dgtk_doc=false -Dbsymbolic_functions=true -Dforce_posix_threads=true -Dfam=false -Dinstalled_tests=false -Dnls=disabled -Doss_fuzz=disabled -Dglib_assert=false -Dglib_checks=false -Dxattr=true -Db_lto=true /home/ouzy/kiddin9/openwrt/build_dir/target-x86_64_musl/glib-2.66.4/openwrt-build /home/ouzy/kiddin9/openwrt/build_dir/target-x86_64_musl/glib-2.66.4/openwrt-build/..
Traceback (most recent call last):
  File "/home/ouzy/kiddin9/openwrt/staging_dir/hostpkg/lib/meson/meson.py", line 18, in <module>
    from mesonbuild._pathlib import Path
  File "/home/ouzy/kiddin9/openwrt/staging_dir/hostpkg/lib/meson/mesonbuild/_pathlib.py", line 38
    def resolve(self, strict: bool = False) -> 'Path':
                            ^
SyntaxError: invalid syntax
make[2]: *** [Makefile:133: /home/ouzy/kiddin9/openwrt/build_dir/target-x86_64_musl/glib-2.66.4/.configured_efb2d3cf01c707184fb77fd0fe149f94] Error 1
make[2]: Leaving directory '/home/ouzy/kiddin9/openwrt/feeds/packages/libs/glib2'
time: package/feeds/packages/glib2/compile#0.54#0.25#0.67
    ERROR: package/feeds/packages/glib2 failed to build.
make[1]: *** [package/Makefile:114: package/feeds/packages/glib2/compile] Error 1
make[1]: Leaving directory '/home/ouzy/kiddin9/openwrt'
make: *** [/home/ouzy/kiddin9/openwrt/include/toplevel.mk:230: package/feeds/packages/glib2/compile] Error 2
```

**nano /home/ouzy/kiddin9/openwrt/staging_dir/hostpkg/lib/meson/meson.py**

```
#!/usr/bin/env python3

# Copyright 2016 The Meson development team

# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at

#     http://www.apache.org/licenses/LICENSE-2.0

# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

import sys
from mesonbuild._pathlib import Path

# If we're run uninstalled, add the script directory to sys.path to ensure that
# we always import the correct mesonbuild modules even if PYTHONPATH is mangled
meson_exe = Path(sys.argv[0]).resolve()
if (meson_exe.parent / 'mesonbuild').is_dir():
    sys.path.insert(0, str(meson_exe.parent))

from mesonbuild import mesonmain

if __name__ == '__main__':
    sys.exit(mesonmain.main())
[... 32 lines cut ...]
    from pathlib import Path
else:
[... cut ...]
        # operator of pathlib.Path() returns a {Posix,Windows}Path object.
        class Path(type(pathlib.Path())):
            def resolve(self, strict: bool = False) -> 'Path':
                try:
                    return super().resolve()
                except FileNotFoundError:
                    if strict:
                        raise
                    return Path(os.path.normpath(str(self)))

    else:
        from pathlib import Path

from pathlib import PurePath, PureWindowsPath, PurePosixPath

```

paper42, 2021-12-11:
```
  File "/home/ouzy/kiddin9/openwrt/staging_dir/hostpkg/lib/meson/mesonbuild/_pathlib.py", line 38
    def resolve(self, strict: bool = False) -> 'Path':
                            ^
SyntaxError: invalid syntax
```
This sounds like it's trying to use Python 2, because this is valid in Python 3.

OUZYcn, 2021-12-11:
> ```
>   File "/home/ouzy/kiddin9/openwrt/staging_dir/hostpkg/lib/meson/mesonbuild/_pathlib.py", line 38
>     def resolve(self, strict: bool = False) -> 'Path':
>                             ^ [...]

So this is a bug?  
how can it fix or avoid this error so that compile can move on?

BKPepe, 2021-12-12:
@OUZYcn Thanks for reporting. Let's take a look.

- Could you check which Python3 version do you have in lang/python/python3-version.mk?
So, we can rule out my pull request, which was mentioned.

- On your host system, could you please check that you don't have installed ``python3-minimal``? If you do, please use full featured variant and be sure that you installed python3-distutils.

OUZYcn, 2021-12-12:
> @OUZYcn Thanks for reporting. Let's take a look.
> 
> * Could you check which Python3 version do you have in lang/python/python3-version.mk?

```
ouzy@ouzy:~/kiddin9$ find | grep python3-version.mk
./openwrt/feeds/packages/lang/python/python3-version.mk
```

>   So, we can rule out my pull request, which was mentioned. Also, it would be good which commit hashes from feeds are you trying to compile.

ref: https://github.com/kiddin9/OpenWrt_x86-r2s-r4s/blob/master/onekey/build.sh
```
REPO_BRANCH="$(curl -s https://api.github.com/repos/openwrt/openwrt/tags | jq -r '.[].name' | grep v21 | head -n 1 | sed -e 's/v//')"
git clone -b v$REPO_BRANCH --depth 1 https://github.com/openwrt/openwrt
```

> * On your host system, could you please check that you don't have installed python3-minimal? If you do, be sure that you have installed python3-distutils. Also, you can try it with full-featured Python3 version (if it works in your case).

```
ouzy@ouzy:~/kiddin9$ apt list | grep python3-minimal

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

python3-minimal/groovy,now 3.8.6-0ubuntu1 amd64 [installed,automatic]

```
```
ouzy@ouzy:~/kiddin9$ apt list | grep python3-distutils

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

python3-distutils-extra/groovy 2.45 all
python3-distutils/groovy-updates,groovy-security,now 3.8.10-0ubuntu1~20.10 all [installed]

```

Therefore, should I " sudo apt remove python3-minimal" then openwrt compile should continue to works?

BKPepe, 2021-12-12:
```
ouzy@ouzy:~/kiddin9$ find | grep python3-version.mk
./openwrt/feeds/packages/lang/python/python3-version.mk
```

That's not helpful, though. I was asking which version is in that file. I'm looking for value in ``PYTHON3_VERSION_MICRO``.

___

Unfortunately, I am not able to provide you any support for that script, which you linked.  

1. It says that you need to install `python2.7` for your host system, which is wrong.
Since OpenWrt 21.02, it is not required.

2. It downloads firmware from tencent.com, which is not an official mirror according to https://openwrt.org/downloads

3. It is questionable why you should use snapshots for 21.02 for the router in the production environment.
Because 21.02 is a stable release and even released with fixup release - 21.02.1

4. There are many patches, which I don't know what they are doing.
Those patches are not included in our repositories, there are missing any details what these patches do except their name.

____

I am using Ubuntu 21.10 for development purposes and I am not able to reproduce your issue.

OUZYcn, 2021-12-13:
> ```
> ouzy@ouzy:~/kiddin9$ find | grep python3-version.mk
> ./openwrt/feeds/packages/lang/python/python3-version.mk
> ```

> That's not helpful, though. I was asking which version is in that file. I'm looking for value in `PYTHON3_VERSION_MICRO`.

Sorry , here you go. 
```
#
# Copyright (C) 2007-2016 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

# Note: keep in sync with setuptools & pip
PYTHON3_VERSION_MAJOR:=3
PYTHON3_VERSION_MINOR:=9
PYTHON3_VERSION_MICRO:=7

PYTHON3_VERSION:=$(PYTHON3_VERSION_MAJOR).$(PYTHON3_VERSION_MINOR)

PYTHON3_SETUPTOOLS_PKG_RELEASE:=1
PYTHON3_PIP_PKG_RELEASE:=1

PYTHON3_SETUPTOOLS_VERSION:=57.4.0
PYTHON3_PIP_VERSION:=21.2.3

``` 

> Unfortunately, I am not able to provide you any support for that script, which you linked.
> 
> 1. It says that you need to install `python2.7` for your host system, which is wrong.
>    Since OpenWrt 21.02, it is not required.

OK, I'm not using python2.7 for dev works on host. so I will remove it. 

> 2. It downloads firmware from tencent.com, which is not an official mirror according to https://openwrt.org/downloads

Well, about official mirror or not, maybe off topic but i can told you that Tencent , Ali, Huawei could services providers and many .edu all had many linux dist mirrors. For Tencent cloud mirror, i did checked some files, sha256 checksum is same as openwrt.org repo. 

> 3. It is questionable why you should use snapshots for 21.02 for the router in the production environment.
>    Because 21.02 is a stable release and even released with fixup release - 21.02.1

thx for remind me that, I'll contact the script maintainer to looking for reasons of why. 

> 4. There are many patches, which I don't know what they are doing.
>    Those patches are not included in our repositories, there are missing any details what these patches do except their name.
> 
Somehow that patchs , my understanding, its enhancement mod such like [fullconenat ](https://github.com/Chion82/netfilter-full-cone-nat). I'm not sure which already upstream but some certainly isn't.

> I am using Ubuntu 21.10 for development purposes and I am not able to reproduce your issue.

OUZYcn, 2021-12-14:
Hi  @BKPepe 
May i have you update about this issue?

jefferyto, 2021-12-20:
meson in 21.02 uses your build host (system) Python, not host Python (compiled from the python3 package).

What do you see if you run at the terminal:

```
python3 --version
```

Also ping @dhewg

OUZYcn, 2021-12-24:
> meson in 21.02 uses your build host (system) Python, not host Python (compiled from the python3 package).
> 
> What do you see if you run at the terminal:
> 
> ```
> python3 --version
> ```
> 
> Also ping @dhewg

@jefferyto 

Sorry for reply late. 

ouzy@ubuntuesxi:~/kiddin9$ python --version
Python 2.7.18

eli-schwartz, 2021-12-24:
```
/home/ouzy/kiddin9/openwrt/staging_dir/host/bin/python /home/ouzy/kiddin9/openwrt/staging_dir/hostpkg/lib/meson/meson.py  --buildtype plain --native-file /home/ouzy/kiddin9/openwrt/build_dir/target-x86_64_musl/glib-2.66.4/openwrt-native.txt --cross-file /home/ouzy/kiddin9/openwrt/build_dir/target-x86_64_musl/glib-2.66.4/openwrt-cross.txt -Ddefault_library=both -Diconv=external -Dselinux=disabled -Dlibmount=disabled -Dinternal_pcre=true -Dman=false -Ddtrace=false -Dsystemtap=false -Dgtk_doc=false -Dbsymbolic_functions=true -Dforce_posix_threads=true -Dfam=false -Dinstalled_tests=false -Dnls=disabled -Doss_fuzz=disabled -Dglib_assert=false -Dglib_checks=false -Dxattr=true -Db_lto=true /home/ouzy/kiddin9/openwrt/build_dir/target-x86_64_musl/glib-2.66.4/openwrt-build /home/ouzy/kiddin9/openwrt/build_dir/target-x86_64_musl/glib-2.66.4/openwrt-build/..
Traceback (most recent call last):
  File "/home/ouzy/kiddin9/openwrt/staging_dir/hostpkg/lib/meson/meson.py", line 18, in <module>
    from mesonbuild._pathlib import Path
  File "/home/ouzy/kiddin9/openwrt/staging_dir/hostpkg/lib/meson/mesonbuild/_pathlib.py", line 38
    def resolve(self, strict: bool = False) -> 'Path':
                            ^
SyntaxError: invalid syntax
```

You are invoking meson using the (unversioned) python interpreter at /home/ouzy/kiddin9/openwrt/staging_dir/host/bin/python which apparently does not like the python3 syntax.

jefferyto, 2021-12-25:
> After sudo apt remove python and sudo apt autoremove, here is python info

You didn't actually need to remove Python 2. Both `$(STAGING_DIR_HOST)/bin/python` and `$(STAGING_DIR_HOST)/bin/python3` are [mapped](https://github.com/openwrt/openwrt/blob/openwrt-21.02/include/prereq-build.mk#L155-L171) to Python 3 (`python3`) not Python 2.

In any case, I suggest you start again with a clean checkout of buildroot / sdk and try compiling again.

neheb, 2022-03-10:
is this still an issue?