openwrt/packages #17311: fail2ban: does not compile (_2to3) is invalid

BKPepe, 2021-12-11:
Maintainer: @erdoukki
Environment: aarch64, Turris MOX, OpenWrt 21.02.1

Description:

```
Applying ./patches/001-fail2ban-fix-python3-script.patch using plaintext: 
patching file config/filter.d/ignorecommands/apache-fakegooglebot

Applying ./patches/CVE-2021-32749.patch using plaintext: 
patching file config/action.d/complain.conf
patching file config/action.d/dshield.conf
patching file config/action.d/mail-buffered.conf
patching file config/action.d/mail-whois-lines.conf
patching file config/action.d/mail-whois.conf
patching file config/action.d/mail.conf
error in fail2ban setup command: use_2to3 is invalid.
make[2]: *** [Makefile:84: /home/beast/beast/workspace/turris-os-packages-dragons-mox/build/build_dir/target-aarch64_cortex-a53_musl/fail2ban-0.11.2/.built] Error 1
time: package/feeds/packages/fail2ban/compile#1.82#0.20#2.86
```

erdoukki, 2021-12-11:
Thanks for the report.

May be it is related to https://github.com/openwrt/packages/issues/15099 ?

 I will look at this issue...

erdoukki, 2021-12-11:
Fail2ban upstream look to have already resolved the issue : https://github.com/fail2ban/fail2ban/issues/3098

I will test and propose a new fail2ban package based on latest upstream release !

erdoukki, 2021-12-12:
> Fail2ban upstream look to have already resolved the issue : [fail2ban/fail2ban#3098](https://github.com/fail2ban/fail2ban/issues/3098)
> 
> I will test and propose a new fail2ban package based on latest upstream release !

I do not know if the OpenWrt packaging proposal must follow upstream development or upstream release.
So I just take care of the reported issue and look to fix it only !

BKPepe, 2021-12-12:
I think you should take time to analyze it, prefer how you want to fix it, and share the final proposal with us instead of three comments in a row because then you are making it very difficult to read and understandable. You should also consider using edit.

erdoukki, 2021-12-16:
Resolved-by: https://github.com/openwrt/packages/pull/17341

closed 2022-01-10 by jefferyto