openwrt/packages #17331: zsh: Error loading shared library libncursesw.so.5: No such file or directory (needed by /bin/zsh)

xtaran, 2021-12-13:
Maintainer: @msva
Environment:

```
$ cat /etc/openwrt_version 
f14bc5cf5635edbb3ab2e29c14a260e2640e588f
$ cat /etc/openwrt_release 
DISTRIB_RELEASE='5.3.2'
DISTRIB_REVISION='r11388+90-f14bc5cf56'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='TurrisOS 5.3.2 f14bc5cf5635edbb3ab2e29c14a260e2640e588f'
DISTRIB_TAINTS='busybox'
```

Description:

zsh crashes due to missing `libncursesw.so.5` library:

```
$ zsh
Error loading shared library libncursesw.so.5: No such file or directory (needed by /bin/zsh)
Error relocating /bin/zsh: tgoto: symbol not found
Error relocating /bin/zsh: tgetflag: symbol not found
Error relocating /bin/zsh: setupterm: symbol not found
Error relocating /bin/zsh: tigetflag: symbol not found
Error relocating /bin/zsh: tgetstr: symbol not found
Error relocating /bin/zsh: tigetnum: symbol not found
Error relocating /bin/zsh: tigetstr: symbol not found
Error relocating /bin/zsh: tputs: symbol not found
Error relocating /bin/zsh: tgetent: symbol not found
Error relocating /bin/zsh: tgetnum: symbol not found
Error relocating /bin/zsh: putp: symbol not found
Error relocating /bin/zsh: tparm: symbol not found
Error relocating /bin/zsh: strcodes: symbol not found
Error relocating /bin/zsh: numcodes: symbol not found
Error relocating /bin/zsh: boolcodes: symbol not found
Error relocating /bin/zsh: numnames: symbol not found
Error relocating /bin/zsh: boolnames: symbol not found
Error relocating /bin/zsh: strnames: symbol not found
```

This issue seems also visible via https://openwrt.org/packages/pkgdata/zsh as the link below `libncursesw6` (https://openwrt.org/packages/pkgdata/libncursesw6) is colored in red and that page no more exists. I assume that this issue is also the reason why zsh is (currently) not listed on https://openwrt.org/packages/index/utilities---shells despite it belongs to that category according to https://openwrt.org/packages/pkgdata/zsh.

This breaks logging in via SSH to a user with `zsh` as login shell even if I do `ssh hostname -t /bin/sh` or try to use `scp` or `sftp`. It hence breaks SSH access completely if all non-technical (i.e. real) users have `zsh` as login shell (as I usually do).

Was able to do a `chsh -s /bin/bash root` via LuCI "Custom Commands" page to fix the followup SSH breakage at least.

hnyman, 2023-01-14:
Related to EoL 19.07  (or actually TurrisOS 5.3.2, apparently)

(and libncurses has provided libncursesw since 2016, so there has probably been something faulty in the Turris compilation)

closed 2023-01-14 by hnyman

msva, 2023-01-16:
1) it **is** listed on https://openwrt.org/packages/index/utilities---shells 
2) try to install modern owrt build and modern zsh version, please