openwrt/packages #17332: php8-pecl-http: Compilation failed.

zdz2019, 2021-12-13:
I use the latest snapshot version, 
Target System: x86_64
Build system: Ubuntu 20.04.3 LTS
Compile php8-pecl-http: compilation failed.

#### Log information:
checking for openssl/crypto.h... yes
checking for CURLOPT_TLSAUTH_TYPE in curl/curl.h... yes
checking whether CURLOPT_TLSAUTH_TYPE expects CURL_TLSAUTH_SRP... (cached) yes
checking for CURL_LOCK_DATA_SSL_SESSION in curl/curl.h... yes
checking whether curl_share accepts CURL_LOCK_DATA_SSL_SESSION... (cached) yes
checking for CURLOPT_TLS13_CIPHERS in curl/curl.h... yes
checking whether curl_easy_setopt accepts CURLOPT_TLS13_CIPHERS... configure: error: in `/home/test/openwrt/build_dir/target-x86_64_musl/pecl-php8/pecl_http-4.2.1':
configure: error: cannot run test program while cross compiling
See `config.log' for more details
make[3]: *** [Makefile:63: /home/test/openwrt/build_dir/target-x86_64_musl/pecl-php8/pecl_http-4.2.1/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 1

#### Key information in config.log:
configure:2928: x86_64-openwrt-linux-musl-gcc -V >&5
x86_64-openwrt-linux-musl-gcc: error: unrecognized command line option '-V'
x86_64-openwrt-linux-musl-gcc: fatal error: no input files
compilation terminated.
configure:2939: $? = 1
configure:2928: x86_64-openwrt-linux-musl-gcc -qversion >&5
x86_64-openwrt-linux-musl-gcc: error: unrecognized command line option '-qversion'; did you mean '--version'?
x86_64-openwrt-linux-musl-gcc: fatal error: no input files
compilation terminated.

conftest.c:9:10: fatal error: ac_nonexistent.h: No such file or directory
 #include <ac_nonexistent.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.

conftest.c:9:10: fatal error: ac_nonexistent.h: No such file or directory
 #include <ac_nonexistent.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.

x86_64-openwrt-linux-musl-gcc: error: unrecognized command line option '-R'
configure:3636: $? = 1
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME ""
| #define PACKAGE_TARNAME ""
| #define PACKAGE_VERSION ""
| #define PACKAGE_STRING ""
| #define PACKAGE_BUGREPORT ""
| #define PACKAGE_URL ""
| /* end confdefs.h.  */
| 
| int
| main ()
| {
| 
|   ;
|   return 0;
| }

conftest.c:55:20: error: expected expression before ')' token
 if (sizeof ((off_t)))
                    ^
configure:4328: $? = 1
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME ""
| #define PACKAGE_TARNAME ""
| #define PACKAGE_VERSION ""
| #define PACKAGE_STRING ""
| #define PACKAGE_BUGREPORT ""
| #define PACKAGE_URL ""
| #define STDC_HEADERS 1
| #define HAVE_SYS_TYPES_H 1
| #define HAVE_SYS_STAT_H 1
| #define HAVE_STDLIB_H 1
| #define HAVE_STRING_H 1
| #define HAVE_MEMORY_H 1
| #define HAVE_STRINGS_H 1
| #define HAVE_INTTYPES_H 1
| #define HAVE_STDINT_H 1
| #define HAVE_UNISTD_H 1
| /* end confdefs.h.  */
| #include <stdio.h>
| #ifdef HAVE_SYS_TYPES_H
| # include <sys/types.h>
| #endif
| #ifdef HAVE_SYS_STAT_H
| # include <sys/stat.h>
| #endif
| #ifdef STDC_HEADERS
| # include <stdlib.h>
| # include <stddef.h>
| #else
| # ifdef HAVE_STDLIB_H
| #  include <stdlib.h>
| # endif
| #endif
| #ifdef HAVE_STRING_H
| # if !defined STDC_HEADERS && defined HAVE_MEMORY_H
| #  include <memory.h>
| # endif
| # include <string.h>
| #endif
| #ifdef HAVE_STRINGS_H
| # include <strings.h>
| #endif
| #ifdef HAVE_INTTYPES_H
| # include <inttypes.h>
| #endif
| #ifdef HAVE_STDINT_H
| # include <stdint.h>
| #endif
| #ifdef HAVE_UNISTD_H
| # include <unistd.h>
| #endif
| int
| main ()
| {
| if (sizeof ((off_t)))
| 	    return 0;
|   ;
|   return 0;
| }

mhei, 2021-12-18:
Sorry, for late response. I've setup a VM and tried to reproduce but I failed: I can compile successfully without any issues. Can you provide more details about your configuration, e.g. diffconfig?

zdz2019, 2021-12-19:
I changed the compilation target to "mvebu" "arm_cortex-a9_vfpv3-d16", everything is normal, compile successfully.

Later, I will try to compile the x86_64 platform.

zdz2019, 2021-12-19:
I found the problem. When libcurl enables openssl, it will fail to compile, and if the default WOLFSSL is used, it will compile successfully.

mhei, 2021-12-19:
Thanks for your investigation, I'll have a look...

mhei, 2021-12-19:
I found a solution, let's see what buildbots think about...

commit 3083278f (2021-12-20): php8-pecl-http: add compile-time hint about TLS13 ciphers (fixes #17332)

When cURL is built with OpenSSL as backend SSL/TLS library,
pecl_http's configure tries to detect whether TLS 1.3 ciphers
are enabled. This does not work when cross-compiling so let's
pass it based on OpenSSL build configuration.
commit 9e9b9dc5 (2021-12-20): Merge pull request #17395 from mhei/fix-issue-17332-php8-pecl-http-compile-failure

php8-pecl-http: add compile-time hint about TLS13 ciphers (fixes #17332)
closed 2021-12-20 by commodo