openwrt/packages #17352: packages broken with Python 3.10 because of collections.abc

paper42, 2021-12-16:
Maintainer: @dddaniel @BKPepe @jmarcet
Environment: master

Description:

Python 3.10 removed some deprecated names from collections, so we should either update them or import upstream patches (which often can't be applied because versions in master are too old, 3/4 of these packages didn't get a single update since they were added o this repository):

- [ ] python-awscli @dddaniel 
https://github.com/aws/aws-cli/commit/0c9c08856bff1e60cb14ea9a2fdfa640c9af9eeb
- [ ] python-botocore @dddaniel
https://github.com/boto/botocore/commit/ebd12d8d539b2135db32c84b94f33319179e2eb3
- [ ] python-dns (dnspython) @BKPepe
https://github.com/rthalley/dnspython/commit/62d3d41382d919af4662c12876f3de6ca854fd54
- [ ] python-stem @jmarcet
https://github.com/torproject/stem/commit/36bcb170ba9097885902513640075eac2e6ce38

BKPepe, 2021-12-30:
>  python-dns (dnspython) @BKPepe

I tried a few examples mentioned on [their website](https://www.dnspython.org/examples/) and it seems that it works. :thinking: 

```
root@omnia:~# opkg install python3-dns
Installing python3-dns (1.16.0-3.10-2) to root...
Downloading https://repo.turris.cz/crashlab/omnia/packages/packages/python3-dns_1.16.0-3.10-2_arm_cortex-a9_vfpv3-d16.ipk
Configuring python3-dns.
root@omnia:~# python3
Python 3.10.0 (default, Dec 29 2021, 00:36:57) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import dns.resolver
>>> n = dns.reversename.from_address("127.0.0.1")
>>> print(n)
1.0.0.127.in-addr.arpa.
>>> print(dns.reversename.to_address(n))
127.0.0.1
```

But I will update it anyway shortly. Thanks for letting me know.

BKPepe, 2021-12-30:
PR sent - https://github.com/openwrt/packages/pull/17458

jefferyto, 2022-01-03:
Patch backported for python-stem in #17460.

closed 2026-04-05 by commodo