openwrt/packages #17402: mwan3: flush_conntrack broken

champtar, 2021-12-20:
Maintainer: @feckert @aaronjg 
Environment: 21.02

Description:
mwan3_flush_conntrack call was moved in 39f58789e603fd31b763887282fc6b042a4d9d54 and logic was inverted,
it's called only when mwan3 is globally disabled so right now flush_conntrack is broken

feckert, 2021-12-20:
You are absolutely right! Thanks for the hint, I'll have a look at it asap. 
Can you move the line from
https://github.com/openwrt/packages/blob/master/net/mwan3/files/etc/hotplug.d/iface/15-mwan3#L29
after line
https://github.com/openwrt/packages/blob/master/net/mwan3/files/etc/hotplug.d/iface/15-mwan3#L87
does this fix the issue on your use case?

champtar, 2021-12-20:
Doing your change conntrack is now flushed:
```
Mon Dec 20 17:10:45 2021 user.info mwan3-hotplug[6167]: Connection tracking flushed for interface 'wancamp1' on action 'connected'
```

closed 2021-12-28 by feckert

feckert, 2021-12-29:
@champtar Pushed to stable branch with commit https://github.com/openwrt/packages/commit/7a15a271a6c6682cb170b1a410b1c77b6bfb666e

PR #17418 (2021-12-28): mwan3: fix mwan3 flush conntrack table call
Maintainer: me and @aaronjg 
Compile tested: no
Run tested: x86_64, APU3

Description:
fixes #17402 thanks to @champtar for reporting this
We should also backport this to openwr-21.02 after merge