openwrt/packages #17406: crowdsec-firewall-bouncer: remove crowdsec package dependency

erdoukki, 2021-12-21:
Maintainer: @erdoukki 
Environment: (OpenWrt 21.02)

Description:
https://forum.openwrt.org/t/crowdsec-initial-packages-v1-2-0-for-openwrt/102648/46

> using crowdsec-firewall-bouncer on many OpenWRT devices connected with my domain LAPI server (which collect many crowdsec machines, mostly nginx), it works great. Actually, crowdsec package is not mandatory for that usage, it would be great if it was not a dependency.

erdoukki, 2021-12-21:
Resolved by https://github.com/openwrt/packages/pull/17407

iamnague, 2021-12-21:
Hi @erdoukki, not tested yet,  don't you think you should also change /etc/firewall.cs file as it enables and restarts crowdsec service ?

erdoukki, 2021-12-21:
sure...
good advice, thanks.
I modify the script and send an update to this PR.

ne20002, 2022-03-04:
Do we have a chance to get this into 21.02 as well? The size of the GeoIp database downloaded with the main program is too much for my WRT3200ACM and I just need the baouncer anyway.

erdoukki, 2022-03-04:
> Do we have a chance to get this into 21.02 as well?
Sure, thanks @ne20002 to ask, it was just missed for 21.02.

> The size of the GeoIp database downloaded with the main program is too much for my WRT3200ACM and I just need the baouncer anyway.

I also plan to propose today, actually testing the fix
see https://github.com/openwrt/packages/issues/17681#issuecomment-1058958900

erdoukki, 2022-03-04:
> > Do we have a chance to get this into 21.02 as well?
> 
> Sure, thanks @ne20002 to ask, it was just missed for 21.02.

Done: https://github.com/openwrt/packages/pull/17986

ne20002, 2022-03-06:
Hi
Dependency is gone and space seems to be no problem now.

The installation ends with an error message.

[image]

erdoukki, 2022-03-06:
two points here.
first:
crowdsec, because of cscli tool *is* required !
the crowdsec service may be disabled in your case.
I will revert the dependency remove !
Packaging only the cscli tool is not a good alternative but I will look also at it.

second:
your problem with ipset is an openWrt issue, maybe...
 not specific to CrowdSec.

ne20002, 2022-03-06:
Fixed the dependency problem.

If the dependency is really required for installing only a bouncer ... ok for me. But for Openwrt devices I assume the install including the geoIp database is too much. Routers usually don't have that much free space.

Anyway: thank you for your work and help.

erdoukki, 2022-03-06:
the issue about geoip database is also resolved in a PR of CrowdSec package already proposed and waiting for review.
the default will be, now without geoip, which must be, if needed and possible, installed manually

ne20002, 2022-03-06:
I'll wait and give it a try when it's available. :)

Also: current crowdsec pakage is 16.7 MB ... I still think a minimum package for only a bouncer talking to a lapi on another device would be a good thing for a router.

erdoukki, 2022-03-07:
a cscli command line only package will be proposed and added to the CrowdSec package.
Instead of a revert of the removal of crowdsec "full" package dependency, this may give a solution of the usages of bouncer only.
A depends of crowdsec *or* crowdsec-cli will be added to the bouncers (actually only crowdsec-firewall-bouncer is the only merged in master).

ne20002, 2022-03-07:
This sounds like a good solution.

erdoukki, 2022-03-07:
> Also: current crowdsec pakage is 16.7 MB ... I still think a minimum package for only a bouncer talking to a lapi on another device would be a good thing for a router.

on the actual (tests) build on mvebu / armada37xx, I get the two packages as:
```
- 30M crowdsec_1.3.2-11_aarch64_cortex-a53.ipk
- 7,0M crowdsec-cli_1.3.2-11_aarch64_cortex-a53.ipk
```

erdoukki, 2022-03-07:
https://github.com/openwrt/packages/pull/17989

ne20002, 2022-03-07:
I'll wait for the new packages and give it a try on my Wrt3200ACM, Thank you very much.

I still wonder if the cscli is really neccesary. I'm new to Crowdsec and just setting this up for my network, I indeed love the concept. Isn't the cscli only required for registering the bouncer at the lapi? If so, I believe this can be done manually on the lapi and copying the credentials to the bouncer's config. If so, cscli wouldn't be essential and migth be removed?

erdoukki, 2022-03-07:
> On the main LAPI server run cscli bouncers add test then on the bouncer edit the APIkey and host. cscli is not required on the bouncer node

I will remove the cscli command from the firewall-bouncer if the cscli command line is not present, then without forced dependency to crowdsec or crowdsec-cli, it will get all uses cases possible.
If crowdsec-cli cli only) is installed, the cscli command line will be used to register locally the bouncer.
If crowdsec (full agent) is present, the normal registering will work.
If no cscli command line is present, I will simply not try to use it, may be I'll just add a user message.

I will also document the different possiblity on OpenWrt Wiki...

ne20002, 2022-03-28:
Hi @erdoukki
I got it running with the following solution: https://discourse.crowdsec.net/t/install-bouncer-on-openwrt-21-02-2/721.

closed 2022-05-02 by erdoukki