openwrt/packages #17413: miniupnpd not working with upstream router/ modem with public IP

professor-jonny, 2021-12-21:
Miniupnpd is not working with my providers upstream router with a public ip address range.
if i put in my config my public ip address into my extarnal ip upnp fails to start.
if i leave it todetect its own it says it is a public address and upnp is not possible.
if i use stun to get an external interface it says i have a nat issue:

`
STUN: ext interface wan with private IP address 192.168.0.142 is now behind restrictive or symmetric NAT with public IP address 115.189.82.94 which does not support port forwarding
NAT on upstream router blocks incoming connections set by miniupnpd
Turn off NAT on upstream router or change it to full-cone NAT 1:1 type
`
There is no nat on my upstream router it is using dmz to forward all ports to my router.
I do have an upstream CGnat but if a port is requested it dynamically opens when requested from the client as I'm told  and it seems to work with my providers router, and works with my xbox with out issue.
my provider requires me to use their router (tag locked sim card)
upnp works with the providers router with no issue.

brada4, 2021-12-22:
Most likely UPNP is blocked at your provider due to inherent security issues during last years.
Functional replacement is any kind of  VPN, like wireguard or openvpn.
As per specification UPNP should work via multicast, if yours accepted commands on unicast address it would plainly open all your internal network to the world of cyber attacks.

brada4, 2022-02-01:
REF: https://www.bleepingcomputer.com/news/security/277-000-routers-exposed-to-eternal-silence-attacks-via-upnp/

DiegoJp, 2022-12-05:
> Most likely UPNP is blocked at your provider due to inherent security issues during last years. Functional replacement is any kind of VPN, like wireguard or openvpn. As per specification UPNP should [...]

He said it works with provider's router. 

Hello I'm also facing problems with UPnP, however my ISP router is not behind a cgNAT and has a public IP, OpenWRT router is behind ISP's and DMZ'ed to it.

I can easily open ports within Firewall options in LuCI and they work ok.

However it seems the culprit is the example STUN server (`stun.stunprotocol.org:3478`)

I changed it to google's `stun.l.google.com:19302` and it's ~~punching holes~~ mapping ports again!

Good luck!

professor-jonny, 2022-12-05:
I think my issue is I need some sort of UPNP/PCP proxy or relay that miniupnpd does not currently provide.
I believe the ISP provided router does some sort of forwarding to the upstream CGNAT to open ports.

Testing with the ISP provided router with my Xbox sugests all ports are open but with a third party router or a cascaded router behind the ISP router set to DMZ reveals blocked ports.

DiegoJp, 2022-12-05:
But can you do regular port forwarding?

brada4, 2022-12-06:
Yes,  you need to set openwrt wan interface in cgnat network. You know the way usb 4g works. You need to punch holss there in that network and in your router. Obviousky 1:1 NAT hides gateway address in cgnat that supposedly needs to get command with that IP

rmilecki, 2025-04-28:
A new config option (`ignore_private_ip`) was added in the upstream by https://github.com/miniupnp/miniupnp/pull/813 and is present in the 2.3.9 release. I think it may help.