openwrt/packages #17420: Kernel version and package update

BI7PRK, 2021-12-22:
尊敬的大佬们，我非常抱歉地用中文提交一个问题，希望得到解答：
## >>我的设备
<html><body>

项 | 版本
-- | --
型号 | Raspberry Pi 3 Model B Rev 1.2
架构 | ARMv8 Processor rev 4
固件版本 | OpenWrt 21.02.1 r16325-88151b8303 / LuCI openwrt-21.02 branch git-21.340.48972-61cc3b1
内核版本 | 5.4.154

</body>
</html>

## >>安装方式：
原来是OpenWrt 21.02.0 全新安装，后来再由 rpi-3-ext4-sysupgrade.img.gz 刷入升级到OpenWrt 21.02.1。

## >> 遇到的问题：

- 软件包需要更新，但是警告 **kernel 版本不兼容**

[image]

- 然而系统信息显示内核版本为**5.4.154**
[image]

### 请问题，是不是从升级补丁安装会导致内核版本号获取不正确呢？我应该如何解决这个问题呢？

feckert, 2021-12-22:
In English please!

BI7PRK, 2021-12-23:
My device's Pi 3 Model B Rev 1.2 . Firmware version is OpenWrt 21.02.1 r16325-88151b8303. In the packages update list to update warning:  **[kernel incompatible version](https://user-images.githubusercontent.com/24989210/147119524-ce9f421d-8d2b-48e1-b69b-ceee70267a02.png)**. `need= 5.4.154-1-ba6c757dfe6f69b325d19a7f237bd331 version, has been installed 5.4.143-1-ba6c757dfe6f69b325d19a7f237bd331`. but my [kernel version's 5.4.154](https://user-images.githubusercontent.com/24989210/147119972-e8ff4858-68c6-476c-a3af-e8861af30a7e.png)

My system installation is from OpenWrt 21.02.0, then upgrade to OpenWrt 21.02.1 from rpi-3-ext4-sysupgrade.img.gz.

Thanks @feckert

BKPepe, 2021-12-24:
I don't think that we can help you anyhow based on the screenshot. It would be nice if we have some more outputs (I mean logs) from ``opkg list-installed | grep kernel``, etc., if you rebooted your router, what did you do and so on to be able to reproduce.

In any case, this issue does not belong to this repository. I suggest asking for help on [OpenWrt forum](https://forum.openwrt.org/).

closed 2021-12-24 by BKPepe