openwrt/packages #17423: [RFC] CI: Add test builds for macOS

jefferyto, 2021-12-22:
Maintainer: @aparcar 
Environment: N/A

Description:
While building on macOS isn't officially supported, there is documentation in the [wiki](https://openwrt.org/docs/guide-developer/toolchain/buildroot.exigence.macosx) on how to set up a build system on macOS, and we have users doing builds on macOS and filing issues (e.g. #17306).

GitHub offers macOS runners for GitHub Actions.

Having test builds for macOS as part of the CI would make catching and fixing issues easier.

aparcar, 2021-12-23:
Hi, while I like the idea it's a bit more involved than "just" using macOS builders. For now the OpenWrt SDKs are only compiled for x86/64. If we want *fast* CI checks we'd need macOS compatible SDKs. And even then we'd only test Intel Macs. The newest ARM64 CPUs may introduce yet another path of build failures.

We could daily/weekly/monthly build SDKs for macOS {x86_64/arm64) and build test with those in the packages.git CI.

Anyone interested in working on this?

svlobanov, 2021-12-24:
I'm interested in this proposal. I can help as far as I can in fixing bugs and testing a little

I use macos (arm64) to build openwrt images, but I don't use OpenWrt SDK

svlobanov, 2022-01-06:
One more idea related to MacOS CI. We can compare packages checksum (built on normal linux host and built on macos). In theory, macos build might be different due to different cflags after autoreconf/configure scripts

aparcar, 2022-01-06:
@svlobanov since we build most of our tools within the build root system this shouldn't be a big problem, however OpenWrt itself isn't fully reproducible. We should work on that in general.

svlobanov, 2022-01-06:
@aparcar currently I'm working on fixing packages from 'packages' repo to build it on macos. As I observed some ./configure scripts have some special checks like "if `uname -s` = darwin..." regardless of cross-compiling so as a result, compiled binary might be different even if it is reproducible on Linux build systems.

brada4, 2022-01-28:
Those configure scripts are typically generated by auto* tools, it is not something that should be patched around downstream on every package.

svlobanov, 2022-01-28:
@brada4 you can check my PRs related to macos https://github.com/openwrt/packages/pulls?page=1&q=is%3Apr+author%3Asvlobanov+macos . Some packages don't have ./configure scripts, some packages use system ar/ranlib instead of openwrt toolchain ar/ranlib and so on. Most of patches are trivial, but some packages require more time to fix.

brada4, 2022-01-28:
@svlobanov I am just saying that in cases where it is about patching configure a more pragmatic approach would be to try newer autotools and pass the news upstream instead. It takes longer for initial fix of OSX build but requires no maintenance work after.

BKPepe, 2023-09-02:
This would be really helpful. 🙏 Right now, it is unclear to me, if packages are not built for me due to missing host dependency or doing something wrong, or even it work with the latest OpenWrt version, but not working with the stable one. So, overall, what I am looking for is something like what we do for buildbots - failure about packages, etc.

BKPepe, 2026-06-07:
@aparcar Ping. :-)