openwrt/packages #17437: adguardhome: Backport 0.107.0 release to openwrt-21.02?

jamesmacwhite, 2021-12-27:
Maintainer: @dobo90 @alaviss
CC: @mercyground
Environment: N/A

Description:

The recent commit for bumping the package to 0.107.0 and changes that allows easier control of the data folder path, makes the package much better for use in OpenWrt. Given 0.107.0 is a more solid stable release generally, do we think we can backport it to OpenWrt 21.02?

The 0.107.0 releases fixes the issue with the binary being broken: https://github.com/openwrt/packages/issues/16594 which is pretty critical. If the 0.107.0 release can be backported to 21.02, this provides a much more solid package for installing directly with less issues with running AGH from the start.

There has been a lot of effort from various members of the community for documenting custom installations/scripts, given the opkg package has been a bit rough in various places. The ambition is have a detailed wiki: https://openwrt.org/docs/guide-user/services/dns/adguard-home for using AdGuard Home on OpenWrt ideally through the main package install, given it's the most straightforward process. As a starting point though, the 21.02 branch really needs the 0.107.0 release before this can be possible as a starting point.

In addition, I noticed there is a draft PR here: https://github.com/openwrt/packages/pull/17431. Specifically having a config directory path of `/etc/adguardhome`, rather than just `/etc/adguardhome.yaml`. I'd agree the `/etc/adguardhome` path makes more sense and is more contained. Perhaps we should hold until this element is merged if it's agreed? In theory existing installations under the /etc/ directory should continue to function.

Either way though, great work, the adguardhome package is looking in great shape for having a much more straightforward installation to setup process.

mercyground, 2021-12-27:
Agreed on 107 being stable now. 106 has issues and its broken due to the go version being different in 21.02. 

What sort of defaults are proposed for the install of AGH? Out of the box you need to make choices and avoid the conflicts with both luci and dnsmasq. my thread for the manual install of AGH details some of these issues.

https://forum.openwrt.org/t/how-to-updated-2021-installing-adguardhome-on-openwrt/113904

Reverse proxying would allow luci and agh to co-exist. Or just installing the web interface to a different port. (i used 8080 and let them live side by side as it were)

The ideal usage for AGH would make AGH the primary DNS with associated upstream. But this requires the reconfiguration of dnsmasq as a downstream and moving its port. (If you dont do this then having dnsmasq as primary dns you add another hop and increase memory usage.)

Also there is the issue of NTP updates on boot. You can use exceptions so NTP does not use encrypted dns so it can update the time to allow proper encrypted dns usage (it will fail with incorrect time/date etc).

Careful consideration of what defaults we wish to provide to users is key, 
The other way is to just install and let the users set things up their way. Of course this still means we need to choose a port / config to enable them to set it up via the 3000 install port. And guide them with potential issues.

alaviss, 2021-12-27:
> What sort of defaults are proposed for the install of AGH? Out of the box you need to make choices and avoid the conflicts with both luci and dnsmasq.

Best we can do now is to provide guides like https://github.com/openwrt/packages/blob/master/net/unbound/files/README.md#parallel-dnsmasq.

> Careful consideration of what defaults we wish to provide to users is key,
The other way is to just install and let the users set things up their way. Of course this still means we need to choose a port / config to enable them to set it up via the 3000 install port. And guide them with potential issues.

One thing to note that AdGuard Home doesn't seem to support any kind of "seed" configuration. The installation wizard will only run if there is no configuration, and you can only set the user/pass that way.

mercyground, 2021-12-27:
i believe there is an improvement issue for the user/pass (because you cannot change it except via yaml hacking. You also have to generate a bcrypt hash of the pwd). 

Its not the only bug bear i have with yaml editing of parameters that realistically really should be editable via the web interface.

As for a seed config you can take the basic yaml file and build off that. It isnt ideal however given the variety of installations and configurations people could have. 

I gave some thought to this as i did my thread about configuring from install with AGH's script. We shouldn't be choosing for users without informing them. There will /always/ be a user that wants to do things differently. But yet we must do something in order to "plug" AGH into the router. Its one reason why i detailed things and explained whys of my choices so others can change what i did if they wanted other decisions. eg different upstreams, or different ip's etc.

I'm of the opinion we should explore and ask AGH team to improve things on their side. 

For now however till they improve things, my gut says go simple. Install AGH, let user set it up on which ever ports, let it be the parallel dns and have users add dhcp option 6 for ipv4 and ipv6 and take over dns for downstream clients that way. That is the bare minimum required editing required. Obviously to do this the wiki will have to list instructions because there is no real way to fully automate this. Its unfortuate that this results in a only "semi-automated" install but i cant see a better way unless AGH improves.

Only other way to do it would be to ask questions as part of the install and have it script generated a yaml file? but that would require an SSH install and wouldnt allow luci installs (unless you could pull that as part of the install? i dont know that much about luci)

I'll ponder this some more and get back to you all.

BKPepe, 2021-12-27:
I cherry-picked the change from the master branch as indeed it seems that adguardhome is broken in OpenWrt 21.02 (referenced issue). 

It would be nice if you could verify that it works in OpenWrt 21.02 branch. The packages should be compiled within 24/48 hours. If it works, then close this issue.

Unfortunately, I am not using it, though. But next time, it is good to verify that the new version works in the stable branch before asking for backport.

jamesmacwhite, 2021-12-27:
@BKPepe Thanks. I believe @mercyground can verify the 0.107.0 binary works fine on OpenWrt 21.02. They have been running the edge builds and also tested the stable 0.107.0 release without any issues.

It is fairly important because the binary itself in the opkg package is broken in OpenWrt 21.02 and you essentially have to download the AdGuardHome binary from GitHub to fix it in OpenWrt under the 0.106.3 release currently, the 0.107.0 release will fix that.

mercyground, 2021-12-27:
confirmed. 107 edge builds work nicely. I've not tried their official 107 stable but i imagine should be just fine. I am wanting to test with a clean 21.02 build to help figure out some defaults and passing issues upstream to AGH for them to enable better integration with OpenWrt.

mercyground, 2021-12-29:
wiki entry has been vastly updated and we had a user submitted SSH install guide added. https://openwrt.org/docs/guide-user/services/dns/adguard-home#command-line_instructions_using_ssh

I think AGH should check for either ca-certificates ca-bundle as a requirement and install. Currently I just included them as part of the the install process. But as you need one or the other as a necessity, it really be part of install check.

Fixing the config issues with https://github.com/openwrt/packages/pull/17431 would be good to have part of the 107 stable build. Certainly aid in keeping things tidier.

Cheers to all involved with improving this for everyone!

alaviss, 2021-12-29:
> I think AGH should check for either ca-certificates ca-bundle as a requirement and install. Currently I just included them as part of the the install process. But as you need one or the other as a necessity, it really should be part of install check.

It's already a part of the deps, so it should be pulled in automatically

https://github.com/openwrt/packages/blob/2b165b9b3777b5053b67903f0dc7a7cdb890ec21/net/adguardhome/Makefile#L44

mercyground, 2021-12-29:
i shall amend the wiki to drop that part of install. Cheers!

jamesmacwhite, 2021-12-30:
This can be closed as the change was backported and runs OK on 21.02,

commit f7973fc7 (2021-12-27): adguardhome: update to 0.107.0

Full changelog available at: https://github.com/AdguardTeam/AdGuardHome/releases/tag/v0.107.0

packr has been removed from build dependencies, per
https://github.com/AdguardTeam/AdGuardHome/commit/c6888326b034daea9e69de5ee0a8f37aede45892#diff-2873f79a86c0d8b3335cd7731b0ecf7dd4301eb19a82ef7a1cba7589b5252261L2

Also added the ability to configure working directory location and moved
the directory to /var. On most setups this should not change anything,
as /var is symlinked to /tmp. The move mostly benefits setups where /var
is configured to be persistent.

The wor [... cut]
closed 2021-12-30 by jamesmacwhite