openwrt/packages #17453: [19.07] domoticz does not start - Fatal Python error: drop_gil: GIL is not locked

BKPepe, 2021-12-29:
Maintainer: @stintel
Environment: Turris 1.1, powerpc_8540, OpenWrt 19.07.8

Description:
```
root@turris:~# domoticz
2021-12-29 17:29:25.571  Status: Domoticz V4.9700 (c)2012-2018 GizMoCuz
2021-12-29 17:29:25.571  Status: Build Hash: b97777b, Date: 2018-06-23 14:18:06
2021-12-29 17:29:25.572  Status: Startup Path: /usr/bin/
2021-12-29 17:29:25.779  Status: PluginSystem: Started, Python version '3.7.12'.
2021-12-29 17:29:25.787  Active notification Subsystems: gcm, http (2/13)
2021-12-29 17:29:25.788  Status: WebServer(HTTP) started on address: :: with port 8080
2021-12-29 17:29:25.799  Error: [web:443] missing SSL certificate chain file ./server_cert.pem!
2021-12-29 17:29:25.799  Error: [web:443] missing SSL certificate file ./server_cert.pem!
2021-12-29 17:29:25.799  Error: [web:443] missing SSL private key file ./server_cert.pem!
2021-12-29 17:29:25.799  Error: [web:443] missing SSL DH parameters file ./server_cert.pem!
2021-12-29 17:29:25.800  Status: WebServer(SSL) started on address: :: with port 443
2021-12-29 17:29:25.802  Status: Proxymanager started.
2021-12-29 17:29:25.803  Starting shared server on: :::6144
2021-12-29 17:29:25.804  Status: TCPServer: shared server started...
2021-12-29 17:29:25.804  Status: RxQueue: queue worker started...
2021-12-29 17:29:27.805  Status: EventSystem: reset all events...
2021-12-29 17:29:27.806  Status: EventSystem: reset all device statuses...
Fatal Python error: drop_gil: GIL is not locked

Current thread 0xb6d81d60 (most recent call first):
  File "<frozen importlib._bootstrap_external>", line 81 in _path_stat
  File "<frozen importlib._bootstrap_external>", line 87 in _path_is_mode_type
  File "<frozen importlib._bootstrap_external>", line 102 in _path_isdir
  File "<frozen importlib._bootstrap_external>", line 1447 in path_hook_for_FileFinder
  File "<frozen importlib._bootstrap_external>", line 1197 in _path_hooks
  File "<frozen importlib._bootstrap_external>", line 1221 in _path_importer_cache
  File "<frozen importlib._bootstrap_external>", line 1249 in _get_spec
  File "<frozen importlib._bootstrap_external>", line 1280 in find_spec
  File "<frozen importlib._bootstrap>", line 906 in _find_spec
  File "<frozen importlib._bootstrap>", line 963 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 983 in _find_and_load
Aborted
```

Reproduced issue mentioned here: https://forum.turris.cz/t/tos-5-3-3-domoticz/16391
It could be related to this: https://github.com/domoticz/domoticz/pull/2534

stintel, 2021-12-30:
I am no longer maintaining domoticz.

hnyman, 2023-01-14:
Fixed by  #17454

closed 2023-01-14 by hnyman