openwrt/packages #17456: adblock 4.1.13 error -> coreutils sort not found or not executable

sarla999, 2021-12-30:
the adblock cant work when upgraded to 4.1.3

this is my version
adblock runtime information
  + adblock_status  : error
  + adblock_version : 4.1.3
  + blocked_domains : 0
  + active_sources  : -
  + dns_backend     : - (), -
  + run_utils       : download: , sort: /bin/busybox, awk: /bin/busybox
  + run_ifaces      : trigger: -, report: -
  + run_directories : base: /tmp, backup: /tmp/adblock-Backup, report: /tmp/adblock-Report, jail: /tmp
  + run_flags       : backup: ✔, flush: ✘, force: ✘, search: ✘, report: ✘, mail: ✘, jail: ✘
  + last_run        : start, n/a, 511/52/386, 30.12.2021 17:24:37

the error report from  adblock.sh file 1860 line.

when I try to run sort in ash , is't ok . so how can I fix it . thanks

sarla999, 2021-12-30:
I can fix this error by "ln -s /usr/libexec/sort-coreutils /usr/bin/sort"

closed 2021-12-30 by sarla999

jamesmacwhite, 2022-01-01:
@sarla999 Another fix might have been to force reinstall the coreutils and coreutils-sort packages, someone else had a similar issue recently:

https://github.com/openwrt/packages/issues/17468