openwrt/packages #17468: adblock.sh

skyark21, 2021-12-31:
Maintainer: @dibdot
Environment: openwrt-21.02 branch (git-21.357.58218-b3cd473) / OpenWrt 21.02.1 r16325-88151b8303

Description:

Can't start adblock, the Log View say this:
Fri Dec 31 09:42:04 2021 user.err adblock-4.1.3[17869]: coreutils sort not found or not executable

Looking in the  github page I found this code.
```
# sort check
#
adb_sort="$(command -v sort)"
if [ ! -x "${adb_sort}" ] || [ "$("${adb_sort}" --version 2>/dev/null | grep -c "coreutils")" = "0" ]
then
	f_log "err" "coreutils sort not found or not executable"
fi
```

when i run the 
```
sort --version
```
It return this:
```
sort: unrecognized option: version
BusyBox v1.33.2 (2021-12-21 11:58:22 UTC) multi-call binary.

Usage: sort [-nru] [FILE]...

Sort lines of text

	-n	Sort numbers
	-r	Reverse sort order
	-s	Stable (don't sort ties alphabetically)
	-u	Suppress duplicate lines
	-z	Lines are terminated by NUL, not newline
```

I don't know if there is the bug source.
 checking the site https://man7.org/linux/man-pages/man1/sort.1.html
there is indeed the --version option, but it doesnt appere to be compiled in this package.

jamesmacwhite, 2021-12-31:
You might only have the busybox sort binary. The coreutils sort package does have the version flag:

```
sort --version
sort (GNU coreutils) 8.30
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Mike Haertel and Paul Eggert.
```

Do you have coreutils installed?

skyark21, 2022-01-01:
I tried that and the result is the same.
Acctualy I allready installed but I did it from the lua interface.
```
opkg install coreutils coreutils-sort
Package coreutils (8.32-6) installed in root is up to date.
Package coreutils-sort (8.32-6) installed in root is up to date.
``` 
I did a list installed maybe I'm missing something.
``` 
root@router:/tmp# opkg list-installed
adblock - 4.1.3-4
base-files - 1434-r16325-88151b8303
busybox - 1.33.2-1
ca-bundle - 20210119-1
cgi-io - 2021-09-08-98cef9dd-20
coreutils - 8.32-6
coreutils-sort - 8.32-6
ddns-scripts - 2.8.2-12
ddns-scripts-services - 2.8.2-12
dnsmasq - 2.85-8
dropbear - 2020.81-2
firewall - 2021-03-23-61db17ed-1
fstools - 2021-01-04-c53b1882-1
fwtool - 2019-11-12-8f7fe925-1
getrandom - 2020-10-25-9ef88681-2
hostapd-common - 2020-06-08-5a8b3662-38
ip6tables - 1.8.7-1
iptables - 1.8.7-1
iptables-mod-conntrack-extra - 1.8.7-1
iptables-mod-ipopt - 1.8.7-1
iw - 5.9-8fab0c9e-3
iwinfo - 2021-04-30-c45f0b58-2.1
jshn - 2021-05-16-b14c4688-2
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 5.4.154-1-79c5dc6db69102eb2943a96b9bec8b63
kmod-ath - 5.4.154+5.10.68-1-1
kmod-ath9k - 5.4.154+5.10.68-1-1
kmod-ath9k-common - 5.4.154+5.10.68-1-1
kmod-cfg80211 - 5.4.154+5.10.68-1-1
kmod-gpio-button-hotplug - 5.4.154-3
kmod-ifb - 5.4.154-1
kmod-ip6tables - 5.4.154-1
kmod-ipt-conntrack - 5.4.154-1
kmod-ipt-conntrack-extra - 5.4.154-1
kmod-ipt-core - 5.4.154-1
kmod-ipt-ipopt - 5.4.154-1
kmod-ipt-nat - 5.4.154-1
kmod-ipt-offload - 5.4.154-1
kmod-ipt-raw - 5.4.154-1
kmod-lib-crc-ccitt - 5.4.154-1
kmod-mac80211 - 5.4.154+5.10.68-1-1
kmod-nf-conntrack - 5.4.154-1
kmod-nf-conntrack6 - 5.4.154-1
kmod-nf-flow - 5.4.154-1
kmod-nf-ipt - 5.4.154-1
kmod-nf-ipt6 - 5.4.154-1
kmod-nf-nat - 5.4.154-1
kmod-nf-reject - 5.4.154-1
kmod-nf-reject6 - 5.4.154-1
kmod-nls-base - 5.4.154-1
kmod-phy-ath79-usb - 5.4.154-1
kmod-ppp - 5.4.154-1
kmod-pppoe - 5.4.154-1
kmod-pppox - 5.4.154-1
kmod-sched-cake - 5.4.154-1
kmod-sched-core - 5.4.154-1
kmod-slhc - 5.4.154-1
kmod-usb-core - 5.4.154-1
kmod-usb-ehci - 5.4.154-1
kmod-usb-ledtrig-usbport - 5.4.154-1
kmod-usb2 - 5.4.154-1
libblobmsg-json20210516 - 2021-05-16-b14c4688-2
libc - 1.1.24-3
libgcc1 - 8.4.0-3
libip4tc2 - 1.8.7-1
libip6tc2 - 1.8.7-1
libiwinfo-data - 2021-04-30-c45f0b58-2.1
libiwinfo-lua - 2021-04-30-c45f0b58-2.1
libiwinfo20210430 - 2021-04-30-c45f0b58-2.1
libjson-c5 - 0.15-2
libjson-script20210516 - 2021-05-16-b14c4688-2
liblua5.1.5 - 5.1.5-9
liblucihttp-lua - 2021-06-11-3dc89af4-1
liblucihttp0 - 2021-06-11-3dc89af4-1
libmnl0 - 1.0.4-2
libncurses6 - 6.2-3
libnl-tiny1 - 2020-08-05-c291088f-2
libpthread - 1.1.24-3
libubox20210516 - 2021-05-16-b14c4688-2
libubus-lua - 2021-06-30-4fc532c8-2
libubus20210630 - 2021-06-30-4fc532c8-2
libuci20130104 - 2020-10-06-52bbc99f-5
libuclient20201210 - 2021-05-14-6a6011df-1
libustream-wolfssl20201210 - 2020-12-10-68d09243-1
libwolfssl4.8.1.66253b90 - 4.8.1-stable-4
libxtables12 - 1.8.7-1
logd - 2020-10-25-9ef88681-2
lua - 5.1.5-9
luci-app-adblock - git-21.191.52202-07063ae
luci-app-ddns - git-21.349.33342-b5a40b3
luci-app-firewall - git-21.312.70727-3eac573
luci-app-opkg - git-21.312.69848-4745991
luci-app-sqm - git-21.188.55209-f161b40
luci-base - git-21.357.58218-b3cd473
luci-lib-base - git-20.232.39649-1f6dc29
luci-lib-ip - git-20.250.76529-62505bd
luci-lib-ipkg - git-18.318.71164-4bbe325
luci-lib-jsonc - git-19.317.29469-8da8f38
luci-lib-nixio - git-20.234.06894-c4a4e43
luci-mod-admin-full - git-19.253.48496-3f93650
luci-mod-network - git-21.335.80427-5091496
luci-mod-status - git-21.295.66779-853a128
luci-mod-system - git-21.305.74567-388dae9
luci-theme-bootstrap - git-21.320.44446-0cee46b
luci-theme-material - git-21.295.66888-fc702bc
mtd - 26
nano - 6.0-1
netifd - 2021-10-30-8f82742c-1
odhcp6c - 2021-01-09-53f07e90-16
odhcpd-ipv6only - 2021-07-18-bc9d317f-3
openwrt-keyring - 2021-02-20-49283916-2
opkg - 2021-06-13-1bf042dd-1
ppp - 2.4.8.git-2020-10-03-3
ppp-mod-pppoe - 2.4.8.git-2020-10-03-3
procd - 2021-02-23-37eed131-1
px5g-wolfssl - 3
rpcd - 2021-03-11-ccb75178-1
rpcd-mod-file - 2021-03-11-ccb75178-1
rpcd-mod-iwinfo - 2021-03-11-ccb75178-1
rpcd-mod-luci - 20210614
sqm-scripts - 1.5.1-1
swconfig - 12
tc-mod-iptables - 5.11.0-3
tc-tiny - 5.11.0-3
terminfo - 6.2-3
uboot-envtools - 2021.01-10
ubox - 2020-10-25-9ef88681-2
ubus - 2021-06-30-4fc532c8-2
ubusd - 2021-06-30-4fc532c8-2
uci - 2020-10-06-52bbc99f-5
uclient-fetch - 2021-05-14-6a6011df-1
uhttpd - 2021-03-21-15346de8-2
uhttpd-mod-ubus - 2021-03-21-15346de8-2
urandom-seed - 3
urngd - 2020-01-21-c7f7b6b6-1
usign - 2020-05-23-f1f65026-1
wireless-regdb - 2021.08.28-1
wpad-basic-wolfssl - 2020-06-08-5a8b3662-38
```

I'm still getting the same result

```
Sat Jan  1 09:35:41 2022 user.err adblock-4.1.3[6771]: coreutils sort not found or not executable
```

jamesmacwhite, 2022-01-01:
It could be that the busybox sort binary is taking priority in your path, even though it's installed.

Have you tried force reinstalling coreutils and coreutils-sort to see if that fixes it?

```
opkg install coreutils coreutils-sort --force-reinstall
```

Other than that a dirty fix would be to symlink the busybox sort to the coreutils one, but not sure that's the best idea.

skyark21, 2022-01-01:
That did the trick!!!!

```
root@router:~# opkg install coreutils coreutils-sort --force-reinstall
Removing package coreutils from root...
Removing package coreutils-sort from root...
Installing coreutils (8.32-6) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/mips_24kc/packages/coreutils_8.32-6_mips_24kc.ipk
Installing coreutils-sort (8.32-6) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/mips_24kc/packages/coreutils-sort_8.32-6_mips_24kc.ipk
Configuring coreutils.
Configuring coreutils-sort.
root@router:~# sort --version
sort (GNU coreutils) 8.32
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Mike Haertel and Paul Eggert.
```
And it's running

```
Sat Jan  1 09:48:02 2022 user.info adblock-4.1.3[8139]: adblock instance started ::: action: restart, priority: 0, pid: 8139
```
Thanks @jamesmacwhite

closed 2022-01-01 by skyark21

jamesmacwhite, 2022-01-01:
Sweeeeeeet. Glad to help!